evcc-io/evcc #11096: Porsche vehicle not available: cannot create vehicle 'porsche': unexpected status: 502 (Bad Gateway)

snapscanii, 2023-12-13:
### Describe the bug

Seit einigen Wochen kann evcc leider nicht mehr auf meinen Porsche zugreifen. Ich bekomme keinen SOC etc.. Gibt es eine Lösung? Danke für eure Hilfe.

### Steps to reproduce

1.
2.
3.
...
Porsche vehicle anlegen... dann kommt der Fehler, bzw. er kam von heute auf morgen :)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

interval: 5s # control cycle interval

sponsortoken: XX

meters:
- name: grid1
  type: template 
  template: solaredge-hybrid
  id: 1  
  host: 192.168.0.XX  
  port: 1502
  usage: grid  
  modbus: tcpip   
  
#- name: my_grid #alte Konfiguration mit Tibber als grid
#  type: template
#  template: tibber-pulse
#  usage: grid
#  token: XX
#  homeid: XX # Optional 

- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.0.XX  
  port: 1502
  usage: pv  
  modbus: tcpip  
  name: pv2
  
- type: template
  template: solaredge-hybrid 
  id: 2  
  host: 192.168.0.XX  
  port: 1502
  usage: pv  
  modbus: tcpip  
  name: pv3

- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.0.XX
  password: XX
  name: pv4
  
- type: template
  template: sma-inverter #neu, garage hinten
  usage: pv
  host: 192.168.0.XX
  password: XXXXX
  name: pv5
  
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.0.XX  
  port: 1502
  usage: battery
  capacity: 7  
  modbus: tcpip  
  name: battery1

chargers:
- type: template
  template: keba-modbus 
  host: 192.168.0.XX  
  name: wallbox1
  modbus: tcpip
  id: 255
  port: 502
- type: template
  template: keba-modbus
  host: 192.168.0.XX
  name: wallbox2
  modbus: tcpip
  id: 255
  port: 502

vehicles:
- name: my_car1smart
  type: smart
  #template: smart  
  title: smart
  user: XX
  password: XX
  vin: WMEXX
  icon: car
  identifiers: 
  - '0000000000000000'
  capacity: 17.6 
  phases: 1 #bislang gab es diesen Eintrag nicht
  onIdentify: # set defaults when vehicle is identified
    mode: now #pv # enable PV-charging when vehicle is identified
    #mincurrent: 6 #bislang aus 6
    #maxcurrent: 32 #bislag aus 32    
    minSoc: 30 # immediately charge to 30% regardless of mode unless "off" (disabled)
    targetSoc: 100
    priority: 3
    
- name: my_car2porsche
  type: porsche
  #template: porsche
  title: Taycan
  user: XX
  password: XX
  vin: WP0XXXXXXXXXXXXXXXXXXXX
  icon: car
  identifiers:
  - '0000000000000000'
  capacity: 93.4  
  onIdentify: # set defaults when vehicle is identified
    mode: pv # enable PV-charging when vehicle is identified
    minSoc: 20 # immediately charge to 20% regardless of mode unless "off" (disabled)
    targetSoc: 85 # limit charge to 85% 
    priority: 1  
     
- name: my_car3audi
  type: audi
  title: e-tron # Wird in der Benutzeroberfläche angezeigt (optional)
  user: XX # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: XX # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
  vin: WAUZZZXXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
  icon: car
  #identifiers:
  #- '0000000000000000'
  capacity: 95 # Akkukapazität in kWh (optional) 
  #onIdentify: # set defaults when vehicle is identified
    #mode: pv # enable PV-charging when vehicle is identified
    #minSoc: 20 # immediately charge to 20% regardless of mode unless "off" (disabled)
    #targetSoc: 85 # limit charge to 85% 
    #priority: 2   
      
loadpoints:
- title: Garage aussen
  charger: wallbox1
  mode: off # charge mode (off, now, minpv, pv)
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  priority: 2
  vehicle: my_car1smart
  resetOnDisconnect: true
- title: Garage innen
  charger: wallbox2
  mode: pv # charge mode (off, now, minpv, pv)
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  priority: 1
  vehicle: my_car3audi
  resetOnDisconnect: true

site:
  title: Büro
  meters:
    grid: grid1 # Konfiguration mit Solaredge Wechselrichter
    # grid: my_grid #tibber Konfiguration
    pvs:
    - pv2
    - pv3
    - pv4
    - pv5
    batteries:
    - battery1
  residualpower: 100 #additional houshold usage margin
  #prioritySoc: 50 # give home battery priority up to this soc (empty to disable)
  #bufferSoc: 70 # continue charging on battery above soc (0 to disable)
  #bufferStartSoc: 60 # start charging on battery above soc (0 to disable)

tariffs:
  currency: EUR 
  grid:
    type: tibber
    token: XX # access token
    homeid: XX # optional if multiple homes associated to account  
  feedin:
   # rate for feeding excess (pv) energy to the grid
   type: fixed
   price: 0.09 # 0,09 [currency]/kWh
```

### Log details

```text
rherre@raspberrypi:~ $ evcc vehicle
[main  ] INFO 2023/12/13 09:41:24 evcc 0.122.1
[main  ] INFO 2023/12/13 09:41:24 using config file: /etc/evcc.yaml
[db    ] INFO 2023/12/13 09:41:24 using sqlite database: /home/rherre/.evcc/evcc.db
[main  ] ERROR 2023/12/13 09:41:26 creating vehicle my_car2porsche failed: cannot create vehicle 'porsche': unexpected status: 502 (Bad Gateway)
my_car1smart
------------
Soc:           100%
Capacity:      17.6kWh
Charge status: B
Range:         116km
Odometer:      37228km
Identifiers:   [0000000000000000]
OnIdentified:  MinSoc_:30, TargetSoc:100, Priority:3, Mode:now
Features:      []

my_car2porsche
--------------
.Soc:     vehicle not available: cannot create vehicle 'porsche': unexpected status: 502 (Bad Gateway)
Capacity: 93.4kWh
Features: [Offline]

my_car3audi
-----------
Soc:           58%
Capacity:      95.0kWh
Charge status: A
Range:         189km
Odometer:      48995km
Finish time:   2023-12-12 13:39:00 +0100 CET
Climater:      not available
Target Soc:    100%
Features:      []
```

### What type of operating system are you running?

Linux

### Version

122.1

GrimmiMeloni, 2023-12-14:
502 Bad Gateway zeigt einen Fehler auf der Server Seite bei Porsche an. Ein Client (wie evcc) kann hier nichts tun. Wenn das nicht nur temporär auftritt wäre ggf. einmal Recherche deinerseits notwendig ob es eine völlig neue API gibt. Wenn es da was gibt, müßten wir schauen ob wir das anpassen können.

closed 2023-12-15 by andig

Jimnaldo, 2023-12-20:
> nightly

not available on HA, when will it be in the official release?