evcc-io/evcc #11099: Document issue for Modbus integration

benneharli, 2023-12-13:
### Describe the bug

Looking at the latest documentation, it's stated that all addresses are Base 0, which means we in integration with EcoStruxure Building Operation (which uses PLC addressing) need to add 1.

That seems to be correct for e.g. "Set command" which in the manual is address 4001, and we can control the EVlink writing to register 4002.

However, if we want to read "Last stop cause" which in the manual is address is 4011, we dont' get the correct reading on 4012, but on 4011.

Can we get a complete check of all registers, and make sure the documentation is correct? Or is it a firmware bug?

### Steps to reproduce

1. Define registers for "Set Command" and "Last stop cause".
2. Both should be base 0 (4001 and 4011) or base 1 (4002 and 4012) - not 4002 and 4011

### Configuration details

```yaml
Modbus TCP integration
```

### Log details

```text
No log
```

### What type of operating system are you running?

Linux

andig, 2023-12-13:
@benneharli all registers are defined according to Schneider docs. Are you suggesting the docs are broken? Is there a corrected version?

[image]

Does this issue cause any problem for running evcc?

benneharli, 2023-12-14:
@andig sorry, you're right. The problem is not the implementation, but the documentation and/or functionality of the device. If you need to change something, it should be based on the documentation. I'll close this, and get with the manufacturer.

closed 2023-12-14 by benneharli