evcc-io/evcc #11100: WB Amperfied connect.solar - 1P3P Support

maatinh, 2023-12-13:
### Describe the bug

It looks like evcc cannot make use of the 1P3P capabilities of the Amperfied connect.solar. I defined "phases: 1" on vehicle level but charging happens with 3P.

### Steps to reproduce

Use Amperfied connect.solar

### Configuration details

```yaml
...
vehicles:
  - name: Zoe
    title: Zoe PV-Laden
    type: template
    phases: 1
    template: renault
    capacity: 25 # kWh
    user: 
    password: 
    onIdentify: # set defaults when vehicle is identified
      mode: pv # enable PV-charging when vehicle is identified
      minCurrent: 7
      targetSoc: 85 # limit charge to 90%
  - name: Zoe3P
    title: Zoe Schnell-Laden
    type: template
    phases: 3
    template: renault
    capacity: 25 # kWh
    user: 
    password: 
    onIdentify: # set defaults when vehicle is identified
      mode: now # enable PV-charging when vehicle is identified
      minCurrent: 13
      targetSoc: 85 # limit charge to 90%
...
chargers:
  - name: NRGKick
    type: template
    template: nrgkick-connect
    host: 
    mac: 
    password: 
  - name: connect.solar
    type: template
    template: amperfied
    modbus: tcp
    id: 255
    host: 
    port: 
...
```

### Log details

```text
main  ] INFO 2023/12/13 11:08:31 using config file: /etc/evcc.yaml
[main  ] INFO 2023/12/13 11:08:31 starting ui and api at :7070
[main  ] ERROR 2023/12/13 11:08:32 creating vehicle Zoe3P failed: cannot create vehicle 'template': invalid key: onidentify
[main  ] ERROR 2023/12/13 11:08:32 creating vehicle Zoe failed: cannot create vehicle 'template': invalid key: onidentify
[lp-1  ] WARN 2023/12/13 11:08:32 phases not configured, assuming 3p
[site  ] WARN 2023/12/13 11:08:33 vehicle 'Zoe PV-Laden (offline)' does not support automatic detection
[site  ] WARN 2023/12/13 11:08:33 vehicle 'Zoe Schnell-Laden (offline)' does not support automatic detection
[site  ] INFO 2023/12/13 11:08:33 site config:
[site  ] INFO 2023/12/13 11:08:33   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/12/13 11:08:33     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/12/13 11:08:33     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/12/13 11:08:33   vehicles:
[site  ] INFO 2023/12/13 11:08:33     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2023/12/13 11:08:33     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/12/13 11:08:33 loadpoint 1:
[lp-1  ] INFO 2023/12/13 11:08:33   mode:        pv
[lp-1  ] INFO 2023/12/13 11:08:33   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2023/12/13 11:08:33   meters:      charge ✓
[lp-1  ] INFO 2023/12/13 11:08:33     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2023/12/13 11:08:33 vehicle updated: unknown -> Zoe PV-Laden (offline)
[lp-1  ] INFO 2023/12/13 11:08:33 vehicle updated: Zoe PV-Laden (offline) -> unknown
[lp-1  ] INFO 2023/12/13 11:08:33 car disconnected
[lp-1  ] INFO 2023/12/13 11:08:33 vehicle updated: unknown -> Zoe PV-Laden (offline)
[lp-1  ] INFO 2023/12/13 11:16:43 car connected
[lp-1  ] INFO 2023/12/13 13:32:43 start charging ->
[lp-1  ] WARN 2023/12/13 13:33:33 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2023/12/13 13:34:43 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2023/12/13 13:35:33 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2023/12/13 13:36:43 phase mismatch between 3p measured for 1p vehicle and 3p charger
[site  ] ERROR 2023/12/13 13:37:43 pv 1 energy: read failed: read tcp 172.18.0.2:55402->192.168.40.134:502: i/o timeout
[lp-1  ] WARN 2023/12/13 13:37:43 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2023/12/13 13:38:33 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2023/12/13 13:39:43 phase mismatch between 3p measured for 1p vehicle and 3p charger
[lp-1  ] WARN 2023/12/13 13:40:43 phase mismatch between 3p measured for 1p vehicle and 3p charger
```

### What type of operating system are you running?

Linux

### Version

0.122.1

maatinh, 2023-12-20:
@premultiply Mit der neuen Firmware gibt es neue Modbus-Register, so dass die WB automatisch zwischen 1P/3P umschaltet. Dazu muss die gewünschte Ladeleistung vorgebeben werden - nicht der Strom. Alternativ kann man weiterhin den Strom vorgeben und über Modbus vorgeben ob 1P oder 3P geladen werden soll. Die Doku dazu ist noch nicht veröffentlicht, die Register könnte ich aber benennen.

maatinh, 2024-01-26:
Die Dokumentation von Amperfied wurde veröffentlicht:

https://www.amperfied.de/wp-content/uploads/2023/09/Documentation-Modbus-Register-Layout-connect-series-20240124.pdf

Man kann entweder manuell zwischen 1P/3p switchen (Register 501/261) und den max. Ladestrom setzen oder man gibt die Leistung vor und die WB wechselt automatisch zwischen 1P/3P (Register 500).

cc: @premultiply

andig, 2024-02-03:
@maatinh woran erkennen wird, ob es eine connect.solar ist und die richtige firmware hat?

maatinh, 2024-02-03:
Phase Switching ist verfügbar ab Register-Layout V2.0.1. Die Firmware-Version kann mit Register 004 abgefragt werden:

[image]

[image]

Falls die Umschaltung durch die WB automatisch passieren soll in dem nicht der Strom und die Phasen sondern die Leistung vorgegeben wird, ist mindestens V2.0.2 erforderlich:

[image]

Leider habe ich nichts gefunden, wie die WB-Variante identifiziert werden kann.

Ich habe aber gesehen, dass es jetzt auch eine Disconnect-Simulation für die WB gibt (EV40?) :

[image]
[image]

Wäre toll, wenn das auch eingebaut werden könnte falls das Auto nicht von B2 nach C2 wechselt.

andig, 2024-02-04:
> Falls die Umschaltung durch die WB automatisch passieren soll in dem nicht der Strom und die Phasen sondern die Leistung vorgegeben wird, ist mindestens V2.0.2 erforderlich:

Das wollen wir nicht.

> Leider habe ich nichts gefunden, wie die WB-Variante identifiziert werden kann.

Dann sehe ich nicht wie wir das umsetzen sollen- die Funktion gibts ja nur bei der Solar.

closed 2024-02-04 by andig

maatinh, 2024-02-04:
Ich werde mal den Support kontaktieren. Eventuell kann man die Variante ja doch identifizieren und ihr könntet euch das zweite Template sparen.

maatinh, 2024-02-05:
Hallo @andig 

hier die Zuordnung von Seriennummern zu Amperfied-WB-Varianten:

**Amperfied Wallbox connect.home**
5,0m -> 00.779.2795
7,5m -> 00.779.2963

**Amperfied Wallbox connect.business**
5,0m -> 00.779.2964
7,5m -> 00.779.2965

**Amperfied Wallbox connect.solar**
5,0m -> 00.779.3056
7,5m -> 00.779.3057

Hier die Beschreibung der Modbus-Register um den String auszulesen:

[image]

andig, 2024-02-05:
Ist jetzt als separates Template umgesetzt