evcc-io/evcc #11107: Ladeplanung geht über das Ziel hinaus

mdkeil, 2023-12-14:
### Describe the bug

Seit einigen Tagen fällt mir auf, das trotz eingestelltem Ladeziel (80%) der ZielSOC morgens teilweise auf 85-90% landet.
Ich polle den SOC (via flobz) alle 5min.  Ich konnte also auch im flobz-Log feststellen, welcher SOC während der Ladung vom Auto gemeldet wurde. Dabei habe ich festgestellt, dass auch wenn das Auto bereits > 80% meldet, die Ladung nicht gestoppt wird.

### Steps to reproduce

siehe oben; aus dem Bauch heraus würde ich sagen, dass dieses Verhalten erst aufgetreten ist, nachdem ich die poll-time (when connected) auf 5min gesetzt habe, um die Vorklimatisierung zu detektieren.. und dadurch die Regelung durcheinander zu kommen scheint.

### Configuration details

```yaml
nicht relevant
```

### Log details

```text
Ich habe heute nochmal das nightly aktualisiert und das Log-Level auf debug gesetzt. Das Log liefere ich nach dem nächsten "fehlerhaften" Ladevorgang.
```

### What type of operating system are you running?

Docker container

### Version

0.122.1 nightly (e4af488a)

premultiply, 2023-12-14:
Kein Log. Bis dahin machen wir hier erstmal zu.

deadrabbit87, 2023-12-14:
Geht bei mir bei zwei LPs auch so. Es wird plan als auch Standard Soc ignoriert. Ich schau, dass ihr einen Debug Log bekommt.

mdkeil, 2023-12-14:
Mein zweiter LP ist ein schaltbarer sonoff R2 mit tasmota geflasht, wo bei Bedarf zusätzlich mein Ladeziegel angeschlossen wird.

deadrabbit87, 2023-12-14:
Hier mal mein Log. 

Eingestellt waren im Plan 45% mit Abfahrt 16:15 Uhr. 
[image]

Hier sollte er doch eigentlich aufhören?
```
Dez 14 15:56:00 hades evcc[2655751]: [lp-2  ] DEBUG 2023/12/14 15:56:00 vehicle soc: 48%
Dez 14 15:56:00 hades evcc[2655751]: [lp-2  ] DEBUG 2023/12/14 15:56:00 vehicle range: 88km
Dez 14 15:56:00 hades evcc[2655751]: [lp-2  ] DEBUG 2023/12/14 15:56:00 plan: charge 0s starting at 0001-01-01 00:53:28 +0053 LMT until 2023-12-14 16:15:00 +0100 CET (power: 11040W, avg cost: NaN)
Dez 14 15:56:00 hades evcc[2655751]: [lp-2  ] DEBUG 2023/12/14 15:56:00 plan: continuing for remaining -7m57s
Dez 14 15:56:09 hades evcc[2655751]: [site  ] DEBUG 2023/12/14 15:56:09 ----
```
[evcclog.txt](https://github.com/evcc-io/evcc/files/13674919/evcclog.txt)

mdkeil, 2023-12-15:
@premultiply 
log habe ich oben eingefügt.. nachdem erneut der geplante soc durchgeraucht ist und auch das Auto den aktuellen Soc > planned Soc gemeldet hat (soc estimated war natürlich ebenfalls > planned soc), habe ich manuell gestopt.  Was auch wieder aufgefallen ist, nachdem das Auto angeschlossen wurde, das die letze Planung nicht gelöscht wurde..

mdkeil, 2023-12-16:
Ich bin dann erstmal wieder von der nightly weg.

deadrabbit87, 2023-12-19:
@premultiply Gibt es hierzu schon was neues und braucht ihr noch was?

Mit Version `0.122.1 (19df9236` hat sich nichts geändert.

premultiply, 2023-12-19:
/cc @andig

premultiply, 2023-12-19:
Es fehlt auch noch die Konfig mit `evcc dump --cfg`...

mdkeil, 2023-12-19:
@premultiply config habe ich oben eingefügt, obgleich es für dieses Problem meiner Ansicht nach keine relevanten Punkte enthält.

deadrabbit87, 2023-12-19:
> Es fehlt auch noch die Konfig mit `evcc dump --cfg`...

HIer meine, wüsste aber auch nicht was da relevantes drin steht: 
<details>
<summary>evcc dump --cfg</summary>

```

[main  ] INFO 2023/12/19 08:46:57 evcc 0.122.1 (19df9236)
[main  ] INFO 2023/12/19 08:46:57 using config file: /etc/evcc.yaml

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

mqtt:
  broker: 172.16.0.10:1884
  topic: evcc
  user: *****
  password: *****

log: warn
levels:
  flobz: debug
  renault: debug

# unique installation id
plant: *****

interval: 10s # control cycle interval

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 172.16.10.20  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 172.16.10.30  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 172.16.10.20  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery1
  capacity: 10.2
  minsoc: 20
- type: template
  template: kostal-ksem
  id: 71  
  host: 172.16.10.10  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: shelly-1pm
  host: 172.16.30.30
  channel: 0
  name: WP_Pwr

tariffs:
 grid:
  type: tibber
  token: *****

 feedin:
   type: fixed
   price: 0.0954 # 10,48 Haus und 8,6 Garage (netto)
 co2:
   type: grÃ¼nstromindex
   zip: "83379"

vehicles:
- type: template
  template: renault
  title: Zoe
  name: zoe
  user: *****
  password: *****
  capacity: 41  
  targetSoc: 80
  mincurrent: 8
- type: template
  template: flobz
  title: e-208
  name: e-208
  url: http://172.16.0.10:5000
  vin: *****
  capacity: 50  
  targetSoc: 70
  mincurrent: 6
  wakeup_alt: true
- type: custom
  title: WÃ¤rmepumpe
  name: WP
  icon: waterheater
  phases: 1
  onIdentify: 
    mode: pv
  soc:
    source: http
    uri: http://172.16.30.40/status/
    method: get
    headers:
      - content-type: application/json
    jq: .ext_temperature."0".tC

loadpoints:
- title: Garage
  charger: warp2_Garage
  mode: pv
  phases: 0
  maxcurrent: 16
  vehicle: e-208
- title: Stellplatz
  charger: warp2_Stellplatz
  mode: pv
  phases: 0
  maxcurrent: 16
  vehicle: zoe

- title: Heizung
  charger: WP_SG
  meter: WP_Pwr
  vehicle: WP
  mode: pv
  phases: 1
  mincurrent: 5
  maxcurrent: 16
  enable: 
    threshold: 0
    delay: 1m
  disable: 
    threshold: 0
    delay: 15m
  guardduration: 15m
  priority: 3

chargers:
- name: warp2_Garage
  type: template
  template: tinkerforge-warp
  host: 172.16.0.10
  port: 1884
  topic: warp2/Garage
  energymanager: wem/Garage

- name: warp2_Stellplatz
  type: template
  template: tinkerforge-warp
  host: 172.16.0.10
  port: 1884
  topic: warp2/Stellplatz
  energymanager: wem/Stellplatz

- name: WP_SG
  type: template
  template: shelly
  host: 172.16.30.40
  standbypower: -1

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2
    battery:
    - battery1
  residualPower: 100

# Message config using evcc go-text-template rendering, evcc variables and sprig-functions
messaging:
  events:
    connect: # vehicle connect event
      title: "{{.vehicleTitle}} connected on wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} Verbindung mit Wallbox {{.title}} hergestellt mit {{round (divf .pvPower 1000) 2 }} kW PV.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    disconnect: # vehicle connected event
      title: "{{.vehicleTitle}} disconnected of wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} Verbindung zu Wallbox {{.title}} getrennt nach {{round (.connectedDuration) 1 }}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
  services:
   - type: telegram
     token: *****
     chats: *****
     - 5511037295

       # sponsortoken: *****
sponsortoken: *****

Version: `0.122.1 (19df9236)`

```

</details>

andig, 2023-12-19:
Es wird weiter geladen weil Ladeziel nicht erreicht. Alles wie erwartet.

deadrabbit87, 2023-12-19:
Dann verstehe ich hier was komplett nicht. 

Im Tab Ankunft ist Ziel SoC 80% gesetzt. 

Wenn ich jetzt im Ladeplaner 45% einstelle, dann muss er doch aufhören wenn kein PV Überschuss da ist?

Genau das passiert nämlich nicht.

Außerdem werden die 80% auch nicht eingehalten. Da sind bei mir aktuell 70 % eingetragen und das Auto war heute voll.

andig, 2023-12-19:
Verstehe ich nicht. Dein Log oben sagt:

```
Dez 14 15:56:00 hades evcc[2655751]: [lp-2  ] DEBUG 2023/12/14 15:56:00 vehicle soc: 48%
Dez 14 15:56:00 hades evcc[2655751]: [lp-2  ] DEBUG 2023/12/14 15:56:00 vehicle range: 88km
Dez 14 15:56:00 hades evcc[2655751]: [lp-2  ] DEBUG 2023/12/14 15:56:00 plan: charge 0s starting at 0001-01-01 00:53:28 +0053 LMT until 2023-12-14 16:15:00 +0100 CET (power: 11040W, avg cost: NaN)
Dez 14 15:56:00 hades evcc[2655751]: [lp-2  ] DEBUG 2023/12/14 15:56:00 plan: continuing for remaining -7m57s
```

Da ist der soc also erst bei 48%. Für Suchspiele hab ich leider keine Zeit. Da wir das eh im nächsten Release nochmal komplett umkrempeln bitte neues Issue falls es dann noch auftritt.

mdkeil, 2023-12-19:
`Wenns genug PV und/oder Batterie gibt wird also bis 80% weiter geladen.`

es wurde bei @deadrabbit87 aber mit 11kW weitergeladen, ohne dass es PV-Überschuss / Batterie gegeben hat.

PS: Mein Screenshot oben hatt ich wieder rausgenommen, da ich erst von einer späteren Ladesession ein log hochgeladen habe-- das Verhalten war aber das gleiche.. 

PPS: @deadrabbit87 und ich haben unterschiedliche Screenshots/Logs gepostet.. die aber das gleiche fehlerhafte Verhalten wiederspiegeln-- dies tritt auch nur in der **nightly** auf.

deadrabbit87, 2023-12-19:
... und es wird auch nicht nur bis 80% weiter geladen, sondern auch darüber hinaus. 

Das spiegelt jetzt  mein Log nicht wieder, kann ich aber auch gerne einen Log nachreichen.

andig, 2023-12-19:
> Das spiegelt jetzt mein Log nicht wieder,

Macht bitte ein sauberes issue, mit klarer Fehlerbeschreibung, passendem Logfile. Ich habe leider keine Muße das Problem zu reverseengineeren.

mdkeil, 2023-12-19:
Das muss doch mit diesem issue zusammenhängen, dass eine aktive Ladeplanung im nightly aktuell nicht bei erreichen einer SOC-Grenze (Session oder "global") automatisch beendet wird. Ich habe bei mir als Workaround im Auto direkt den maxSoc eingestellt bis das Problem behoben ist.. Ich habe momentan auch keine Idee, wodurch dies verursacht wird..

mdkeil, 2023-12-20:
Ich glaube, ich habe einen möglichen Ansatzpunkt gefunden:

Ich habe in der Nacht via Planer auf 90% SOC laden lassen, maxSOC stand auf 80%.

Die Ladung wurde dieses Mal auch korrekt bei 90% beendet..

```
[lp-1  ] INFO 2023/12/20 05:02:08 stop charging <-
[lp-1  ] DEBUG 2023/12/20 05:02:08 charge total import: 6261.540kWh
[lp-1  ] DEBUG 2023/12/20 05:02:08 soc estimated: 90.02% (vehicle: 90.00%)
[lp-1  ] DEBUG 2023/12/20 05:02:08 vehicle soc: 90%
[lp-1  ] DEBUG 2023/12/20 05:02:08 vehicle range: 284km
[lp-1  ] DEBUG 2023/12/20 05:02:08 plan: charge 0s starting at 0001-01-01 00:53:28 +0053 LMT until 2023-12-20 07:00:00 +0100 CET (power: 11040W, avg cost: NaN)
[lp-1  ] DEBUG 2023/12/20 05:02:08 **limitSoc reached: 90.0% > 80%**
```
Unterschied zu den vorherigen Ladevorgängen war, dass der Session-SOC immer kleiner maxSOC war.. Ich habe das Gefühl, dass bei aktiver Ladung durch den Planer diese erst beendet wird, wenn der maxSOC erreicht ist, was ja auch passen würde, wenn nach dem Erreichen des Session-SOC ausreichend PV/Battery zur Verfügung stehen würde-- In der Nacht ist es aber meistens dunkel ;)

deadrabbit87, 2023-12-20:
@mdkeil @andig 

Ich kann dazu auch etwas beitragen. Mit dem Gedanken im Hinterkopf ein neues Issue aufzumachen, habe ich heute Nacht beide Fahrzeuge (Zoe & 208) mit einem planSoc und einen limitSoc laden lassen. Bei der Zoe waren beide Socs auf 80% gesetzt und beim 208 auf 70%.

_Beide_ Fahrzeuge habe die Ladung dann bei 90% Soc gestoppt. 

Mir sind diese 90% beim 208er schon vorher aufgefallen, dachte jedoch, dass die Ladung das Fahrzeug selbst beendet hat. 

Dem ist aber nicht so, evcc ignoriert beide gesetzten Socs und stoppt die Ladung beim 90%. Siehe auch log.

[231220 evcc_debug_log.txt](https://github.com/evcc-io/evcc/files/13724388/231220.evcc_debug_log.txt)

@andig Soll ich nochmal ein issue aufmachen wo ich nur meine Konfiguration und Log anfüge?

mdkeil, 2023-12-20:
Es stellt sich also die Frage, wo die 90% limitSoc dann herkommen.. ich dache eigentlich, dass dieser durch maxSOC festgelegt wird (was auch Sinn machen würde).. bei mir hat es dann zufällig gepasst.

mdkeil, 2023-12-20:
Ok, dann hat es bei mir zufälliger Weise übereingestimmt, dass der Session-SOC ebenfalls auf 90% stand-- Nun die Frage, warum die geplante Ladung bis 90% durchzieht.

deadrabbit87, 2023-12-20:
@mdkeil Wie ich oben schon geschrieben habe, ist da mir bei keinem lp 90% eingestellt. Beim 208 ist der limitSoc auf 70% und bei der Zoe auf 80%.

@andig Woher könnten die 90% kommen? Ich denke, dass ihr das irgendwo mal im Code setzt und das dann verwendet wird.

naltatis, 2023-12-20:
Mögt ihr mal die Antwort von `http://evcc.local:7070/api/state` posten, wenn das wieder auftritt?

deadrabbit87, 2023-12-20:
Gerne. planSoc und limitSoc waren hier beim `lp-1` auf 50% gesetzt: 
<details>
<summary>Click to expand</summary>

`{"result":{"auth":{"vehicles":{}},"battery":[{"power":0,"soc":20,"capacity":10.2,"controllable":true}],"batteryCapacity":10.2,"batteryConfigured":true,"batteryDischargeControl":false,"batteryEnergy":0,"batteryMode":"hold","batteryPower":0,"batterySoc":20,"bufferSoc":70,"bufferStartSoc":94,"currency":"EUR","greenShareHome":0,"greenShareLoadpoints":0,"gridConfigured":true,"gridCurrents":[14.9,16.240000000000002,15.44],"gridEnergy":7955.297,"gridPower":10240,"gridPowers":[3270,3540,3420],"homePower":404.0981639999991,"loadpoints":[{"chargeCurrent":16,"chargeCurrents":[15.2407856,14.87061977,14.63061237],"chargeDuration":2310000000000,"chargePower":9827.131836,"chargeRemainingDuration":-1077000000000,"chargeRemainingEnergy":0,"chargeTotalImport":3184.490967,"chargedEnergy":6271.973000000344,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":true,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":50,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":50,"effectivePlanTime":"2023-12-20T16:00:00Z","effectivePriority":0,"enableThreshold":0,"enabled":true,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":50,"maxCurrent":16,"minCurrent":6,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":0,"phasesEnabled":3,"planActive":true,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":75,"sessionEnergy":6271.973000000344,"sessionPrice":1.5209930055000855,"sessionPricePerKWh":0.2425063063090358,"sessionSolarPercentage":0,"title":"Garage","vehicleCapacity":50,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"car","vehicleName":"e-208","vehicleOdometer":27552.1,"vehiclePresent":true,"vehicleRange":92,"vehicleSoc":55.28955140000062,"vehicleTargetSoc":0,"vehicleTitle":"e-208"},{"chargeCurrent":0,"chargeDuration":0,"chargePower":0,"chargeRemainingEnergy":16400,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":false,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":80,"effectiveMaxCurrent":16,"effectiveMinCurrent":8,"effectivePlanSoc":80,"effectivePlanTime":"2023-12-20T05:45:00Z","effectivePriority":0,"enableThreshold":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":0,"phasesEnabled":1,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"Stellplatz","vehicleCapacity":41,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"car","vehicleName":"zoe","vehicleOdometer":82020.14,"vehiclePresent":true,"vehicleRange":85,"vehicleSoc":44,"vehicleTargetSoc":0,"vehicleTitle":"Zoe"},{"chargeCurrent":0,"chargeDuration":0,"chargePower":8.77,"chargeTotalImport":475.41490000000005,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":"","charging":false,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":5,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":3,"enableThreshold":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":5,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":3,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"Heizung","vehicleCapacity":0,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"waterheater","vehicleName":"WP","vehicleOdometer":0,"vehiclePresent":true,"vehicleRange":0,"vehicleSoc":29.31,"vehicleTargetSoc":0,"vehicleTitle":"Wärmepumpe"}],"prioritySoc":50,"pv":[{"power":0},{"power":0}],"pvConfigured":true,"pvEnergy":0,"pvPower":0,"residualPower":100,"siteTitle":"Mein Zuhause","smartCostActive":false,"smartCostLimit":0,"smartCostType":"priceforecast","sponsor":"deadrabbit87","sponsorTokenExpires":0,"statistics":{"30d":{"avgCo2":290.88880145933376,"avgPrice":0.22686622283804794,"chargedKWh":986.9488574596656,"solarPercentage":11.083570896877122},"365d":{"avgCo2":126.6028179890782,"avgPrice":0.15682552813529688,"chargedKWh":3924.4702156499607,"solarPercentage":62.358537209829144},"total":{"avgCo2":126.6028179890782,"avgPrice":0.15682552813529688,"chargedKWh":3924.4702156499607,"solarPercentage":62.358537209829144}},"tariffCo2":75,"tariffCo2Home":75,"tariffCo2Loadpoints":75,"tariffFeedIn":0.0954,"tariffGrid":0.2475,"tariffPriceHome":0.2475,"tariffPriceLoadpoints":0.2475,"vehicles":{"WP":{"title":"Wärmepumpe"},"e-208":{"title":"e-208","minSoc":25,"limitSoc":70,"plans":[{"soc":50,"time":"2023-12-20T16:00:00Z"}]},"zoe":{"title":"Zoe","minSoc":25,"limitSoc":80,"plans":[{"soc":80,"time":"2023-12-20T05:45:00Z"}]}},"version":"0.122.1 (4a04830b)"}}
`

</details>

Wie du vermutlich siehst, war hier der aktuelle SoC vom Fahrzeug bei 55 %. Gestartet habe ich das Zielladen bei 45 % SoC.

naltatis, 2023-12-22:
Ich hab heute nach auch noch mal einen Test gemacht:

Planungsziel: 75%
Ladelimit: 80%
Ladestand nachher: 77%

Heißt, er hat 2% mehr geladen als erwartet. Das "Problem" hier ist, dass wir nach Erreichen des Ziels in das normale PV Verhalten zurückfallen und der Disable Timer (bei mir gerade 15m weil Winter) erst abläuft, bevor er abschaltet.
Heißt die 15m auf 4,2kW (3p min) machen hier die Differenz.

[image]

Ich seh das nicht als großes Problem. Allerdings wäre meine und vmtl. auch die Erwartungshaltung der meisten Nutzer, dass nach dem Erreichen des Ziels hart beendet werden sollte, wenn gerade keine Sonne scheint. Fühlt sich präziser an. Mit dem jetzigen Verhalten werden wir zwangsläufig immer mal wieder verwunderte Nutzermeldungen bekommen.

@andig kann man das einfach mit "Timer überspringen" an der Stelle lösen?

mdkeil, 2023-12-22:
Das Verhalten wäre auch nicht schlimm.. bei @deadrabbit87 und mir ist er aber nicht in den "PV-Modus" zurück gesprungen, was den disable-timer aktivieren würde. Wenn es wieder weniger windig ist und ich meine Fahrzeuge nicht mehr bis 90%+ lade, mache ich mal weitere Test-- vielleicht hat sich das "Problem" bis dato auch erledigt. Ich update auf jeden Fall solange weiterhin täglich auf das aktuelle nightly.

deadrabbit87, 2023-12-22:
@naltatis Wie von @mdkeil schon geschrieben, sind die 2-3 % nicht so schlimm. Das ist mir schon immer aufgefallen, dachte jedoch, dass das mit dem estimater zusammenhängt. Aber mit PV-Modus klingt das logisch. 

In meinem Fall wird da einfach komplett was ignoriert und immer bis 90% weiter geladen (auch wenn z.B. nur 50 % eingestellt ist). 

Das ist auch nur seit eurem Umbau der Fall.

naltatis, 2023-12-22:
@deadrabbit87 das mit den 90% ist wirklich merkwürdig. Die Ausgabe vom State sieht ok aus. Da sind die Limit wie erwartet.

Kannst du mal `sqlite3 ~/.evcc/evcc.db ".dump 'settings'"` ausführen und das Ergebnis hier reinpacken?

Alternative auch deine `evcc.db` hier posten (Achtung Ladehistorie) oder mir per Slack schicken?

andig, 2023-12-22:
> Heißt, er hat 2% mehr geladen als erwartet. Das "Problem" hier ist, dass wir nach Erreichen des Ziels in das normale PV Verhalten zurückfallen und der Disable Timer (bei mir gerade 15m weil Winter) erst abläuft, bevor er abschaltet.

> @andig kann man das einfach mit "Timer überspringen" an der Stelle lösen?

Im master ist das so gelöst:
```go
// minimum or target charging
case lp.minSocNotReached() || plannerActive:
	err = lp.fastCharging()
	lp.resetPhaseTimer()
	lp.elapsePVTimer() // let PV mode disable immediately afterwards
```

Es gibt also nicht diesen Punkt, wo der planner **nicht** mehr aktiv ist, solange er aktiv ist wird allerdings der Timer zurück gesetzt. Deshalb versteht ich eigentlich nicht, wie nach aktivem Planner noch ein Timer aktiv sein kann?

Gibts dazu ein Logfile?

deadrabbit87, 2023-12-22:
Ich hab dir die `evcc.db` geschickt. 

Trotzdem hier noch die Ausgabe: 

```

kerberos@hades:~$ sqlite3 ~/.evcc/evcc.db ".dump 'settings'"
PRAGMA foreign_keys=OFF;
BEGIN TRANSACTION;
CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`));
INSERT INTO settings VALUES('savings.started','2023-01-17T19:30:08+01:00');
INSERT INTO settings VALUES('savings.gridCharged','0.14482558694916306');
INSERT INTO settings VALUES('savings.gridCost','0.04617724218209299');
INSERT INTO settings VALUES('savings.gridSavedCost','0.045518401975366475');
INSERT INTO settings VALUES('savings.selfConsumptionCharged','0.21516479415308515');
INSERT INTO settings VALUES('savings.selfConsumptionCost','0.023668127356839378');
INSERT INTO settings VALUES('vehicle.0.targetSoc','70');
INSERT INTO settings VALUES('vehicle.0.targetEnergy','0');
INSERT INTO settings VALUES('vehicle.0.minSoc','0');
INSERT INTO settings VALUES('vehicle.0.targetTime','0001-01-01T00:00:00Z');
INSERT INTO settings VALUES('vehicle.2.targetSoc','80');
INSERT INTO settings VALUES('vehicle.2.targetEnergy','0');
INSERT INTO settings VALUES('vehicle.2.minSoc','0');
INSERT INTO settings VALUES('vehicle.2.targetTime','0001-01-01T00:00:00Z');
INSERT INTO settings VALUES('vehicle.1.targetSoc','60');
INSERT INTO settings VALUES('vehicle.1.targetEnergy','0');
INSERT INTO settings VALUES('vehicle.1.minSoc','0');
INSERT INTO settings VALUES('vehicle.1.targetTime','0001-01-01T00:00:00Z');
COMMIT;

```

> Gibts dazu ein Logfile?

Ja, siehe [hier](https://github.com/evcc-io/evcc/files/13724388/231220.evcc_debug_log.txt)

Oder meinst du ein anderes Logfile?

andig, 2023-12-22:
Gibts einen relevanten Timestamp wo das beendet hätte werden sollen?

deadrabbit87, 2023-12-22:
Beim `lp-1` hier: 
`Dez 20 04:59:46 hades evcc[1704017]: [lp-1  ] DEBUG 2023/12/20 04:59:46 vehicle soc: 70%`

und beim `lp-2` hier: 
`Dez 20 05:13:26 hades evcc[1704017]: [lp-2  ] DEBUG 2023/12/20 05:13:26 vehicle soc: 80%`

Ab da hätte der Ladevorgang gestoppt werden müssen.

deadrabbit87, 2023-12-23:
Mit `evcc version 0.122.1 (b61546f2)` hat heute Nacht das Zielladen wie gewünscht funktioniert.

Ich besitze beobachte das mal, ob der Fehler damit behoben ist.

mdkeil, 2023-12-23:
targetSoc und minSOC sind bei mir entweder auskommentiert oder bereits entfernt. Ich hatte in der Nacht nur via smartcost laden müssen mit Session Soc auf 90%.. Ladung wurde bei 92% beendet, was ich als normal bezeichnen würde.. wobei es nicht aussagekräftig ist, da sonst die Ladung bis 90% durchgerauscht ist.. ich schaue in der nächsten Woche weiter, wo ich wohl wieder nur bis 80% laden werde.

andig, 2023-12-24:
Ich mache mal zu. Wenns wieder passiert bitte mit Logfile melden.

closed 2023-12-24 by andig