evcc-io/evcc #11111: evcc crashing

maatinh, 2023-12-14:
### Describe the bug

After restarting evcc I get this error. I've enabled modbusproxy but dont think its related to the error below:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0xa57ac]

goroutine 67 [running]:
io.(*teeReader).Read(0x640c420, {0x6241800, 0x200, 0x200})
        io/io.go:620 +0x1c
bytes.(*Buffer).ReadFrom(0x6b026c0, {0x6e6898b8, 0x6b04298})
        bytes/buffer.go:211 +0xa4
github.com/evcc-io/evcc/util/request.drainBody({0x2c6d398, 0x6b04298})
        github.com/evcc-io/evcc/util/request/roundtrip.go:70 +0xac
github.com/evcc-io/evcc/util/request.(*roundTripper).RoundTrip(0x640c410, 0x6522280)
        github.com/evcc-io/evcc/util/request/roundtrip.go:106 +0x13c
net/http.send(0x6522200, {0x2c0f010, 0x640c410}, {0xc156ea4050e24bc0, 0x343d2d09b, 0x476a240})
        net/http/client.go:260 +0x550
net/http.(*Client).send(0x6b06580, 0x6522200, {0xc156ea4050e24bc0, 0x343d2d09b, 0x476a240})
        net/http/client.go:181 +0x98
net/http.(*Client).do(0x6b06580, 0x6522200)
        net/http/client.go:724 +0x8f0
net/http.(*Client).Do(...)
        net/http/client.go:590
github.com/evcc-io/evcc/util/request.(*Helper).DoJSON(0x6b04270, 0x6522200, {0x1e9d628, 0x6522180})
        github.com/evcc-io/evcc/util/request/helper.go:68 +0x90
github.com/evcc-io/evcc/vehicle/renault/kamereon.(*API).request_(0x6b065a0, {0x6aee540, 0x62}, {0x2c0eea8, 0x640c420})
        github.com/evcc-io/evcc/vehicle/renault/kamereon/api.go:54 +0x2f4
github.com/evcc-io/evcc/vehicle/renault/kamereon.(*API).request(0x6b065a0, {0x6aee540, 0x62}, {0x0, 0x0})
        github.com/evcc-io/evcc/vehicle/renault/kamereon/api.go:64 +0xdc
github.com/evcc-io/evcc/vehicle/renault/kamereon.(*API).Person(0x6b065a0, {0x6b8c210, 0x24}, {0x22b583d, 0x7})
        github.com/evcc-io/evcc/vehicle/renault/kamereon/api.go:77 +0xcc
github.com/evcc-io/evcc/vehicle.NewRenaultDaciaFromConfig({0x22b583d, 0x7}, 0x6b8e120)
        github.com/evcc-io/evcc/vehicle/renault.go:78 +0xa98
github.com/evcc-io/evcc/vehicle.init.14.func2(0x6b8e120)
        github.com/evcc-io/evcc/vehicle/renault.go:34 +0x2c
github.com/evcc-io/evcc/vehicle.NewFromConfig({0x6486184, 0x7}, 0x6b8e0c0)
        github.com/evcc-io/evcc/vehicle/config.go:63 +0x1bc
github.com/evcc-io/evcc/vehicle.NewVehicleFromTemplateConfig(0x6b06300)
        github.com/evcc-io/evcc/vehicle/template.go:18 +0x74
github.com/evcc-io/evcc/vehicle.NewFromConfig({0x6626d58, 0x8}, 0x5f72260)
        github.com/evcc-io/evcc/vehicle/config.go:63 +0x1bc
github.com/evcc-io/evcc/cmd.vehicleInstance({{0x6626d30, 0x5}, {0x6626d58, 0x8}, 0x5f72260})
        github.com/evcc-io/evcc/cmd/setup.go:274 +0x2c
github.com/evcc-io/evcc/cmd.configureVehicles.func1()
        github.com/evcc-io/evcc/cmd/setup.go:313 +0x58
golang.org/x/sync/errgroup.(*Group).Go.func1()
        golang.org/x/sync@v0.5.0/errgroup/errgroup.go:75 +0x60
created by golang.org/x/sync/errgroup.(*Group).Go in goroutine 1
        golang.org/x/sync@v0.5.0/errgroup/errgroup.go:72 +0xa0

```

### Steps to reproduce

1.

### Configuration details

```yaml
vehicles:
  - name: Zoe
    title: Zoe-PV-Laden
    type: template
    phases: 1
    template: renault
    capacity: 25 # kWh
    user: ...
    password: ...
    mode: pv # enable PV-charging when vehicle is identified
    minCurrent: 7
    targetSoc: 85 # limit charge to 90%
  - name: Zoe3P
    title: Zoe-Schnell-Laden
    type: template
    phases: 3
    template: renault
    capacity: 25 # kWh
    user: ...
    password: ...

    mode: now # enable PV-charging when vehicle is identified
    minCurrent: 13
    targetSoc: 85 # limit charge to 90%
```

### Log details

```text
[main  ] INFO 2023/12/14 17:54:52 evcc 0.122.1 (e4af488a)
[main  ] INFO 2023/12/14 17:54:52 using config file: /etc/evcc.yaml
[main  ] INFO 2023/12/14 17:54:52 starting ui and api at :7070
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0xa57ac]

goroutine 68 [running]:
io.(*teeReader).Read(0x56d46f0, {0x5bffa00, 0x200, 0x200})
        io/io.go:620 +0x1c
bytes.(*Buffer).ReadFrom(0x6022a98, {0x6e607fe0, 0x601c568})
        bytes/buffer.go:211 +0xa4
github.com/evcc-io/evcc/util/request.drainBody({0x2c6d398, 0x601c568})
        github.com/evcc-io/evcc/util/request/roundtrip.go:70 +0xac
github.com/evcc-io/evcc/util/request.(*roundTripper).RoundTrip(0x56d46e0, 0x5a20880)
        github.com/evcc-io/evcc/util/request/roundtrip.go:106 +0x13c
net/http.send(0x5a20680, {0x2c0f010, 0x56d46e0}, {0xc156eaba74edb973, 0x369fb512c, 0x476a240})
        net/http/client.go:260 +0x550
net/http.(*Client).send(0x600e960, 0x5a20680, {0xc156eaba74edb973, 0x369fb512c, 0x476a240})
        net/http/client.go:181 +0x98
net/http.(*Client).do(0x600e960, 0x5a20680)
        net/http/client.go:724 +0x8f0
net/http.(*Client).Do(...)
        net/http/client.go:590
github.com/evcc-io/evcc/util/request.(*Helper).DoJSON(0x601c540, 0x5a20680, {0x1e9d628, 0x5a20600})
        github.com/evcc-io/evcc/util/request/helper.go:68 +0x90
github.com/evcc-io/evcc/vehicle/renault/kamereon.(*API).request_(0x600e980, {0x5b06e00, 0x62}, {0x2c0eea8, 0x56d46f0})
        github.com/evcc-io/evcc/vehicle/renault/kamereon/api.go:54 +0x2f4
github.com/evcc-io/evcc/vehicle/renault/kamereon.(*API).request(0x600e980, {0x5b06e00, 0x62}, {0x0, 0x0})
        github.com/evcc-io/evcc/vehicle/renault/kamereon/api.go:64 +0xdc
github.com/evcc-io/evcc/vehicle/renault/kamereon.(*API).Person(0x600e980, {0x5cfa390, 0x24}, {0x22b583d, 0x7})
        github.com/evcc-io/evcc/vehicle/renault/kamereon/api.go:77 +0xcc
github.com/evcc-io/evcc/vehicle.NewRenaultDaciaFromConfig({0x22b583d, 0x7}, 0x5d1f740)
        github.com/evcc-io/evcc/vehicle/renault.go:78 +0xa98
github.com/evcc-io/evcc/vehicle.init.14.func2(0x5d1f740)
        github.com/evcc-io/evcc/vehicle/renault.go:34 +0x2c
github.com/evcc-io/evcc/vehicle.NewFromConfig({0x573ae24, 0x7}, 0x5d1f6e0)
        github.com/evcc-io/evcc/vehicle/config.go:63 +0x1bc
github.com/evcc-io/evcc/vehicle.NewVehicleFromTemplateConfig(0x5d1ec20)
        github.com/evcc-io/evcc/vehicle/template.go:18 +0x74
github.com/evcc-io/evcc/vehicle.NewFromConfig({0x58e77e8, 0x8}, 0x58cba20)
        github.com/evcc-io/evcc/vehicle/config.go:63 +0x1bc
github.com/evcc-io/evcc/cmd.vehicleInstance({{0x58e77b4, 0x3}, {0x58e77e8, 0x8}, 0x58cba20})
        github.com/evcc-io/evcc/cmd/setup.go:274 +0x2c
github.com/evcc-io/evcc/cmd.configureVehicles.func1()
        github.com/evcc-io/evcc/cmd/setup.go:313 +0x58
golang.org/x/sync/errgroup.(*Group).Go.func1()
        golang.org/x/sync@v0.5.0/errgroup/errgroup.go:75 +0x60
created by golang.org/x/sync/errgroup.(*Group).Go in goroutine 1
        golang.org/x/sync@v0.5.0/errgroup/errgroup.go:72 +0xa0
```

### What type of operating system are you running?

Docker container

### Version

0.122.1 - nightly

thierolm, 2023-12-14:
Starting evcc with only a vehicle config is NOT working!
Take https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml as example and read the documentation on how to stepwise configure evcc ...

maatinh, 2023-12-14:
> Starting evcc with only a vehicle config is NOT working! Take https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml as example and read the documentation on how to stepwise configure evcc ...

This was a snippet of the vehicles only. not the complete config.

Dodemont, 2023-12-14:
Ich habe das gleiche Problem nach dem aktuellen Update.

Wenn ich den Bereich "vehicles:" aus der Config entferne läuft evcc wieder.

maatinh, 2023-12-14:
Complete config - personal data removed:

```

network:
  schema: http
  host: evcc.local
  port: 7070
interval: 30s # control cycle interval
log: info
levels:
  site: info
  lp-1: info
  charger: debug
  lp-2: error
  cache: error
  db: error
  modbus: info
meters:
  - name: gridmeter
    type: template
    template: goodwe-hybrid
    usage: grid
    modbus: tcpip
    id: 247
    host: 192.168.40.134
    port: 502
  - name: meter_pv
    type: template
    template: goodwe-hybrid
    usage: pv
    modbus: tcpip
    id: 247
    host: 192.168.40.134
    port: 502
chargers:
  - name: NRGKick
    type: template
    template: nrgkick-connect
    host: 192.168.40.132
    mac: XXXX
    password: 'XXXX'
  - name: connect.solar
    type: template
    template: amperfied
    modbus: tcp
    id: 255
    host: 192.168.40.139
    port: 502
vehicles:
  - name: Zoe
    title: Zoe-PV-Laden
    type: template
    phases: 1
    template: renault
    capacity: 25 # kWh
    user: XXXX
    password: XXXX
    mode: pv # enable PV-charging when vehicle is identified
    minCurrent: 7
    targetSoc: 85 # limit charge to 90%
  - name: Zoe3P
    title: Zoe-Schnell-Laden
    type: template
    phases: 3
    template: renault
    capacity: 25 # kWh
    user: XXXX
    password: XXXX
    mode: now # enable PV-charging when vehicle is identified
    minCurrent: 13
    targetSoc: 85 # limit charge to 90%
site:
  title: XXXX# display name for UI
  meters:
    grid: gridmeter # grid meter
    pv:
      - meter_pv # list of pv inverters/ meters
  prioritySoc: # give home battery priority up to this soc (empty to disable)
  bufferSoc: # ignore home battery discharge above soc (empty to disable)
loadpoints:
  - title: Parkplatz
    charger: connect.solar
    vehicle: Zoe
    mode: pv
    soc:
      poll:
          mode: charging
          interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 0
    minCurrent: 7
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: -1000 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 750 # maximum import power (W)
    guardDuration: 2m # switch charger contactor not more often than this (default 5m)
  - title: Garage # display name for UI
    charger: NRGKick # charger
    mode: "pv" # set default charge mode, use "off" to disable by default if charger is publicly available
    resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      poll:
          mode: charging
          interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: -1200 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 250 # maximum import power (W)
    guardDuration: 2m # switch charger contactor not more often than this (default 5m)
    minCurrent: 7 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)
    vehicle: Zoe

```

thierolm, 2023-12-14:
Dann liegt es an der nightly ... geht auf die stable

andig, 2023-12-14:
Closed in https://github.com/evcc-io/evcc/commit/19df92369ec7e9e68bb29e3bc776e0c8d7a6f5a2

closed 2023-12-14 by andig