evcc-io/evcc #11123: Anzeige guardtimer, obwohl er nicht aktiv ist

VolkerK62, 2023-12-17:
Im UI wurde "Schaltschutz aktiv 1:53" angezeigt. Nach Ablauf der Zeit wurde aber richtigerweise weitergeladen.
Nach Aktualisierung der Webansicht (F5), war die Anzeige wieder da und zählte wieder runter. Ließ sich beliebig wiederholen.

Laienhafte Vermutung:
Im Log ist der letzte Guard Eintrag um 10:46:39.
Offensichtlich wird der nicht richtig beendet.
Erst als um 11:19 die Ladung gestoppt wurde, war auch die Anzeige im UI weg.

Auch in /api/state war der Eintrag in der Zeit vorhanden und wurde nicht aktualisiert.

[guard_log.txt](https://github.com/evcc-io/evcc/files/13695792/guard_log.txt)

andig, 2023-12-17:
Was sagt denn /api/state?

VolkerK62, 2023-12-17:
>Was sagt denn /api/state?

Ich habe mehrfach /api/state aufgerufen. Leider nicht gesichert.
Die u.g. Einträge haben sich in der Zeit 10:46-11:19  nicht verändert.
Ausgegeben wurde:
`guardAction: "enable"`
`guardRemaining: "eine lange Zahl"` deren Wert sich nicht veränderte

GrimmiMeloni, 2023-12-17:
Vielleicht Regression von #9431 ?

GrimmiMeloni, 2023-12-18:
Jo Volker, da hast Du mit Deiner "laienhaften Vermutung" ins Schwarze getroffen. :wink: 

Was ich sehe:

1. 10:44 Auto wird angesteckt. Ladung startet im PV modus. 
2. Warum der Start erfolgt ist aus dem Log nicht ersichtlich, es greift aber in jedem Fall ein Disable Threshold und der PV Timer fängt sofort an zu laufen.
3. Der PV Timer läuft sauber runter. 
4. Eine Abschaltung des Chargers erfolgt aber noch nicht, weil die Guard Duration noch läuft.
5. Bis hierhin (10:45:40) alles noch OK.

Dann verändert sich um 10:47:09 die Situation. Die Site Power fällt während wir auf den Guard Timer warten unter den Disable Threshold, und der Charge darf somit weiterlaufen. Dies führt dazu das der Codepfad der für den Guard Timer bei der Abschaltung zuständig ist nicht mehr durchlaufen wird.

https://github.com/GrimmiMeloni/evcc/blob/ff38abffd89ac789b8d1af5d17bc2363f6050f65/core/loadpoint.go#L724

@andig ich glaube wir können den Block 724-728 (warten auf Guard + ggf. elapse des Guard) aus dem Enable/Disable if-block rausziehe. Der einzige Unterschied dürfte sein, daß wir dann in jedem Intervall unabhängig ob wir schalten müssen die Betrachtung des Guard durchführen. Tut aber denke ich nicht weh.

PR #11204 (2024-01-28): Loadpoint: publish guardTimer independently from pvTimer
fix #11123
PR #11926 (2024-01-29): Loadpoint: publish guardTimer independently from pvTimer
fix #11123, contains #11204 adjusted for new master
closed 2024-01-28 by andig