evcc-io/evcc #11126: Evcc + Warp2 smart + Id3 + Solaranlage - laden schaltet nicht zwischen 1 Phase und 3 Phase

f-lehmann-gmx-de, 2023-12-17:
### Describe the bug

Ich habe mein system so konfiguriert, dass es zwischen 1 Phasig und 3 Phasig umschaltet wenn mehr Leistung der PV anlage vorhanden ist. Tatsächlich lädt das auto nicht mit 1 phase und erst wenn soviel strom von der PV Anlage kommt das es mit 3 phasig laden kann. Min strom ist auf 6A. Ich habe schon so einige konfigurationen ausprobiert aber es will nicht funktionieren. Ich hadere ob ich etwas falsch eingestellt habe oder ob das ein Software bug ist

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: id3
    type: template
    template: id
    title: ID3 Performance # Wird in der Benutzeroberfläche angezeigt (optional)
    user: xxxxxx@xxxxx.de # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxxxxxxxxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: WVW?????????????????? # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 58 # Akkukapazität in kWh (optional)
    timeout: 30s # optional 
    phases: 0
# site describes the EVU connection, PV and home battery
site:

......

loadpoints:
  - title: Garage # display name for UI
    charger: wallbox1 # charger
    meter: grid # charge meter
    mode: "pv" # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: id3    # set default vehicle (disables vehicle detection)
    resetOnDisconnect: false # set defaults when vehicle disconnects
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

    # # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      # # polling defines usage of the vehicle APIs
      # # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # # poll mode defines under which condition the vehicle API is called:
        # #   charging: update vehicle ONLY when charging (this is the recommended default)
        # #   connected: update vehicle when connected (not only charging), interval defines how often
        # #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: connected
        # # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
     delay: 1m # threshold must be exceeded for this long
     threshold: -1500 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
     delay: 2m # threshold must be exceeded for this long
     threshold: 0 # maximum import power (W)
    guardDuration: 2m # switch charger contactor not more often than this (default 5m)
```

### Log details

```text
habe dafür aktuell kein log
```

### What type of operating system are you running?

Docker container

### Version

0.122.1

closed 2023-12-17 by andig

f-lehmann-gmx-de, 2023-12-17:
@andig ich sehe das der case geschlossen worden ohne jegliche Antwort oder Tipp. Es wäre nett wenn ich einen Tipp bekommen könnte was falsch ist oder ich falsch mache. danke

andig, 2023-12-17:
Ohne Logfile lässt sich dazu nix sagen

premultiply, 2023-12-17:
Ein **Fahrzeug** mit 0 Phasen gibt es nicht.