evcc-io/evcc #11128: Battery discharge control -> Fehler beim Starten

stna1981, 2023-12-17:
### Describe the bug

Ich habe auf die aktuelle 0.122.1 aktualisiert, um beim schnellen Laden die Batterie sperren zu können. Wenn ich aber unter site die entsprechende Zeile einfüge, erscheint immer nur der Fehler

* '' has invalid keys: batterydischargecontrol

im Log und evcc startet nicht mehr.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
site:  
  title: Zuhause 1
  meters:  
    grid: grid3  
    pv:  
    - pv1  
    - pv4  
    battery:  
    - battery2  
  batteryDischargeControl: true
```

### Log details

```text
Dec 17 22:30:58 raspberrypi systemd[1]: evcc.service: Service RestartSec=10s expired, scheduling restart.
Dec 17 22:30:58 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart counter is at 1.
Dec 17 22:30:58 raspberrypi systemd[1]: Stopped evcc.
Dec 17 22:30:58 raspberrypi systemd[1]: Started evcc.
Dec 17 22:30:59 raspberrypi evcc[5720]: [main  ] INFO 2023/12/17 22:30:59 evcc 0.122.1 (0f7729d7)
Dec 17 22:30:59 raspberrypi evcc[5720]: [main  ] INFO 2023/12/17 22:30:59 using config file: /etc/evcc.yaml
Dec 17 22:30:59 raspberrypi evcc[5720]: [main  ] INFO 2023/12/17 22:30:59 starting ui and api at :7070
Dec 17 22:31:00 raspberrypi evcc[5720]: [db    ] INFO 2023/12/17 22:31:00 using sqlite database: /var/lib/evcc/evcc.db
Dec 17 22:31:00 raspberrypi evcc[5720]: [ocpp-1] DEBUG 2023/12/17 22:31:00 waiting for chargepoint: 5m0s
Dec 17 22:33:20 raspberrypi evcc[5720]: [ocpp  ] INFO 2023/12/17 22:33:20 charge point connected, registering: TACW1142821G4233
Dec 17 22:33:25 raspberrypi evcc[5720]: [main  ] FATAL 2023/12/17 22:33:25 failed configuring site: 1 error(s) decoding:
Dec 17 22:33:25 raspberrypi evcc[5720]: * '' has invalid keys: batterydischargecontrol
Dec 17 22:33:25 raspberrypi evcc[5720]: [main  ] FATAL 2023/12/17 22:33:25 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.122.1

StefanSchoof, 2023-12-17:
Ist auch noch nicht in der Version, sondern nur im nightly.

stna1981, 2023-12-18:
Es steht doch im Changelog der 0.122.1 und das ist doch keine nightly?

StefanSchoof, 2023-12-18:
Ja, aber das war ohne irgendein Gerät. Aber du hast auch die nightly version 0f7729d7

stna1981, 2023-12-18:
Hab jetzt die 0.122.1+1702865837 drauf und der Fehler kommt immer noch...

andig, 2023-12-18:
Siehe https://github.com/evcc-io/evcc/discussions/10978, der Eintrag muss einfach raus da er jetzt über's UI eingestellt wird.

closed 2023-12-18 by andig

stna1981, 2023-12-18:
Wenn ich die Option im UI einschalte, dann ist sie nach einem Browser refresh wieder deaktiviert. Und entsprechend funktioniert es auch nicht. Muss man da sonst noch was beachten?

andig, 2023-12-18:
Schau mal bitte in die Browserkonsole auf die HTTP Requests- was wird da bei Knopfdruck ausgetauscht? Was steht vor/nach Ausführung in /api/state?

stna1981, 2023-12-18:
Wenn ich die Funktion aktiviere, wird ein POST auf <IP>:7070/api/batterydischargecontrol/true abgesetzt mit Status 200. Beim deaktivieren entsprechend auf <IP>:7070/api/batterydischargecontrol/false

andig, 2023-12-18:
Wie erwartet. Und gibts einen Rückgabewert beim einschalten? Welchen? Was steht dann in /api/state?

stna1981, 2023-12-18:
Also im UI ist die Option als aktiv gekennzeichnet, das API sagt mir aber sowohl vorher als auch hinterher:

 "batteryDischargeControl" : false

Es wird also offenbar aus irgendwelchen Gründen nicht persistiert...

andig, 2023-12-18:
Wir brauchen ein *vollständiges* Issue. Bitte mal komplette steps to reproduce unter Nutzung von curl einschließlich /api/state in jedem Schritt und passendem Logfile und Config dazu.