evcc-io/evcc #11129: Odd clock icon placement

andig, 2023-12-18:
Current master on Safari:

[image]

naltatis, 2023-12-18:
Hast du nen konkreten Vorschlag? Ich seh das als unproblematisch an. Beides ist gut erkennbar. Die Alternative wäre, die Uhr noch weiter nach unten zu ziehen, was in den anderne Zuständen komisch aussieht.

andig, 2023-12-18:
Keine gute Idee. Kam mir nur merkwürdig vor weil ich das vorher nicht gesehen hatte. Kann das Handle unten vllt. einen weißen Rahmen bekommen der dann sichtbar über der Uhr liegt? Ansonsten gerne schließen- ist nicht so wichtig.

naltatis, 2023-12-20:
Ich mach das erstmal zu. Wenn mir (oder jmd anderem) noch ne elegante Lösung einfällt, machen wir das wieder auf.

closed 2023-12-20 by naltatis