evcc-io/evcc #11130: Planner: clarify vehicle not available on arrival tab

andig, 2023-12-18:
On master, user wonders why he can't select anything while vehicle is disconnected:

[image]

Add a message that settings require a connected vehicle with soc capability?

naltatis, 2023-12-18:
Das Problem ist hier das `"vehicleSoc": 0`. Das ist aktuell der einzige Weg, über den wir im UI erkennen können, ob ein Fahrzeug einen SoC hat. In dem Zustand wird auch nicht nur die Form im Arrival Tab deaktiviert. Auch der Rest der UI läuft auf kWh-basiertem Laden.

https://github.com/evcc-io/evcc/blob/32a88cc2b5623e9275ed29f17b662f97f4678838/assets/js/components/Loadpoint.vue#L244-L250

Wenn wir das sauber lösen wollen, müssen wir `socBasedCharging` und `socBasedPlanning` nur noch im Core ermitteln und jeweils als Boolean publizieren.

**Zum eigentlichen Problem:** Ja, da muss ne klare Meldung angezeigt werden.

naltatis, 2023-12-18:
Ergänzung: Hier ist die Logik zu strikt. Die Felder werden aktuell nur bei `socBasedPlanning` freigegeben. Richtig ist hier `socBasedCharging`, die Akkukapazität spielt für die beiden Funktionen keine Rolle. Ich änder das gleich mit.

PR #11138 (2023-12-18): Arrival Tab: add soc based charging notice
fixes #11130

- 💬 clarify why options are grayed out
- 🔋 allow minsoc / soc limit settings if capacity is unknown

[image]

[image]
closed 2023-12-18 by andig