evcc-io/evcc #11140: error message every few seconds: pv 2 power: strconv.ParseFloat: parsing \"<nil>\": invalid syntax

umrath, 2023-12-19:
### Describe the bug

evcc is running for a few days here now and suddenly (starting tonight I think) it's raining error messages like this:

pv 2 power: strconv.ParseFloat: parsing \"<nil>\": invalid syntax

Every few seconds this error pops up.
I do not see any real functional impact, as charging and visulizing stuff works seemingly fine. But nevertheless, it's worrying.

### Steps to reproduce

I'm not sure if I can provide steps to reproduce it.
The only thing that truly changed was me swapping the wallbox yesterday and asjusting the configuration accordingly. But the error messages started way later, so I do not see any direct connection here.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: <redacted>

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
#modbusproxy:
#    - port: 5200
#      uri: 192.168.178.63:502
#      rtu: true
#      readonly: false

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: Alpha_ESS_grid
    type: template
    template: alpha-ess-smile
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 85
    host: 192.168.179.110
    port: 502 # Port 

  - name: Alpha_ESS_pv
    type: template
    template: alpha-ess-smile
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 85
    host: 192.168.179.110
    port: 502 # Port 

  - name: Alpha_ESS_battery
    type: template
    template: alpha-ess-smile
    usage: battery

    # Modbus TCP
    modbus: tcpip
    id: 85
    host: 192.168.179.110
    port: 502 # Port 

  - type: template
    template: hoymiles-opendtu
    usage: pv
    host: 192.168.179.107
    name: Gartenkraftwerk

# charger definitions
chargers:
#  - name: Elvi
#    type: template
#    template: elvi
#    timeout: 2m # optional
#      #stationId: <redacted>
#      #connector: 1
#      #idtag: '<redacted>'
#    connecttimeout: 2m

  - type: template  
    template: go-e 
    host: 192.168.178.54  
    name: go-e
      
# vehicle definitions
vehicles:
  - name: Tesmeralda
    type: template
    template: tesla
    title: Tesmeralda
    accessToken: <redacted> # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    refreshToken: <redacted> # Siehe https://docs.evcc.io/docs/devices/vehicles#tesla
    vin: <redacted>
      #cache: 5m
    capacity: 77 # Akkukapazität in kWh
    mode: pv # enable PV-charging when vehicle is identified
    minSoc: 20 # immediately charge to value regardless of mode unless "off" (disabled)
               # value can be overwritten by configuration in the ui
    targetSoc: 73 # limit charge
                  # value can be overwritten by configuration in the ui

# site describes the EVU connection, PV and home battery
site:
  residualPower: 400 # 400W house consumption
  title: Zuhause
  meters:
    grid: Alpha_ESS_grid
    pv:
      - Alpha_ESS_pv
      - Gartenkraftwerk
    battery:
      - Alpha_ESS_battery
  prioritySoc: 10 # give home battery priority up to this soc (empty to disable)
  bufferSoc: 100 # ignore home battery discharge above soc (empty to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage # display name for UI
    mode: pv
    phases: 3
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true
    enable:
      delay: 2m
      threshold: 0 # es wird eingeschaltet, wenn 2 Minute lang 3000 W Einspeisung vorhanden sind
    disable:
      delay: 5m
      threshold: 300 # es wird ausgeschaltet, wenn 5 Minuten lang mehr als 300 W Netzstrom bezogen wurde
    charger: go-e
    guardduration: 2m

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.3262 # EUR/kWh
    #zones:
    #  - days: Mo-Fr
    #    hours: 2-5
    #    price: 0.2 # EUR/kWh
    #  - days: Sa,So
    #    price: 0.15 # EUR/kWh

    # or variable via tibber
    #type: tibber
    #token: "<redacted>" # access token
    #homeid: "<redacted>" # optional if multiple homes associated to account

    # # or variable via awattar
    # type: awattar
    # region: de # optional, choose at for Austria
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.079 # EUR/kWh
  planner:
    # planner tariff can be used for target charging if not grid tariff is specified
    # GrünStromIndex (Germany only) or ElectricityMaps provide CO2 intensity forecast

    # type: grünstromindex
    # zip: <zip>

    # type: electricitymaps
    # uri: <uri>
    # token: <token>
    # zone: DE

# mqtt message broker
mqtt:
  broker: 192.168.178.63:1883
  topic: evcc # root topic for publishing, set empty to disable
  clientid: evcc
  user: homeassistant
  password: Keeghim6die6queethoo5iesieluiruoN8IeCharaaZ8vuTh3Booba6fooPiG6bi

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# push messages
messaging:
  events:
    start: # charge start event
      title: Laden gestartet
      msg: Gestartet im Modus "${mode}"
    stop: # charge stop event
      title: Laden beendet
      msg: Laden beendet ${chargedEnergy:%.1fk}kWh in ${chargeDuration}
    connect: # vehicle connect event
      title: Auto verbunden
      msg: Auto verbunden um ${pvPower:%.1fk}kW PV
    disconnect: # vehicle connected event
      title: Auto abgesteckt
      msg: Auto abgesteckt nach ${connectedDuration}
    soc: # vehicle soc update event
      title: Ladestand geändert
      msg: Batterie auf ${vehicleSoc:%.0f}% geladen
    guest: # vehicle could not be identified
      title: Unbekanntes Fahrzeug
      msg: unbekanntest Fahrzeug verbunden, Gast?
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
I cannot provide that as I run evcc inside Homeassistant and I do not know how to run this command within this environment.
```

### What type of operating system are you running?

Linux

### Version

0.122.1

umrath, 2023-12-19:
Addendum: Adjusting the interval setting to 30 seconds seemingly changes the ratio at which the error messages appear.

mdkeil, 2023-12-19:
Ich glaube das Problem liegt im hoymiles-Template.. von dort kommt zumindest meiner Ansicht nach die Fehlermeldung.

```
  - type: template
    template: hoymiles-opendtu
    usage: pv
    host: 192.168.179.107
    name: Gartenkraftwerk
```

umrath, 2023-12-19:
> Ich glaube das Problem liegt im hoymiles-Template.. von dort kommt zumindest meiner Ansicht nach die Fehlermeldung.
> 
> ```
>   - type: template
>     template: hoymiles-opendtu
>     usage: pv
>     host: 192.168.179.107
>     name: Gartenkraftwerk
> ```

Huh, das hatte ich überhaupt nicht auf dem Radar, weil ich es gar nicht angefasst hatte.
Interessant!

Irgendeine Idee, was das sein könnte bzw. wie ich das fixen kann?
Ich habe die DTU auch in HA konfiguriert. Kommen die sich irgendwie ins Gehege? Allerdings wäre das neu, denn bisher war das kein Problem.

umrath, 2023-12-19:
Fehler sehr wahrscheinlich gefunden: Die DTU ist abgestürzt.

closed 2023-12-19 by umrath

mdkeil, 2023-12-19:
Das würde auch den Fehler erklären ;)

umrath, 2023-12-19:
interval wieder bei 10 Sekunden - keine Fehler mehr.

Danke, @mdkeil !