evcc-io/evcc #11148: Connection to Ostrom

umrath, 2023-12-19:
Much like Tibber, Ostrom provides hourly tariffs with variable prices.
I would very much like to see this getting integrated into evcc, if possible.

Link to the API: https://docs.ostrom-api.io/reference/introduction

andig, 2023-12-20:
Das bräuchte Api Credentials.

umrath, 2023-12-20:
Ist das ein Problem?
Als Entwickler kann man die relativ problemlos anfragen (der Ostrom-Support ist üblicherweise ziemlich responsiv).
Und als Nutzer brauche ich für andere Dienste ja teilweise auch Zugangsdaten, die ich mir auch relativ problemlos über die App holen kann.

andig, 2023-12-20:
Kein Problem- muss nur jemand machen und zur Verfügung stellen.

closed 2023-12-26 by andig

habz0c, 2024-01-29:
Hi,
ich bin neu hier.
Hoffe es entspricht der Etiquette, wenn ich schreibe, dass auch ich mich freuen würde Ostrom integriert zu wissen im evcc. 
In dem Blogbeitrag „Feature Highlights 10/2023“ hat der Michael beim Punkt -Smartes Netzladen- dazu aufgefordert ein issue zu eröffnen, sofern ein dynamischer Stromtarif noch nicht unterstützt ist, wenn dieser aber eine Schnittstelle bietet.
Wie eingangs erwähnt, ich bin neu hier und weiß nun nicht ob die von „umrath“ genannte API Doku jene Schnittstelle darstellt, welche das evcc-Team benötigt. 
Bin aufjedenfall sehr dankbar für das starke evcc-Tool!!

andig, 2024-01-29:
Danke. Am letzen Zustand bzgl. Anforderungen hat sich allerdings nichts geändert, s.o.

umrath, 2024-01-29:
Ich verstehe allerdings nicht ganz, was an den Voraussetzungen nicht erfüllt ist.
Eine offene Schnittstelle ist doch vorhanden.

andig, 2024-01-29:
https://github.com/evcc-io/evcc/issues/11148#issuecomment-1864132013

umrath, 2024-01-29:
Genau das verstehe ich nicht.
Eine Mail an partner@ostrom.de sollte doch völlig genügen.

andig, 2024-01-30:
https://github.com/evcc-io/evcc/issues/11148#issuecomment-1864375234

habz0c, 2024-01-30:
Alles klar. Habe da mal ganz freundlich eine Nachricht hinterlassen. Vielleicht melden die sich ja. 
Sobald die es tun, gebe ich hier ein Zeichen.

kallelix, 2024-02-14:
@habz0c haben sie es getan?

habz0c, 2024-02-15:
Nein

umrath, 2024-03-01:
Noch immer nicht?
Sonst würde ich mal nachhaken. Bisher war der Support immer sehr auf zack.

habz0c, 2024-03-01:
Leider noch immer keine Antwort von denen.

arboeh, 2024-03-16:
Ich bin auch sehr an einer Integration interessiert. An Mai bin ich auch Kunde von Ostrom.
Wer an einem Referral Code interessiert ist, kann mich gerne über Telegram unter @arend_boehmer kontaktieren.

kallelix, 2024-03-21:
Ich werde Anfang April, wenn mein Vertrag dann läuft, einen PR dafür machen können. Im Grunde ist das aktuell etwas sinnfrei, weil die API relativ exakt das liefert, was auch die awattar API liefert. Man könnte also auch einfach die awattar API für eine Strompreisindikation verwenden (die benötigt keinen Authentifizierung). Es sind eben einfach nur die EPEX SPOT Preise für MWh bzgl. Tageszeit gemittelt über alle Strombörsen in Europa.

Ich denke der Strommarkt muss sich aber hier weiter entwickeln und es müssen die Regionen wo der Strom erzeugt und abgenommen wird mit in den Preis einfließen. So könnten die APIs pro Stromanbieter irgendwann dann wieder Sinn machen.

Im Grunde muss hier aber evcc sagen wo die Reise hingehen soll. Es müssen in Deutschland ab 2025 alle Energieanbieter gesetzlich verpflichtend dynamische Stromtarife anbieten. Es ist unrealistisch für alle eine API zu implementieren.

habz0c, 2024-04-01:
Nabend,
Wenn ich da so lese stellt sich mir die Frage ob es möglich ist die Werte von EPEX direkt abzurufen.

kallelix, 2024-04-01:
Die APIs von netztransparenz.de könnte hier eine sinnvolle Anlaufstelle sein. Das scheint ein Projekt der Übertragungsnetzbetreiber zu sein.

arboeh, 2024-04-01:
> Nabend, Wenn ich da so lese stellt sich mir die Frage ob es möglich ist die Werte von EPEX direkt abzurufen.

Die Idee hatte ich auch schon. Es gibt für Home Assistant schon eine fertige Integration: https://github.com/mampfes/ha_epex_spot

andig, 2024-04-01:
Ich verstehe die Diskusssion nicht. Reicht Entso-E nicht?

kallelix, 2024-04-01:
Ich denke schon, dass das reicht. Leider war das für mich nicht offensichtlich und damit scheine ich ja nicht alleine zu sein.

andig, 2024-04-01:
Wir sind noch nicht gut darin, die verfügbaren Tarife zu vermarkten.

/cc @naltatis

habz0c, 2024-04-02:
Dann ist ja vorerst alles geklärt. Danke euch. :)

arboeh, 2024-04-02:
> Ich denke schon, dass das reicht. Leider war das für mich nicht offensichtlich und damit scheine ich ja nicht alleine zu sein.

Ebenso... 😂

andig, 2024-06-02:
Zwischenzeitlich können „custom“ Tarife auch selbst per Template implementiert werden.

AndyKausG, 2024-06-05:
Huhu Leute, ich bin auch Ostrom Kunde und baue eventuell selbst noch eine Integration für HASIO. Aber ich hab nun auch mal EVCC angeschaut und finde das recht nett. Vielleicht kann man ja auch hier eine Preisabfrage für Ostrom mit einbauen?
Die API ist recht einfach. Rest API mit Bearer Token und dann muss man die Preise abfragen. Dann muss man nicht zwei Werte aus dem Preisarray addieren. Und hat bereits den Endkundenpreis inkl. Netzgebühr.... hab das mit dem Ostrom Support validiert.

andig, 2024-06-05:
Ja, kann man 👍🏻

arboeh, 2024-06-05:
> Zwischenzeitlich können „custom“ Tarife auch selbst per Template implementiert werden.

Ja, das funktioniert.

Statischer Tarif (monatlich):
```
tariffs:
  currency: EUR
  grid:
    type: custom
    price:
      source: http
      uri: https://api.ostrom.de/v1/tariffs/city-id?cityId=11111&usage=1000
      jq: .ostrom[0].workingPriceBrutto
      scale: 0.01
```

Dynamischer Tarif (stündlich):
```
tariffs:
  currency: EUR
  grid:
    type: custom
    forecast:
      source: http
      uri: https://api.ostrom.de/v1/spot-prices/ostrom-product-tariff?cityId=11111&postalCode=55555
      jq: '.day.prices[ ] | {start: .date, price: .priceWithTaxesAndLevies}'
```

premultiply, 2024-06-05:
@naltatis @andig Lohnt es (schon) dafür ggf. Aufwand in Tariff-Templates zu stecken?

naltatis, 2024-06-05:
Ja, sollten wir nutzen. Die werden inzwischen ja auch automatisch in der Doku angezeigt. Das macht's für neue Nutzer viel einfacher. Und wir haben dann gleich die Grundlage für die UI.

premultiply, 2024-06-05:
Hab mal einen groben Entwurf erstellt: #14206

SerrII, 2024-06-11:
Hi,

bei mir kommt folgende Fehlermeldung 

`yaml: line 98: mapping values are not allowed in this context`

Kann mir einer helfen?

Danke im Voraus

Dici71, 2024-06-14:
> Ja, das funktioniert.
>
> Statischer Tarif (monatlich):
>
> ```
> tariffs:
>   currency: EUR
>   grid:
>     type: custom
>     price:
>       source: http
>       uri: [...]

> Ja, das funktioniert.
>
> Statischer Tarif (monatlich):
>
> ```
> tariffs:
>   currency: EUR
>   grid:
>     type: custom
>     price:
>       source: http
>       uri: [...]

Bei mir will der Server mit den Anpassungen in der Config einfach nicht mehr starten

Error during startup. Check your configuration and restart.
cannot create tariff type 'custom': jq: query failed: cannot iterate over: null

arboeh, 2024-06-14:
Es darf nur ein "type: custom" eingebunden werden. "forecast" ODER "price", also entweder "statischer Tarif" ODER "dynamischer Tarif". Die URL-Parameter "cityId" und "postalCode" müssen individuell angepasst werden.
Ich bekomme im Moment beim Abruf des dynamischen Tarifs aber auch einen Fehler:
`cannot create tariff type 'custom': jq: too many results`
Das müsste ich mir genauer angucken.

Dici71, 2024-06-14:
> Es darf nur ein "type: custom" eingebunden werden. "forecast" ODER "price", also entweder "statischer Tarif" ODER "dynamischer Tarif". Die URL-Parameter "cityId" und "postalCode" müssen individuell [...]

Vielen Dank für dein Feedback.....

ich habe im Grunde nur die 2. Konfiguration genutzt, die sich auf den dynamischen Tarif bezieht...die Fehlermeldung bezieht sich auch auf diese Zeile:
jq: '.day.prices[ ] | {start: .date, price: .priceWithTaxesAndLevies}'

P.S. was ist mit CityID gemeint ?

Gruß

arboeh, 2024-06-14:
> Vielen Dank für dein Feedback.....
>
> ich habe im Grunde nur die 2. Konfiguration genutzt, die sich auf den dynamischen Tarif bezieht...die Fehlermeldung bezieht sich auch auf diese Zeile: jq: [...]

Guck mal auf https://www.ostrom.de/ und gib deine Daten für die Berechnung deines Tarifs ein. Wenn man dann auf "Tarif berechnen" klickt, wird man auf eine Seite weitergeleitet und in der Adresszeile steht dann etwas wie:
`https://join.ostrom.de/tariff-plan?usage=1000&cityId=11111&postalCode=55555&cityName=Musterstadt&language=de_DE`
Da siehst du dann deine cityId.

Dici71, 2024-06-14:
> Guck mal auf https://www.ostrom.de/ und gib deine Daten für die Berechnung deines Tarifs ein. Wenn man dann auf "Tarif berechnen" klickt, wird man auf eine Seite weitergeleitet und in der [...]

Oh ja....da muss man erstmal drauf kommen ;-))

An der Fehlermeldung ändert sich allerdings dadurch nichts ;-(

arboeh, 2024-06-17:
> Oh ja....da muss man erstmal drauf kommen ;-))
>
> An der Fehlermeldung ändert sich allerdings dadurch nichts ;-(

I tried it in https://jqkungfu.com/ and it works. Error using this in evcc is
`jq: too many results`
Resultset is an array with 24 entries, but "too many" is relative :-)

andig, 2024-06-17:
Sounds like a peoblem with the query, not the amount of data.

arboeh, 2024-06-17:
This gives back valid data format in online jq tester, but doesn't work in evcc:
`'[.day.prices[ ] | {start: .date, price: .priceWithTaxesAndLevies}]'` -> `invalid character 'm' looking for beginning of value`

andig, 2024-06-17:
Thats a different error? What's the data? As usual, log files are king.

arboeh, 2024-06-17:
Data is:

```
{
  "day": {
    "averagePrice": 30.38,
    "prices": [
      {
        "date": "2024-06-17T00:00:00.000Z",
        "dateUTC": "2024-06-16T22:00:00.000Z",
        "price": 7.67,
        "grossKwhPrice": 9.12,
        "mWhPriceInEuros": 76.62,
        "hour": "00",
        "taxesAndLevies": 20.59,
        "priceWithTaxesAndLevies": 28.26
      },
      {
        "date": "2024-06-17T01:00:00.000Z",
        "dateUTC": "2024-06-16T23:00:00.000Z",
        "price": 6.68,
        "grossKwhPrice": 7.95,
        "mWhPriceInEuros": 66.73,
        "hour": "01",
        "taxesAndLevies": 20.41,
        "priceWithTaxesAndLevies": 27.09
      },
      {
        "date": "2024-06-17T02:00:00.000Z",
        "dateUTC": "2024-06-17T00:00:00.000Z",
        "price": 6.89,
        "grossKwhPrice": 8.19,
        "mWhPriceInEuros": 68.81,
        "hour": "02",
        "taxesAndLevies": 20.45,
        "priceWithTaxesAndLevies": 27.34
      },
      {
        "date": "2024-06-17T03:00:00.000Z",
        "dateUTC": "2024-06-17T01:00:00.000Z",
        "price": 7.19,
        "grossKwhPrice": 8.55,
        "mWhPriceInEuros": 71.81,
        "hour": "03",
        "taxesAndLevies": 20.5,
        "priceWithTaxesAndLevies": 27.69
      },
      {
        "date": "2024-06-17T04:00:00.000Z",
        "dateUTC": "2024-06-17T02:00:00.000Z",
        "price": 7.36,
        "grossKwhPrice": 8.76,
        "mWhPriceInEuros": 73.56,
        "hour": "04",
        "taxesAndLevies": 20.54,
        "priceWithTaxesAndLevies": 27.9
      },
      {
        "date": "2024-06-17T05:00:00.000Z",
        "dateUTC": "2024-06-17T03:00:00.000Z",
        "price": 9.01,
        "grossKwhPrice": 10.72,
        "mWhPriceInEuros": 90.02,
        "hour": "05",
        "taxesAndLevies": 20.85,
        "priceWithTaxesAndLevies": 29.86
      },
      {
        "date": "2024-06-17T06:00:00.000Z",
        "dateUTC": "2024-06-17T04:00:00.000Z",
[... 240 lines cut ...]
        "priceWithTaxesAndLevies": 30.36
      },
      {
        "price": 7.76,
        "month": "2023-07",
        "taxesAndLevies": 20.61,
        "priceWithTaxesAndLevies": 28.37
      },
      {
        "price": 10.63,
        "month": "2023-06",
        "taxesAndLevies": 21.16,
        "priceWithTaxesAndLevies": 31.79
      }
    ]
  }
}
```

Log:
```
[tariff] ERROR 2024/06/17 16:42:50 invalid character 'm' looking for beginning of value
[main  ] FATAL 2024/06/17 16:42:50 cannot create tariff type 'custom': invalid character 'm' looking for beginning of value
[main  ] FATAL 2024/06/17 16:42:50 will attempt restart in: 15m0s
```

andig, 2024-06-17:
Please provide full log, showing data.

arboeh, 2024-06-18:
Log file attached.
[log.txt](https://github.com/user-attachments/files/15880929/log.txt)

arboeh, 2024-06-18:
> Please provide full log, showing data.

Log file attached.
[log.txt](https://github.com/user-attachments/files/15880934/log.txt)

andig, 2024-06-18:
Can I test this in any way? Which config/credentials?

arboeh, 2024-06-26:
> Seems this needs the debugger. Can I test this in any way? Which config/credentials?

What do you need to debug it? I could send you my "cleaned" config (home assistant addon) without IPs, user names and passwords.

andig, 2024-06-30:
Alle Parameter um den Aufruf zu machen. Ich kann nicht hellsehen :)

xerion3800, 2024-07-09:
> Bei mir will der Server mit den Anpassungen in der Config einfach nicht mehr starten
>
> Error during startup. Check your configuration and restart. cannot create tariff type 'custom': jq: query [...]

Ich habe eine Frage zu den Daten. ich nutze aktuell Tibber und schaue mit gerade ostrom an. Bei Tibber habe ich ja über die API eine Vorschau bis morgen. So wie ich das in dieser Abfrage sehe ich das ja immer nur der aktuelle Tag und nicht bis morgen. Wie läuft das mit dem Ladeplanung ab wenn man nicht genügend Vorschau hat. Hast du da schon Erfahrungen? Oder bekommt man als Kunde eine andere API Schnittstelle mit mehr Informationen?

kofferjunge, 2024-07-29:
Bitte seid nachsichtig mit mir, da ich noch komplett unerfahren mit dem Thema bin.
Ich bin ebenfalls Ostrom-Kunde und habe versucht, den Strompreis in evcc anzeigen zu lassen.
Ich bekomme den gleichen Fehler, wie arboeh „jq: too many results“.
Ich habe einfach mal versucht, weiterzukommen, dabei ist mir aber aufgefallen, dass das Datumsformat in den Daten von Ostrom nicht mit der gewünschten Form von evcc übereinstimmt:

„Das Plugin muss eine JSON-Struktur zurückgeben, welches eine Liste von Zeiträumen und Preisen enthält. Die Datumsfelder müssen in der Form YYYY-MM-DDTHH:MM:SSZ und der Preis in der korrekten Währungseinheit (bspw. EUR) angegeben werden.“ (https://docs.evcc.io/docs/devices/tariffs)

WiskyViking, 2024-08-19:
für tagesaktuelle Preise (bis 0200 des nöchsten Tages) funktioniert es bei mir mit:

tariffs:
  grid:
    type: custom
    forecast:
      source: http
      uri: https://api.ostrom.de/v1/spot-prices/ostrom-product-tariff?cityId=5867&postalCode=24939
      jq: '[ .day .prices ] | transpose | map({ start: ( .[0] | .date [:19] | strptime("%Y-%m-%dT%H:%M:%S") | strftime("%Y-%m-%dT%H:%M:%SZ") ), end: ( .[0] | .date [:19] | strptime("%Y-%m-%dT%H:%M:%S") | mktime + 3600 | strftime("%Y-%m-%dT%H:%M:%SZ") ), price: ( .[0] | .priceWithTaxesAndLevies / 100 ) }) | tostring'

WiskyViking, 2024-08-19:
für den Folgetag muss man als registrierter User einen Authentikation Token erstellen. Dazu benutzt man (in der Sandbox) einen POST Aufruf über folgenden Server:

https://auth.sandbox.ostrom-api.io/oauth2/token

und erhält einen JSON im folgenden Format:

{
  "access_token": "XYZ",
  "token_type": "Bearer",
  "expires_in": 3599
}

diesen Access token benötigt man dann um die Preise abzurufen über (in der Sandbox) einen GET Aufruf über folgenden Server:

https://sandbox.ostrom-api.io/spot-prices

und erhält einen JSON im folgenden Format:

{
  "data": [
    {
      "date": "2024-08-10T00:00:00.000Z",
      "netMwhPrice": 87.28,
      "netKwhPrice": 8.73,
      "grossKwhPrice": 10.39,
      "netKwhTaxAndLevies": 19.1,
      "grossKwhTaxAndLevies": 22.73,
      "netMonthlyOstromBaseFee": 5.05,
      "grossMonthlyOstromBaseFee": 6,
      "netMonthlyGridFees": 8.6,
      "grossMonthlyGridFees": 10.24
    },

stündlich bis:

    {
      "date": "2024-08-10T21:00:00.000Z",
      "netMwhPrice": 106.64,
      "netKwhPrice": 10.67,
      "grossKwhPrice": 12.7,
      "netKwhTaxAndLevies": 19.1,
      "grossKwhTaxAndLevies": 22.73,
      "netMonthlyOstromBaseFee": 5.05,
      "grossMonthlyOstromBaseFee": 6,
      "netMonthlyGridFees": 8.6,
      "grossMonthlyGridFees": 10.24
    }
  ]
}

testen kann man das unter:

Obtain OAuth2 Access Token
https://docs.ostrom-api.io/reference/createoauth2token

und 

Retrieve day-ahead spot price information
https://docs.ostrom-api.io/reference/getspotprices

Leider bin ich noch nicht dahinter gekommen wie man das in EVCC realisieren kann.

Hat jemand eine Idee?

WiskyViking, 2024-08-26:
Hier die JQ Abfrage korrigiert für UTC Zeit:

jq: '[ .day .prices ] | transpose | map({ start: ( .[0] | .dateUTC [:19] | strptime("%Y-%m-%dT%H:%M:%S") | strftime("%Y-%m-%dT%H:%M:%SZ") ), end: ( .[0] | .dateUTC [:19] | strptime("%Y-%m-%dT%H:%M:%S") | mktime - 14400 | strftime("%Y-%m-%dT%H:%M:%SZ") ), price: ( .[0] | .priceWithTaxesAndLevies / 100 ) }) | tostring'

Da es nur die Preise des aktuellen Tages und der EVCC Planer default mäßig des Folgetag betrifft gibt wirft EVCC einen Error aus: 'planner: no matching rate'.

Schlotze83, 2024-10-22:
> "priceWithTaxesAndLevies": 25.69

Sehe ich das richtig, dass ich 25,69 cent pro kWh zahlen muss, wenn ich den dynamischen Tarif von Ostrom nutze?
Habe immer die Preise der Börse verfolgt, die ja deutlich niedriger sind.
Wenn ich das nun mit den aktuellen Preisen vergleiche, die Ostrom liefert, bin ich zu jeder Stunde über den 30 Cent, die ich zur Zeit für meinen "normalen" Tarif zahle...

nilaq, 2024-11-03:
Hi zusammen (@umrath, @habz0c, @kallelix, @arboeh, @Schlotze83, @WiskyViking, @kofferjunge, @xerion3800) , 

ich bin einer der Mitgründer von [Nomos](https://www.nomos.energy/) - wir sind ein API-first Stromanbieter und arbeiten hauptsächlich mit Installateuren von Energiesystemen zusammenarbeitet um dynamische Tarife "white-label" mit den richtigen Integrationen anzubieten. ([PV Magazin Artikel](https://www.pv-magazine.de/2024/10/15/nomos-bietet-photovoltaik-und-waermepumpen-unternehmen-loesungen-fuer-eigene-dynamische-stromtarife/)). Eine große Differenzierung bei uns sind vernünftige APIs, die Basis für mögliche Optimierungen auf Stromseite abdecken.

Jeder unserer Stromkunden hat Zugang zu einem [API Endpunkt](https://docs.nomos.energy/api-reference/endpoints/retrieve-price-time-series), der die Zeitreihen der variablen Preise (incl. EPEX; Netzentgelte und Steuern und Abgaben) übersichtlich zur Verfügung stellt. Mit der Struktur können wir auch die zeitvariablen Netzentgelte supporten, sobald diese verfügbar sind.

Falls jemand hier noch bestehende Probleme mit der Ostrom (oder einer anderen) Integration hat, könnt ihr mir sehr gerne eine kurze Email an [nils@nomos.energy](mailto:nils@nomos.energy) schicken. Da wir noch relativ jung sind als Unternehmen, bieten wir euch auch sehr gerne einen Discount für den Stromvertrag im Gegenzug für Feedback auf unsere API Integrationen an. 

Ich freu mich von euch zu hören, falls das für jemand hier relevant sein sollte.