evcc-io/evcc #11151: EVCC startet nicht, bei konfiguriertem OCPP Charger

steveh80, 2023-12-20:
### Describe the bug

Ich habe versucht mit dem Standard-Template aus der Dokumentation für einen OCPP Charger den Docker-Container hochzufahren. Leider stürzt EVCC ab und die Weboberfläche lässt sich nicht mehr aufrufen. Docker logs zeigt aber keine Fehlermeldung. Docker ps sagt container unhealth.

Andere Konfigurationen funktionieren auch der Demo Modus ohne evcc.yaml. 

Was mache ich falsch?

### Steps to reproduce

1. create evcc.yaml with default templete for evcc charger
2. docker run -d --name evcc -v ./evcc.yaml:/etc/evcc.yaml -v ./.evcc:/.evcc -p 7070:7070 -p 8887:8887 -p 7090:7090/udp -p 9522:9522/udp evcc/evcc:latest 
3. open web interface

### Configuration details

```yaml
chargers:
  - name: my_charger
    type: template
    template: ocpp
    timeout: 2m # optional
```

### Log details

```text
docker logs evcc
[main  ] INFO 2023/12/20 12:35:54 evcc 0.122.1
[main  ] INFO 2023/12/20 12:35:54 using config file: /etc/evcc.yaml
[main  ] INFO 2023/12/20 12:35:54 starting ui and api at :7070
[ocpp-1] DEBUG 2023/12/20 12:35:54 waiting for chargepoint: 5m0s
```

### What type of operating system are you running?

Docker container

### Version

latest

premultiply, 2023-12-20:
Da stürzt nix ab.
Wie die Logmeldung besagt wartet evcc auf die eingehende OCPP-Verbindung.

Erstmal ohne Docker probieren.

closed 2023-12-20 by premultiply

premultiply, 2023-12-20:
Nein, denn Ladesteuerung ohne Ladepunkt wäre auch irgendwie sinnlos, oder?

premultiply, 2023-12-20:
> [ocpp-1] DEBUG 2023/12/20 12:35:54 waiting for chargepoint: 5m0s

Erstmal immer ohne Docker probieren um die selbstgestellten Fallen mit dem Profiwerkzeug zu umschiffen.
Wenn ohne alles funktioniert kann man sich dransetzen und mit dem Profiwerkzeug langsam vorantasten.