evcc-io/evcc #11159: Ladestand wird nicht (mehr) angezeigt

711didi26, 2023-12-21:
### Describe the bug

Update auf v0.122.1 "vor Tagen/Wochen"
Seit ~ dieser Woche, wird der Ladestand nicht mehr angezeigt.
Die Volkwagen App funktioniert und zeigt den Ladestand an.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: da46f6d6724a5b0201d80888186890fb8e28a046c873bf3473bd8d636b4acfc5

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjI5ODM4OTUwODYsImlhdCI6MTY4Nzg5NTA4NiwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IjcxMWRpZGkyNiIsInNwZSI6dHJ1ZX0.xT9vXwwlcUzWQa6QInu8qoJqT-nnEorpkstICy44MyY

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.178.35
  port: 502
  name: grid1
- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.178.35
  port: 502
  name: pv2
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.178.35
  port: 502
  name: battery3

chargers:
- type: template
  template: bender
  host: 192.168.178.80
  port: 502
  name: wallbox5

vehicles:
- type: template
  template: id
  title: ID3
  user: XXXXX
  password: XXXXXXX
  capacity: 58
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Teck
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
Dec 18 22:06:10 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:10 charge power: 3462W
Dec 18 22:06:10 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:10 pv power: 0W
Dec 18 22:06:10 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:10 battery soc: 0%
Dec 18 22:06:10 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:10 battery power: 0W
Dec 18 22:06:10 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:10 grid power: 4445W
Dec 18 22:06:10 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:10 grid currents: [7.47 5.83 6.25]A
Dec 18 22:06:10 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:10 site power: 4445W
Dec 18 22:06:11 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:11 charge voltages: [232 232 233]V
Dec 18 22:06:11 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:11 detected connected phases: 3p
Dec 18 22:06:11 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:11 charge currents: [5.01 5.03 5.04]A
Dec 18 22:06:11 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:11 detected active phases: 3p
Dec 18 22:06:11 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:11 charge total import: 2764.420kWh
Dec 18 22:06:11 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:11 charger status: C
Dec 18 22:06:11 raspberry evcc[945]: [lp-1  ] ERROR 2023/12/18 22:06:11 vehicle soc: missing fuel status
Dec 18 22:06:11 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:11 pv charge current: 0A = 6A + -6.44A (4445W @ 3p)
Dec 18 22:06:20 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:20 ----
Dec 18 22:06:20 raspberry evcc[945]: [lp-1  ] DEBUG 2023/12/18 22:06:20 charge power: 3450W
Dec 18 22:06:20 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:20 pv power: 0W
Dec 18 22:06:20 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:20 battery soc: 0%
Dec 18 22:06:20 raspberry evcc[945]: [site  ] DEBUG 2023/12/18 22:06:20 battery power: 0W
```

### What type of operating system are you running?

Linux

### Version

0.122.1

StefanSchoof, 2023-12-21:
Ist im nightly gefixt: #11132

711didi26, 2023-12-21:
[StefanSchoof](https://github.com/StefanSchoof) THANKS, for this lighting fast response 
Looks like a duplicate to https://github.com/evcc-io/evcc/discussions/11131

closed 2023-12-21 by andig