evcc-io/evcc #11160: Auswertung Ladeenergie - Verbesserung Berechnung Energiepreis / Ersparnis

An-Wi, 2023-12-21:
Wunschgemäß hier das Thema nochmal neu angelegt.

Die Berechnung des Energiepreises und der Ersparnis sollte intelligenter erfolgen und aus den jeweiligen Ladevorgängen den zu Zeitpunkt gültigen Strompreis bzw. die dabei entstandene reale Ersparnis aufaddieren und daraus einen realen Energiepreis ermitteln.

**Hintergrund:** Preisänderungen beim Strom (durch Netzanbieter- oder Vertragswechsel) führen teilweise zu "negativen" Ersparnissen.
Auch dynam. Strompreise würden dadurch genauer abgebildet.

Anbei der Screenshot aus meinem alten Issue #10562 
[image]

mxwi, 2024-05-03:
Do you think that the values will be corrected after implementation, or is the information lost?

best regards

naltatis, 2024-05-03:
Since we haven't stored the reference prices from the past, we'll not be able to fill these data in retroactively. Manually adding them later (in database) will be possible.

BrickTop87, 2024-09-06:
In theory it's possible to reconstruct the old energy prices from the sessions database. But only assuming that the feed-in tariff remained constant (which should be the case for the majority of private pv systems at least in Germany).

price_per_kwh = solar_percentage * feed-in_tariff + (1 - solar_percentage) * x
--> x = (price_per_kwh - solar_percentage * feed-in_tariff) / (1 - solar_percentage)

x is the electricity tariff we're looking for.