evcc-io/evcc #11163: VW e-up: "missing fuel status"

olmo1501, 2023-12-21:
### Describe the bug

This is the same issue as  #9817 with the difference that it doesn't automagically resolve.

Some time ago I had to switch from template "vw" to template "id" which was expected after VW switched of the "old" platform. Ever since I can not manage to retrieve any information about the car.

- VW-App works perfectly fine
- Web (my Volkswagen) works perfectly fine, nothing waiting to be accepted

I even raised the timeout to 30s to no avail. "evcc vehicle" returns after about 7-8 seconds with "missing status". This makes me think that this would have to be a VW-issue. But if so why doesn't anybody else mention this? Did I miss something? Anything else I need to do after switching to the "id" template?

Thanks a lot for your phantastic work.

### Steps to reproduce

1. set template to "id" in evcc.yaml
2. run "evcc vehicle"

### Configuration details

```yaml
vehicles:
- name: e-up
  type: template
  template: id
  title: VW e-up!
  phases: 2
  user: <schnipp>
  password: <schnipp>
  vin: <schnapp>
  capacity: 32
  targetSoC: 85
  timeout: 30s
```

### Log details

```text
[main  ] INFO 2023/12/21 11:42:06 evcc 0.122.1
[main  ] INFO 2023/12/21 11:42:06 using config file: /etc/evcc.yaml
[db    ] INFO 2023/12/21 11:42:07 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/12/21 11:42:07 connecting evcc-1266624932 at tcp://192.168.241.2:1883
[mqtt  ] DEBUG 2023/12/21 11:42:07 tcp://192.168.241.2:1883 connected
.Soc:          missing fuel status
Capacity:      32.0kWh
Charge status: missing charging status
Range:         not available
Odometer:      not available
Finish time:   missing charging status
Climater:      not available
Target Soc:    missing charging status
OnIdentified:  TargetSoc:85
Features:      []
```

### What type of operating system are you running?

Linux

### Version

0.122.1

StefanSchoof, 2023-12-21:
Ist im nightly gefixt: https://github.com/evcc-io/evcc/pull/11132

closed 2023-12-21 by andig

olmo1501, 2023-12-21:
Thanks a lot to both of you @StefanSchoof @andig

bananajoe86, 2023-12-31:
someone should change the documentation. it still says "tamplate = vw" but should be "tamplate = id"