evcc-io/evcc #11165: Tibber Preisvorhersage und <= Ladepreis stimmen nicht

marcocupra, 2023-12-21:
### Describe the bug

EVCC zeigt mir für heute bei Ladestart <= 15ct lediglich eine Stunde an. Bei Tibber sind jedoch mehrere Stunden bei genau 15.0ct.
[image]
[image]

### Steps to reproduce

1. EVCC öffnen
2. Ladepreiseinstellung öffnen
3.
...

### Configuration details

```yaml
EVCC mit
TibberApi
Elli Wallbox
```

### Log details

```text
kein Log
```

### What type of operating system are you running?

Linux

### Version

0122.1

marcocupra, 2023-12-21:
[image]

marcocupra, 2023-12-21:
[image]

andig, 2023-12-21:
<=15ct ist doch korrekt bei Preis =15ct. Bitte Problem klar beschreiben.

andig, 2023-12-21:
Im Zweifel mal

    evcc tariff

aufrufen

VolkerK62, 2023-12-21:
Das ist keine Ladeplanung. Das ist "smart Laden". Da bekommt man normalerweise alle Slots angezeigt, die dem eingestellten Limit entsprechen.

andig, 2023-12-21:
Wir raten. Ohne passendes 

    evcc tariff

lässt dich das nicht beurteilen. Ich mache mal zu bis es Details gibt.

marcocupra, 2023-12-21:
Sorry, ich hatte den Bug übers Smartphone gemeldet, da war es mir zu buckelig ein Logauszug aus dem Docker zu ziehen ;-) Also mein Problem ist, dass zwischen 01.00 Uhr und 06.59 Uhr der Strom genau 15ct kostet. EVCC zeigt mir aber nur die Stunde 4 an. Das heißt das automatische Laden würde ja nur 04.00-04.59 Uhr funktionieren. Nach meiner Sicht müsste es aber zwischen 01.00-06.59 Uhr laden. Und genau 15ct entspricht dem angezeigten <=.

andig, 2023-12-21:
Das mag sein oder auch nicht- es braucht Belege dafür- vllt. schickt uns ja auch Tibber was Falsches...

marcocupra, 2023-12-21:
Ich habe doch im Protokoll die Daten von evcc tariff angehangen. Tibber scheint das Richtige zu senden. Es ist klar 0.150 zu sehen. Ob noch weitere Nachkommastellen da sind, sehe ich nicht.

Hofyyy, 2023-12-21:
Ich klemme mich hier mal rein da ich auch Tibber habe. Lasst uns das mal sortieren.
- Tibber scheint richtig zu senden
- Un dich denke EVCC hat es auch richt, das sieht man ja auch in der Grafik (die Balken zeigen den richtigen Preis)

Die Frage die du hast ist: Warum schlägt mir der Ladeplaner nur vor eine Stunde zu laden, wo ich doch 5 Stunden günstig laden könnte.

Hier wäre meine Frage: Wieviel KWh brauchte denn dein Fahrzeug noch?

Könnte ich hier richtig liegen?

marcocupra, 2023-12-21:
Ja genau, EVCC würde ja die Wallbox nur eine Stunde freischalten, obwohl die Preisgrenze 5 Stunden erreicht wird.

andig, 2023-12-21:
> Ich habe doch im Protokoll die Daten von evcc tariff angehangen.

ah, sorry. Die posthumen Updates sehe ich im Emailstream nicht. Es hat schon seinen Grund, dass wir sofort nach Logs fragen…

andig, 2023-12-21:
/cc @naltatis

andig, 2023-12-21:
> Hier wäre meine Frage: Wieviel KWh brauchte denn dein Fahrzeug noch?

Und?

VolkerK62, 2023-12-21:
@Hofyyy   Das hat doch mit Ladeplaung nichts zu tun.
Das ist `smartCostLimit` das stellt man global ein und gilt dann für alle Loadpoints.

Hofyyy, 2023-12-21:
@VolkerK62
Naja man könnte die smartCostsLimit durchaus als Ladeplanung bezeichnen. Aber ich weiss was du meinst.

@marcocupra 
Ok Die Frage ist jetzt aber wie du deinen Bedarf bei dem Gastfahrzeug definierst. Da bin ich raus, da ich die "Ladeplanung" :-) via smartCostlimit nicht kenne.

Ich denke aber das hat rein garnichts mit Tibber zu tun :-9

marcocupra, 2023-12-21:
@Hofyyy 
Ich lasse immer bis 80% laden. Tesla beendet dann den Ladevorgang. Das hat aber m.E. nichts mit dem Problem zu tun. Es müssten alle Balken grün werden die <= 15.0ct sind, egal wie voll das Auto ist.

naltatis, 2023-12-21:
@marcocupra Kannst du mal die Antwort von `http://evcc.local:7070/api/tariff/grid` hier posten. Vielleicht haben wir hier ein Rundungsproblem?

Hofyyy, 2023-12-21:
Hab gerade bei mir mal geschaut, vielleicht hilfreich zum Vergleich:
Wobei ich natürlich abhängig von der Postleitzahl auch andere Preise haben werde.

Tibber rundet bei den Werten unten in der eigenen UI immer auf eine Kommastelle und zeigt: 16.9 (durchgängig)
  {
                "start": "2023-12-22T00:00:00+01:00",
                "end": "2023-12-22T01:00:00+01:00",
                "price": 0.1691
            },
            {
                "start": "2023-12-22T01:00:00+01:00",
                "end": "2023-12-22T02:00:00+01:00",
                "price": 0.169
            },
            {
                "start": "2023-12-22T02:00:00+01:00",
                "end": "2023-12-22T03:00:00+01:00",
                "price": 0.169
            },
            {
                "start": "2023-12-22T03:00:00+01:00",
                "end": "2023-12-22T04:00:00+01:00",
                "price": 0.1689
            },
            {
                "start": "2023-12-22T04:00:00+01:00",
                "end": "2023-12-22T05:00:00+01:00",
                "price": 0.1688
            },

marcocupra, 2023-12-21:
{"result":{"rates":[{"start":"2023-12-21T00:00:00+01:00","end":"2023-12-21T01:00:00+01:00","price":0.1862},{"start":"2023-12-21T01:00:00+01:00","end":"2023-12-21T02:00:00+01:00","price":0.1801},{"start":"2023-12-21T02:00:00+01:00","end":"2023-12-21T03:00:00+01:00","price":0.1751},{"start":"2023-12-21T03:00:00+01:00","end":"2023-12-21T04:00:00+01:00","price":0.1544},{"start":"2023-12-21T04:00:00+01:00","end":"2023-12-21T05:00:00+01:00","price":0.1504},{"start":"2023-12-21T05:00:00+01:00","end":"2023-12-21T06:00:00+01:00","price":0.1507},{"start":"2023-12-21T06:00:00+01:00","end":"2023-12-21T07:00:00+01:00","price":0.1787},{"start":"2023-12-21T07:00:00+01:00","end":"2023-12-21T08:00:00+01:00","price":0.204},{"start":"2023-12-21T08:00:00+01:00","end":"2023-12-21T09:00:00+01:00","price":0.2085},{"start":"2023-12-21T09:00:00+01:00","end":"2023-12-21T10:00:00+01:00","price":0.1946},{"start":"2023-12-21T10:00:00+01:00","end":"2023-12-21T11:00:00+01:00","price":0.1911},{"start":"2023-12-21T11:00:00+01:00","end":"2023-12-21T12:00:00+01:00","price":0.1813},{"start":"2023-12-21T12:00:00+01:00","end":"2023-12-21T13:00:00+01:00","price":0.1783},{"start":"2023-12-21T13:00:00+01:00","end":"2023-12-21T14:00:00+01:00","price":0.164},{"start":"2023-12-21T14:00:00+01:00","end":"2023-12-21T15:00:00+01:00","price":0.1628},{"start":"2023-12-21T15:00:00+01:00","end":"2023-12-21T16:00:00+01:00","price":0.1703},{"start":"2023-12-21T16:00:00+01:00","end":"2023-12-21T17:00:00+01:00","price":0.1704},{"start":"2023-12-21T17:00:00+01:00","end":"2023-12-21T18:00:00+01:00","price":0.192},{"start":"2023-12-21T18:00:00+01:00","end":"2023-12-21T19:00:00+01:00","price":0.1971},{"start":"2023-12-21T19:00:00+01:00","end":"2023-12-21T20:00:00+01:00","price":0.208},{"start":"2023-12-21T20:00:00+01:00","end":"2023-12-21T21:00:00+01:00","price":0.199},{"start":"2023-12-21T21:00:00+01:00","end":"2023-12-21T22:00:00+01:00","price":0.1897},{"start":"2023-12-21T22:00:00+01:00","end":"2023-12-21T23:00:00+01:00","price":0.1845},{"start":"2023-12-21T23:00:00+01:00","end":"2023-12-22T00:00:00+01:00","price":0.1575},{"start":"2023-12-22T00:00:00+01:00","end":"2023-12-22T01:00:00+01:00","price":0.1503},{"start":"2023-12-22T01:00:00+01:00","end":"2023-12-22T02:00:00+01:00","price":0.1502},{"start":"2023-12-22T02:00:00+01:00","end":"2023-12-22T03:00:00+01:00","price":0.1502},{"start":"2023-12-22T03:00:00+01:00","end":"2023-12-22T04:00:00+01:00","price":0.1501},{"start":"2023-12-22T04:00:00+01:00","end":"2023-12-22T05:00:00+01:00","price":0.15},{"start":"2023-12-22T05:00:00+01:00","end":"2023-12-22T06:00:00+01:00","price":0.1502},{"start":"2023-12-22T06:00:00+01:00","end":"2023-12-22T07:00:00+01:00","price":0.1512},{"start":"2023-12-22T07:00:00+01:00","end":"2023-12-22T08:00:00+01:00","price":0.188},{"start":"2023-12-22T08:00:00+01:00","end":"2023-12-22T09:00:00+01:00","price":0.1976},{"start":"2023-12-22T09:00:00+01:00","end":"2023-12-22T10:00:00+01:00","price":0.1927},{"start":"2023-12-22T10:00:00+01:00","end":"2023-12-22T11:00:00+01:00","price":0.1817},{"start":"2023-12-22T11:00:00+01:00","end":"2023-12-22T12:00:00+01:00","price":0.1815},{"start":"2023-12-22T12:00:00+01:00","end":"2023-12-22T13:00:00+01:00","price":0.1821},{"start":"2023-12-22T13:00:00+01:00","end":"2023-12-22T14:00:00+01:00","price":0.1804},{"start":"2023-12-22T14:00:00+01:00","end":"2023-12-22T15:00:00+01:00","price":0.1846},{"start":"2023-12-22T15:00:00+01:00","end":"2023-12-22T16:00:00+01:00","price":0.1909},{"start":"2023-12-22T16:00:00+01:00","end":"2023-12-22T17:00:00+01:00","price":0.2007},{"start":"2023-12-22T17:00:00+01:00","end":"2023-12-22T18:00:00+01:00","price":0.2014},{"start":"2023-12-22T18:00:00+01:00","end":"2023-12-22T19:00:00+01:00","price":0.2102},{"start":"2023-12-22T19:00:00+01:00","end":"2023-12-22T20:00:00+01:00","price":0.1985},{"start":"2023-12-22T20:00:00+01:00","end":"2023-12-22T21:00:00+01:00","price":0.1796},{"start":"2023-12-22T21:00:00+01:00","end":"2023-12-22T22:00:00+01:00","price":0.1807},{"start":"2023-12-22T22:00:00+01:00","end":"2023-12-22T23:00:00+01:00","price":0.1865},{"start":"2023-12-22T23:00:00+01:00","end":"2023-12-23T00:00:00+01:00","price":0.1646}]}}

marcocupra, 2023-12-21:
Ahh, jetzt sehe ich das Problem 🤪

naltatis, 2023-12-21:
Ja, das was ich vermutet habe. Das ist ganz knapp über 15ct, in der Konsole und UI zeigen wir aber nicht mehr an.

marcocupra, 2023-12-21:
@naltatis 
Kannst Du nicht abschneiden? Das sind ja sowieso unsinnige Millicent 🤪

naltatis, 2023-12-21:
@andig wir könnten überlegen, die Werte nur mit drei Nachkommastellen Präzision zu speichern. Wir müssen immer dran denken, dass es auch andere Währungen gibt, bei denen das ggf. Probleme macht. Tibber gibts ja nicht nur in DE. Wobei Kronen eher größere Zahlen sind als kleinere.

Hofyyy, 2023-12-21:
Wenn ich das mit meinen Werten vergleiche scheint Tibber mathematisch zu runden. 0.88 -> 0.9 und 0.01 -> 0.0

andig, 2023-12-22:
> @andig wir könnten überlegen, die Werte nur mit drei Nachkommastellen Präzision zu speichern.

Das lässt sich umsetzen. Dann kommt aber der nächste Anwender und besteht auf der 15ct Schwelle. Ich fürchte, der Konflikt ist nicht auflösbar.

Da evcc wie erwartet funktioniert: Issue schließen?

mdkeil, 2023-12-22:
Wo man ggfs. allen gerecht werden kann, wenn man es bei der Auswahl der Preisgrenze bei vollen Cents belässt aber z.B. daneben ein optionales Eingabefeld einfügt, wo man dann eine Grenze mit 2Dezimalstellen eingeben kann, um die Auswahl zu überschreiben. Intern müsste dann natürlich auch mit 2 Dezimalstellen gerechnet werden.

andig, 2023-12-22:
Was wir bestimmt nicht wollen sind mehr Knöpfe und Schalter ;)

mdkeil, 2023-12-22:
> Was wir bestimmt nicht wollen sind mehr Knöpfe und Schalter ;)

Dem stimme ich zu.. es gibt doch aber bestimmt auch kombinierte Eingabe/Auswahlfelder ;)

andig, 2023-12-22:
Wir wollen auch keine komplizierteren Eingabefelder ;)

naltatis, 2023-12-22:
Ja, das sehe ich auch so. Ich würd das nicht konfigurierbar machen oder generell mehr Stellen anzeigen. Wenns für alle OK ist würd ich auch vorschlagen das hier erstmal zu schließen. Technisch funktioniert alles korrekt. Wenn noch mal wieder zu Verwirrung kommt müssen wir ggf. noch mal an das Thema Rundung ran.

closed 2023-12-22 by andig