evcc-io/evcc #11168: OpenWB: SoC-Abfrage Cupra Born schlägt fehl

IAmKonni, 2023-12-21:
### Describe the bug

Seit dem 19.12. gegen 23 Uhr klappt die SoC-Abfrage meines Cupra Borns über die OpenWB nicht mehr. Mein Skoda Enyaq hingegen klappt noch einwandfrei.

Ich hatte vor einigen Monaten schonmal den gleichen Fehler, damals war es ein Server-Problem bei evcc.

Die Log-Meldung besagt:

`Lp2: Error from EVCC: 2023/12/21 23:02:39 rpc error: code = Unknown desc = Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/f9ae0043-fb86-4fb6-8709-8f28ba69ac15/vehicles/VSSZZZK1ZNP007453/mycar": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving (LV0)`

OpenWB ist die letzte 1.9er Version.

Ist schon was bekannt bzw. noch jemand betroffen?

### Steps to reproduce

SoC über OpenWB für einen Cupra Born abfragen

### Configuration details

```yaml
openWB Version: 1.9.303.0
```

### Log details

```text
Lp2: Error from EVCC: 2023/12/21 23:02:39 rpc error: code = Unknown desc = Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/f9ae0043-fb86-4fb6-8709-8f28ba69ac15/vehicles/VSSZZZK1ZNP007453/mycar": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving (LV0)
```

### What type of operating system are you running?

Linux

andig, 2023-12-22:
Depends on https://community.fly.io/t/dns-resolution-fails-inside-app-part-3/17386/1

mairj2345, 2023-12-25:
Leider besteht das Problem immer noch, hab seit Tagen die gleichen Probleme und Fehlermeldungen.

andig, 2023-12-25:
Ja, deshalb ist ie Fehlermeldung auch noch offen.

gz2k2, 2023-12-25:
Habe bei meinem Cupra Born seit ein paar Tagen folgendes Problem. Hat das evtl auch damit zu tun?

[https://github.com/evcc-io/evcc/discussions/11239](url)

andig, 2023-12-25:
Wenn du nicht diese Fehlermeldung hast: nein.

IAmKonni, 2023-12-25:
Mittlerweile habe ich nicht mehr den gleichen Fehler, jetzt kommt:

`Error from EVCC: 2023/12/25 23:36:49 rpc error: code = Unknown desc = cannot create vehicle 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v1/users/f9ae0043-fb86-4fb6-8709-8f28ba69ac15/garage/vehicles": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com: i/o timeout (LV0)`

martin-zh, 2023-12-27:
Kann ich nicht bestätigen. Bei mir immer noch:
2023-12-27 08:14:05: PID: 12659: Lp2: Error from EVCC: 2023/12/27 08:14:05 rpc error: code = Unknown desc = cannot create vehicle 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v1/usersxxxx/garage/vehicles": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving (LV0)

andig, 2023-12-27:
Sollte jetzt wieder gehen?

IAmKonni, 2023-12-27:
> Sollte jetzt wieder gehen?

Danke, läuft wieder! 👍

IAmKonni, 2023-12-27:
Damit geschossen. 😉

closed 2023-12-27 by IAmKonni