evcc-io/evcc #11174: Crash nach Weihnachtsupdate

AniGerm, 2023-12-22:
### Describe the bug

Hallo,
meine eMH3 ist ja schon immer etwas „bockig“. 
Heute habe ich das Weihnachtsupdate installiert und jetzt startet evcc nicht mehr. 
Anbei meine Konfiguration. 
Hat jemand eine Idee? Bzw. gibt es die Möglichkeit eines Downgrades?
Vielen Dank und frohe Weihnachten;)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: trace  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 5s # control cycle interval  
  
sponsortoken: *****  
  
# sponsors can set telemetry: true to enable anonymous data aggregation  
# see https://github.com/evcc-io/evcc/discussions/4554  
telemetry: true  
  
meters:  
- type: template  
  template: huawei-dongle-powersensor   
  id: 1    
  host: 192.168.1.139    
  port: 502    
  usage: grid    
  modbus: tcpip    
  name: grid1  
- type: template  
  template: huawei-dongle-powersensor   
  id: 1    
  host: 192.168.1.139    
  port: 502    
  usage: pv    
  modbus: tcpip    
  name: pv2  
- type: template  
  template: huawei-dongle-powersensor   
  id: 1    
  host: 192.168.1.139    
  port: 502    
  usage: battery    
  modbus: tcpip    
  name: battery3  
  
chargers:  
- type: template  
  template: abl   
  id: 1    
  host: 192.168.1.194    
  port: 502    
  modbus: rs485tcpip    
  name: wallbox6  
  
vehicles:  
- type: template  
  template: etron   
  title: E-Tron 55    
  user: *****  
  password: *****  
  capacity: 86.5    
  name: ev4  
- type: template  
  template: bmw   
  title: iX1    
  user: *****  
  password: *****  
  vin: *****  
  capacity: 66.5    
  name: ev5  
  
loadpoints:  
- title: Garten  
  charger: wallbox6  
  mode: pv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 32  
  resetOnDisconnect: true  
  
site:  
  title: Südweg  
  meters:  
    grid: grid1  
    pv:  
    - pv2  
    battery:  
    - battery3  
  residualPower: 100  
  bufferSoc: 80  
  prioritySoc: 70  
  
tariffs:  
  currency: EUR # (default EUR)  
  grid:  
    # static grid price  
    type: fixed  
    price: 0.32 # [currency]/kWh  
  
  feedin:  
    # rate for feeding excess (pv) energy to the grid  
    type: fixed  
    price: 0.08 # [currency]/kWh
```

### Log details

```text
[main  ] INFO 2023/12/22 14:48:23 evcc 0.123.0
[main  ] INFO 2023/12/22 14:48:23 using config file: /etc/evcc.yaml
[main  ] INFO 2023/12/22 14:48:23 starting ui and api at :7070
[db    ] INFO 2023/12/22 14:48:23 using sqlite database: /home/max/.evcc/evcc.db
[main  ] ERROR 2023/12/22 14:48:35 creating vehicle ev5 failed: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot get vehicles: Get "https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles?apptimezone=120&appDateTime=1703252910350": dial tcp: lookup cocoapi.bmwgroup.com: i/o timeout
[main  ] FATAL 2023/12/22 14:48:35 failed configuring loadpoints: failed configuring loadpoint: 1 error(s) decoding:

* '' has invalid keys: resetondisconnect
[main  ] FATAL 2023/12/22 14:48:35 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.123

andig, 2023-12-22:
Da chrashed nichts. Bitte Release Notes lesen.

closed 2023-12-22 by andig

AniGerm, 2023-12-22:
Ich dachte das würde alles noch so funktionieren. Kann ich notfalls erstmal downgraden?
Danke!