evcc-io/evcc #11179: Vehicle Settings Menü im UI

VolkerK62, 2023-12-22:
> Ich verstehe auch den Begriff "Ankunft" des Konfigurationreiters nicht im Zusammenhang mit der Konfiguration für "Min Ladelimit" und "Standard Ladelimit".

Das sind Werte, die beim Verbinden des Autos (Ankunft) eine Rolle spielen. Den Begriff Session würde ich ungern zusätzlich noch einführen. Lieber den Beschreibungstext bei den Advanced Funktionen (Fahrzeug-Settings / Ankunft) klarer bzw. ausführlicher machen.

_Originally posted by @naltatis in https://github.com/evcc-io/evcc/discussions/10978#discussioncomment-7928610_

Ich finde es "unglücklich" gewählt, dass sich die Fahrzeugeinstellungen im Loadpoint hinter "Plan" verstecken.
Schöner wäre dafür ein eigenes Fenster (analog Loadpoint Settings). Oder alternativ (später mal) in der Fahrzeug-Konfiguration.

naltatis, 2023-12-22:
> Schöner wäre dafür ein eigenes Fenster (analog Loadpoint Settings). Oder alternativ (später mal) in der Fahrzeug-Konfiguration.

Ja, das sehe ich genau so. Den jetzigen Stand würd ich als Zwischenschritt dorthin sehen.

naltatis, 2023-12-24:
Ich mach den hier mal zu. Die Menüstruktur (auch Grid, Battery) wird sich über die nächstesn Releases mit neuen Funktionen sicher noch ändern. Mit dem jetzigen Stand (MinSoc, Limit) ein separates Menü anzubieten, finde ich noch nicht sinnvoll.

closed 2023-12-24 by naltatis

VolkerK62, 2023-12-24:
check