evcc-io/evcc #11183: Hoymiles DTU-Pro-S - Request canceled

Praktifix2, 2023-12-22:
### Describe the bug

Hello, 

I'm trying to display the power of a Hoymiles HMS-2000-4T micoinverter in addition to my already existing SENEC-power-plant, in evcc. 

The microinverter is linked to a Hoymiles DTU-Pro-S WiFi. The DTU-Pro is linked to my home network via ethernet (IP 192.168.178.100).

I added the DTU-Pro as "gewaechshaus" with the template "hoymiles-opendtu" into my evcc.yaml. In the userinterface I see the following error:

pv 2 power: \"http://192.168.178.100:502/api/livedata/sstatus\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

Evcc is running on MacOSX using homebrew.

Do you have any suggestions, why evcc can not read the DTU-Pro's data? Does somebody else have this problem?
Thanks in advance.

### Steps to reproduce

See description above.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: ***

sponsortoken: ***

interval: 5s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.39 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

meters:
- type: template
  template: senec-home
  usage: grid  
  host: 192.168.178.2
  name: grid1
- type: template
  template: senec-home
  usage: pv  
  host: 192.168.178.2
  name: pv2
- type: template
  template: senec-home
  usage: battery  
  host: 192.168.178.2
  name: battery3
 

- name: gewaechshaus
  type: template
  template: hoymiles-opendtu
  usage: pv
  host: 192.168.178.100:502 # IP-Adresse oder Hostname

chargers:
- name: wallbox5
  type: template
  template: ellipro
  ski: ***
  ip: 192.168.178.4
 
- name: wallbox1
  type: template
  template: tapo
  host: 192.168.178.10
  user: ***
  password: ***
  standbypower: 1
  integrateddevice: true  
  icon: bike
 
- name: wallbox2
  type: template
  template: tapo
  host: 192.168.178.9
  user: ***
  password: ***
  standbypower: 1
  integrateddevice: true  
  icon: cooler
   
vehicles:
- type: template
  template: tesla
  title: Tesla Model Y  
  accessToken: ***
  refreshToken: ***
  vin: ***
  capacity: 79  
  name: ev4

loadpoints:
- title: Auto (dreiphasig)
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 5
  maxcurrent: 16
  resetOnDisconnect: true
  priority: 5
  guardduration: 10s
  enable:
      delay: 10s
  disable:
      delay: 10s
 
#- title: Klimaanlage
#  charger: wallbox2
#  mode: off
#  phases: 1
#  mincurrent: 3
#  maxcurrent: 16
#  resetOnDisconnect: true
#  priority: 4
#  guardduration: 10s
#  enable:
#      delay: 10s
#  disable:
#      delay: 10s
 
- title: E-Bike
  charger: wallbox1
  mode: off
  phases: 1
  mincurrent: 0.5
  maxcurrent: 16
  resetOnDisconnect: true
  guardduration: 10s
  enable:
      delay: 10s
  disable:
      delay: 10s

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - gewaechshaus
    battery:
    - battery3
  residualPower: 100
  bufferSoc: 100
  prioritySoc: 80

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
         ***
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          ***
          -----END CERTIFICATE-----
```

### Log details

```text
[site  ] ERROR 2023/12/22 17:13:22 pv 2 power: Get "http://192.168.178.100:502/api/livedata/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.122.0

0x3d13f, 2023-12-22:
Are you sure that the API of the DTU is listening on port 502? This ist usually the modbus port, not http...

andig, 2023-12-22:
pv2 ist ein senec-home. Was hat das mit Hoymiles zu tun?