evcc-io/evcc #11184: Since update to 0.123.0 can't set "Gastfahrzeug" over MQTT

vistalba, 2023-12-22:
### Describe the bug

I use EVCC in combination with Homeassistant. So EVCC is connected to a MQTT broker. 

I use  topic `evcc/loadpoints/1/vehicle/set` to set which vehicle is connected. 
Before 0.123.0 I had to send:

```
0 = Gastfahrzeug
1 = VehicleName1
2 = VehicleName2
```

So far, so good. With 0.123.0 now I've to send not the ID. I must now send the `vehicleName`. Okay so far. 

I can now send `vehicleName1` and `vehicleName2`. But I can't set it back to "Gastfahrzeug". Nothing works. Tested with "0", "Gastfahrzeug", "Guest", "Gast", "unknown", "Default" and "NULL".

### Steps to reproduce

As described. Switch between vehicles over MQTT set and try to set "Gastfahrzeug".

### Configuration details

```yaml
nothing special. Two Cars configured.
```

### Log details

```text
[mqtt  ] ERROR 2023/12/22 17:43:33 set evcc/loadpoints/1/vehicle/set: not found: Gastfahrzeug
[mqtt  ] ERROR 2023/12/22 17:43:34 set evcc/loadpoints/1/vehicle/set: not found: Guest
[mqtt  ] ERROR 2023/12/22 17:43:35 set evcc/loadpoints/1/vehicle/set: not found: Gast
[mqtt  ] ERROR 2023/12/22 17:43:35 set evcc/loadpoints/1/vehicle/set: not found: unknown
[mqtt  ] ERROR 2023/12/22 17:43:43 set evcc/loadpoints/1/vehicle/set: not found: Default
[mqtt  ] ERROR 2023/12/22 17:43:48 set evcc/loadpoints/1/vehicle/set: not found: NULL
[lp-1  ] INFO 2023/12/22 17:43:52 vehicle updated: vehicleName2 -> vehicleName1
[lp-1  ] INFO 2023/12/22 17:45:52 vehicle updated: vehicleName1 -> unknown
[lp-1  ] INFO 2023/12/22 17:46:30 vehicle updated: unknown -> vehicleName2
[lp-1  ] INFO 2023/12/22 17:46:44 vehicle updated: vehicleName2 -> unknown
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.0

andig, 2023-12-22:
Did you try empty name?

```go
if payload == "" {
	lp.SetVehicle(nil)
	return nil
}
```

vistalba, 2023-12-22:
Oh, yes. Also tried empty, "" and NIL with MQTT Explorer. 
I'm unsure if it sends an empty set at all. As I do not see anything in the log.
In Homeassistant I tried seting nothing, "" and NIL. 

[image]

The Homeassistant MQTT config now looks like this. It was like this as before I used it to translate 0, 1, 2 to the names.

```yaml
  - command_topic: "evcc/loadpoints/1/vehicle/set"
    name: evcc_vehicle
    unique_id: uniqueid__evcc_vehicle_select
    icon: mdi:car
    availability_topic: "evcc/status"
    state_topic: "evcc/loadpoints/1/vehicleName"
    value_template: >
        {% if value == '' %}
          Kein Fahrzeug
        {% elif value == 'vehicleName1' %}
          vehicleName1
        {% elif value == 'vehicleName2' %}
          vehicleName2
        {% else %}
          Kein Fahrzeug
        {% endif %}
    command_template: >
        {% if value == 'Kein Fahrzeug' %}
          0
        {% elif value == 'vehicleName1' %}
          vehicleName1
        {% elif value == 'vehicleName2' %}
          vehicleName2
        {% else %}
          0
        {% endif %}
    options:
      - "Kein Fahrzeug"
      - "vehicleName1"
      - "vehicleName2"
```

0x3d13f, 2023-12-22:
I tried to send an empty message with the mosquitto_pub option "-n", that also did not do the trick. This does not even trigger an error in the EVCC GUI (other values for the vehicle title do).

vistalba, 2023-12-22:
> I tried to send an empty message with the mosquitto_pub option "-n", that also did not do the trick. This does not even trigger an error in the EVCC GUI (other values for the vehicle title do).

Exactly what I also tried now. Doesn't work for me too. :(
EVCC doesn't log any error it just does nothing. Otherwiese when sending invalid options it shows in the logs.

andig, 2023-12-22:
Here's what's happening:

```go
// ListenSetter creates a /set listener that resets the payload after handling
func (m *Client) ListenSetter(topic string, callback func(string) error) error {
	topic += "/set"
	err := m.Listen(topic, func(payload string) {
		if err := callback(payload); err != nil {
			m.log.ERROR.Printf("set %s: %v", topic, err)
		}
		if err := m.Publish(topic, true, ""); err != nil {
			m.log.ERROR.Printf("clear: %s: %v", topic, err)
		}
	})
	return err
}
```

and

```go
// listen attaches listener to topic
func (m *Client) listen(topic string) paho.Token {
	token := m.Client.Subscribe(topic, m.Qos, func(c paho.Client, msg paho.Message) {
		payload := string(msg.Payload())
		m.log.TRACE.Printf("recv %s: '%v'", topic, payload)
		if len(payload) > 0 {
			m.mux.Lock()
			callbacks := m.listener[topic]
			m.mux.Unlock()

			for _, cb := range callbacks {
				cb(payload)
			}
		}
	})
	return token
}
```

Problem is that the listener never calls the callback when msg is empty to avoid double-roundtrip and clearing the set value. It is logged though:

    recv evcc/loadpoints/1/vehicle/set: ''

I'm not sure what's the best way to allow empty messages for setters.

andig, 2023-12-22:
Um es nicht zu kompliziert zu machen: wie wäre es mit `-` als Synonym für "löschen"?

closed 2023-12-22 by andig

emcepe, 2023-12-23:
Ich habe gerade ein Dejavue:

https://github.com/evcc-io/evcc/issues/4848#issuecomment-1280428142

vistalba, 2023-12-24:
Works for me with 0.123.1. Thanks for resolving @andig.

If anyone else needs the HA MQTT template:

```yaml
  - command_topic: "evcc/loadpoints/1/vehicle/set"
    name: evcc_vehicle
    unique_id: uniqueid__evcc_vehicle_select
    icon: mdi:car
    availability_topic: "evcc/status"
    state_topic: "evcc/loadpoints/1/vehicleName"
    value_template: >
        {% if value == '' %}
          Kein Fahrzeug
        {% elif value == 'vehicleName1' %}
          vehicleName1
        {% elif value == 'vehicleName2' %}
          vehicleName2
        {% else %}
          Kein Fahrzeug
        {% endif %}
    command_template: >
        {% if value == 'Kein Fahrzeug' %}
          -
        {% elif value == 'vehicleName1' %}
          vehicleName1
        {% elif value == 'vehicleName2' %}
          vehicleName2
        {% else %}
          -
        {% endif %}
    options:
      - "Kein Fahrzeug"
      - "vehicleName1"
      - "vehicleName2"
```