evcc-io/evcc #11189: Diverse Probleme seit 0.123.0

mhibu, 2023-12-22:
### Describe the bug

Seit 22.12.2023 funktioniert auch schon vor dem Update auf 0.123.0 folgendes nicht mehr:

Fahrzeugabfrage zu Kia Niro EV,
Fahrzeugabfrage zu Seat Mii,
Preisabfrage von Tibber,
Verbindung zu Go-e Charger V3 im lokalen Netz

Die config habe ich bereits um die mir bekannten breaking Changes geändert.
Haben andere ähnliche Probleme?

### Steps to reproduce

1. Restart EVCC

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: SungrowWR.fritz.box  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: SungrowWR.fritz.box  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: SungrowWR.fritz.box  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: wallbe 
  host: Wallbe.fritz.box  
  port: 502  
  name: wallbox6
- name: wallbox7
  type: template
  template: go-e-v3
  host: go-echarger-059095.fritz.box

vehicles:
- name: ev4
[... 207 lines cut ...]
[main  ] ERROR 2023/12/22 21:37:21 version check failed: Get "https://api.github.com/repos/evcc-io/evcc/releases/latest": dial tcp: lookup api.github.com: i/o timeout (installed: 0.123.0)
[lp-2  ] ERROR 2023/12/22 21:37:31 charge meter: Get "http://go-echarger-059095.fritz.box/status": dial tcp: lookup go-echarger-059095.fritz.box: i/o timeout
[site  ] DEBUG 2023/12/22 21:37:31 pv power: 0W
[site  ] DEBUG 2023/12/22 21:37:31 battery soc: 47%
[site  ] DEBUG 2023/12/22 21:37:31 battery power: 0W
[site  ] DEBUG 2023/12/22 21:37:31 grid power: 466W
[site  ] DEBUG 2023/12/22 21:37:31 site power: 766W
[lp-1  ] DEBUG 2023/12/22 21:37:31 charger status: B
[lp-1  ] INFO 2023/12/22 21:37:31 car connected
[lp-1  ] DEBUG 2023/12/22 21:37:31 pv timer elapse
[lp-1  ] DEBUG 2023/12/22 21:37:31 pv timer inactive
[lp-1  ] DEBUG 2023/12/22 21:37:31 charger: guard elapse
[lp-1  ] DEBUG 2023/12/22 21:37:31 guard timer inactive
[lp-1  ] DEBUG 2023/12/22 21:37:31 pv charge current: 0A = 0A + -1.11A (766W @ 3p)
[lp-1  ] DEBUG 2023/12/22 21:37:31 pv enable timer reset
[lp-1  ] DEBUG 2023/12/22 21:37:31 pv timer inactive
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.0

StefanSchoof, 2023-12-22:
Du hast viele Meldungen in der Art `dial tcp: lookup api.tibber.com`. Wahrscheinlich klappt gerade etwas bei deinem DNS nicht.

mhibu, 2023-12-22:
Vielen Dank, FritzBox Neustart hat geholfen! :-)

closed 2023-12-22 by andig