evcc-io/evcc #11191: Unexpected reboor/crash of ecvv since 0.123.0

vistalba, 2023-12-22:
### Describe the bug

Laden mit Ladeplan hat gerade zwei Mal evcc zum neustarten gebracht. Keine Logs vom Crash verfügbar.
Ich sehe nur, dass es neu gestartet hat.

### Steps to reproduce

-

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.0

vistalba, 2023-12-25:
@andig Kannst du diesen Bug bitte wieder öffnen. Ich bin daran, entsprechende Logs zu generieren. Danke

vistalba, 2023-12-25:
> @andig Kannst du diesen Bug bitte wieder öffnen. Ich bin daran, entsprechende Logs zu generieren. Danke

Vielen Dank für deinen Support. Leider sind auch im Debug Log keine Infos ersichtlich.

Das Log Ende mit dem crash vom evcc Conteiner mit folgenden Log-Zeilen:

```text
[site  ] DEBUG 2023/12/25 13:35:50 ----
[lp-1  ] DEBUG 2023/12/25 13:35:50 charge power: 10309W
[site  ] DEBUG 2023/12/25 13:35:50 pv power: 12435W
[site  ] DEBUG 2023/12/25 13:35:50 grid power: -182W
[site  ] DEBUG 2023/12/25 13:35:50 grid currents: [-3.07 -2.86 -1.36]A
[site  ] DEBUG 2023/12/25 13:35:50 site power: -32W
[lp-1  ] DEBUG 2023/12/25 13:35:50 charge currents: [14.6 14.5 14.7]A
[lp-1  ] DEBUG 2023/12/25 13:35:50 detected active phases: 3p
[lp-1  ] DEBUG 2023/12/25 13:35:50 charge total import: 979.171kWh
[lp-1  ] DEBUG 2023/12/25 13:35:50 charger status: C
[lp-1  ] DEBUG 2023/12/25 13:35:50 soc estimated: 39.43% (vehicle: 34.00%)
[lp-1  ] DEBUG 2023/12/25 13:35:50 vehicle soc: 39%
[lp-1  ] DEBUG 2023/12/25 13:35:50 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/12/25 13:35:50 vehicle range: 119km
[lp-1  ] DEBUG 2023/12/25 13:35:50 set Born plan soc: 0 @ 0001-01-01 00:34:08 +0034 LMT
[lp-1  ] DEBUG 2023/12/25 13:35:50 pv charge current: 15A = 15A + 0.0464A (-32W @ 3p)
[site  ] DEBUG 2023/12/25 13:35:55 ----
[lp-1  ] DEBUG 2023/12/25 13:35:55 charge power: 10322W
[site  ] DEBUG 2023/12/25 13:35:55 pv power: 12435W
[site  ] DEBUG 2023/12/25 13:35:55 grid power: -182W
[site  ] DEBUG 2023/12/25 13:35:55 grid currents: [-3.07 -2.86 -1.36]A
[site  ] DEBUG 2023/12/25 13:35:55 site power: -32W
[lp-1  ] DEBUG 2023/12/25 13:35:55 charge currents: [14.6 14.6 14.7]A
[lp-1  ] DEBUG 2023/12/25 13:35:55 detected active phases: 3p
[lp-1  ] DEBUG 2023/12/25 13:35:55 charge total import: 979.186kWh
[lp-1  ] DEBUG 2023/12/25 13:35:55 charger status: C
[lp-1  ] DEBUG 2023/12/25 13:35:55 soc estimated: 39.46% (vehicle: 34.00%)
[lp-1  ] DEBUG 2023/12/25 13:35:55 vehicle soc: 39%
[lp-1  ] DEBUG 2023/12/25 13:35:55 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/12/25 13:39:10 charge total import: 979.741kWh

```

Kein "Stop Charging" oder ähnlich. Der Go-E Charger V3 hat auch nie aufgehört zu laden. Die Ladesession beim Go-E ist durchgängig. Von 13:35 bis 13:39:10 gab es einfach keine Logs mehr. Vorher war es alle 5 Sekunden, gemäss konfiguriertem Interval. 

Folgendes ist bis jetzt klar:
- nur Cupra Born betroffen (Mercedes hat ohne Probleme stundenlang geladen).
- Das Problem tritt unabhängig vom gewählten Modus auf. Getestet mit PV/PV+Min/Schnell. 
- An der Wallbox wird die Ladung nicht unterbrochen. Was dafür spricht, dass EVCC nicht aktiv "regelt" sondern crashed. Nach dem neustart bemerkt, dass ein Ladevorgang am laufen ist und weiter regelt. Bis es wieder crashed.
- Die Zeit bis das Problem auftritt ist unterschiedlich. Teilweise läd er 4.1kWh, das andere Mal nur 0.7kWh bis er crashed. Komisch ist auch das Datum der Ladesession im UI:

[image]
Ausnahmslos alle diese Sessions haben "Enddatum" am Montag, 1 Januar um 00:34. Wohl auch der Grund, warum er keine weiteren Infos wie "Ladegeschwindigkeit" etc. berechnen kann.

Hast du mir tipps, wie ich hier zu besseren Logs kommen könnte?

thierolm, 2023-12-25:
Ich hatte mit der 0.123.1 vorgestern das Gleiche verhalten. Im Log kein Fehlerhinweis, einfach zu einer Uhrzeit ist das Log stehen geblieben. Die UI war aber weiter aktiv. Das Auto hat weiter geladen aber Vehicle SOC wurde nicht mehr aktualisiert, so dass er nicht bei 50% Min-Soc aufgehört hat zu laden. Ich beobachte das die kommenden Tage mal und versuche Infos beizusteuern, falls es wieder passiert. Hatte zunächst einen Hickup in meinem Netzwerk in Verdacht.

vistalba, 2023-12-25:
Ein Downgrade auf 0.122.1 und `resetOnDisconnect: true` zurück in die Konfig und das Problem ist wieder weg. Es tritt also definitiv nur mit 0.123.0 und 0.123.1 auf.

andig, 2023-12-25:
Aus https://github.com/evcc-io/evcc/discussions/11237#discussioncomment-7944334:

```
2 @ 0x50aa8 0x189c4 0x18594 0x1af9a78 0x86344
#	0x1af9a77	github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe+0x47	github.com/evcc-io/evcc/util/pipe/limiter.go:116

1 @ 0x50aa8 0x17c14 0x17838 0x1ad5330 0x1ad52e1 0x1ad52cd 0x1ad0008 0x1ad0ea4 0x1ae101c 0x1ae1cbc 0x1b3b63c 0x86344
#	0x1ad532f	github.com/evcc-io/evcc/core.(*Loadpoint).publish+0xcf			github.com/evcc-io/evcc/core/loadpoint.go:388
#	0x1ad52e0	github.com/evcc-io/evcc/core.(*Loadpoint).setRemainingEnergy+0x80	github.com/evcc-io/evcc/core/loadpoint_api.go:415
#	0x1ad52cc	github.com/evcc-io/evcc/core.(*Loadpoint).SetRemainingEnergy+0x6c	github.com/evcc-io/evcc/core/loadpoint_api.go:408
#	0x1ad0007	github.com/evcc-io/evcc/core.(*Loadpoint).publishSocAndRange+0x5a7	github.com/evcc-io/evcc/core/loadpoint.go:1424
#	0x1ad0ea3	github.com/evcc-io/evcc/core.(*Loadpoint).Update+0x3d3			github.com/evcc-io/evcc/core/loadpoint.go:1539
#	0x1ae101b	github.com/evcc-io/evcc/core.(*Site).update+0x42b			github.com/evcc-io/evcc/core/site.go:788
#	0x1ae1cbb	github.com/evcc-io/evcc/core.(*Site).Run+0x23b				github.com/evcc-io/evcc/core/site.go:898
#	0x1b3b63b	github.com/evcc-io/evcc/cmd.runRoot.func4+0x2b				github.com/evcc-io/evcc/cmd/root.go:281

1 @ 0x50aa8 0x17c14 0x17838 0x1ae19c4 0x86344
#	0x1ae19c3	github.com/evcc-io/evcc/core.(*Site).Prepare.func1+0x113	github.com/evcc-io/evcc/core/site.go:859

1 @ 0x50aa8 0x17c14 0x17838 0x1ae1a3c 0x86344
#	0x1ae1a3b	github.com/evcc-io/evcc/core.(*Site).loopLoadpoints+0x5b	github.com/evcc-io/evcc/core/site.go:875

1 @ 0x50aa8 0x17c14 0x17838 0x1af98c8 0x86344
#	0x1af98c7	github.com/evcc-io/evcc/util/pipe.(*Deduplicator).pipe+0x237	github.com/evcc-io/evcc/util/pipe/limiter.go:55

1 @ 0x50aa8 0x17c14 0x17838 0x1af9aec 0x86344
#	0x1af9aeb	github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe+0xbb	github.com/evcc-io/evcc/util/pipe/limiter.go:121

1 @ 0x50aa8 0x17c14 0x17838 0x4b90b0 0x86344
#	0x4b90af	github.com/evcc-io/evcc/util.(*Tee).Run+0x1ef	github.com/evcc-io/evcc/util/tee.go:46

1 @ 0x50aa8 0x189c4 0x18574 0x1af8fac 0x86344
#	0x1af8fab	github.com/evcc-io/evcc/server/updater.(*watch).watchReleases+0x4b	github.com/evcc-io/evcc/server/updater/watch.go:31

1 @ 0x50aa8 0x189c4 0x18574 0x1b3b7ec 0x86344
#	0x1b3b7eb	github.com/evcc-io/evcc/cmd.runRoot.func2+0x2b	github.com/evcc-io/evcc/cmd/root.go:235

1 @ 0x50aa8 0x189c4 0x18574 0x1b3b900 0x86344
#	0x1b3b8ff	github.com/evcc-io/evcc/cmd.runRoot.func1+0x8f	github.com/evcc-io/evcc/cmd/root.go:229

1 @ 0x50aa8 0x189c4 0x18594 0x19a8fb0 0x86344
#	0x19a8faf	github.com/evcc-io/evcc/server.(*Influx).Run.func1+0x4f	github.com/evcc-io/evcc/server/influxdb.go:144

1 @ 0x50aa8 0x189c4 0x18594 0x1aa571c 0x86344
#	0x1aa571b	github.com/evcc-io/evcc/push.(*Hub).Run+0x5b	github.com/evcc-io/evcc/push/hub.go:78

1 @ 0x50aa8 0x189c4 0x18594 0x1af8eac 0x86344
#	0x1af8eab	github.com/evcc-io/evcc/server/updater.Run+0x1ab	github.com/evcc-io/evcc/server/updater/run.go:22

1 @ 0x50aa8 0x189c4 0x18594 0x1b3e124 0x86344
#	0x1b3e123	github.com/evcc-io/evcc/cmd.configureDatabase.func2+0x63	github.com/evcc-io/evcc/cmd/setup.go:466

1 @ 0x50aa8 0x189c4 0x18594 0x4b4bac 0x86344
#	0x4b4bab	github.com/evcc-io/evcc/util.(*Cache).Run+0x5b	github.com/evcc-io/evcc/util/cache.go:35

1 @ 0x50aa8 0x61e18 0x19968c0 0x86344
#	0x19968bf	github.com/influxdata/influxdb-client-go/v2/api.(*WriteAPIImpl).bufferProc+0xef	github.com/influxdata/influxdb-client-go/v2@v2.13.0/api/write.go:147

1 @ 0x50aa8 0x61e18 0x1996bec 0x86344
#	0x1996beb	github.com/influxdata/influxdb-client-go/v2/api.(*WriteAPIImpl).writeProc+0xab	github.com/influxdata/influxdb-client-go/v2@v2.13.0/api/write.go:190

1 @ 0x50aa8 0x61e18 0x19ad024 0x86344
#	0x19ad023	github.com/evcc-io/evcc/server.(*SocketHub).Run+0x83	github.com/evcc-io/evcc/server/socket.go:159

1 @ 0x50aa8 0x61e18 0x31f3cc 0x86344
#	0x31f3cb	net/http.(*persistConn).writeLoop+0x9b	net/http/transport.go:2421

1 @ 0x50aa8 0x61e18 0x4ee9e0 0x86344
#	0x4ee9df	database/sql.(*DB).connectionOpener+0x7f	database/sql/sql.go:1218

1 @ 0x50aa8 0x62c48 0x62c25 0x817c8 0x1ad18e8 0x1ad1895 0x19a8e2c 0x86344
#	0x817c7		sync.runtime_SemacquireRWMutexR+0x27			runtime/sema.go:82
#	0x1ad18e7	sync.(*RWMutex).RLock+0x77				sync/rwmutex.go:71
#	0x1ad1894	github.com/evcc-io/evcc/core.(*Loadpoint).Title+0x24	github.com/evcc-io/evcc/core/loadpoint_api.go:18
#	0x19a8e2b	github.com/evcc-io/evcc/server.(*Influx).Run+0x18b	github.com/evcc-io/evcc/server/influxdb.go:156
```

Anscheinend möchte Influx den Fahrzeugtitle abfragen/ mit loggen und dafür das Mutex sperren. Das wiederrum hängt noch in `setRemainingEnergy`. Warum es da nicht raus kommt ist mir allerdings ein Rätsel.

Wastus, 2023-12-26:
Habe ebenfalls das Problem. Wenn man Influx deaktiviert läuft es, das ist zumindest für mich ein akzeptabler Workaround bis zum nächsten Update.

vistalba, 2023-12-26:
> Habe ebenfalls das Problem. Wenn man Influx deaktiviert läuft es, das ist zumindest für mich ein akzeptabler Workaround bis zum nächsten Update.

Ich habe einfach das „Gastfahrzeug“ geladen und anschliessend in der Ladehistorie das Fahrzeug für deo Statistik geändert. Hat auch funktioniert.

closed 2023-12-26 by andig