evcc-io/evcc #11198: 0.123 on homeasistant do not start

Ralf-Th, 2023-12-23:
### Describe the bug

the addon do not start after update 

### Steps to reproduce

1. install and reboot on latest hass
2.
3.
...

### Configuration details

```yaml
nothing changed
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/12/22 20:45:17 evcc 0.123.0
[main  ] INFO 2023/12/22 20:45:17 using config file: /config/evcc.yaml
[main  ] INFO 2023/12/22 20:45:17 starting ui and api at :7070
[db    ] INFO 2023/12/22 20:45:17 using sqlite database: /data/evcc.db
[main  ] FATAL 2023/12/22 20:45:21 failed configuring loadpoints: failed configuring loadpoint: 1 error(s) decoding:

* '' has invalid keys: resetondisconnect
[main  ] FATAL 2023/12/22 20:45:21 will attempt restart in: 5m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.0

VolkerK62, 2023-12-23:
pls check https://github.com/evcc-io/evcc/discussions/10978

nreese1994, 2023-12-23:
Same here. Not starting in Home Assistant!

parjanos, 2023-12-23:
I have the same problem...

nreese1994, 2023-12-23:
You'd need to just your evcc.yaml ... a bit unfortunate I'd say if you have no time right now but need EVCC... but is what it is...

I just "downgraded" back to 0.122.1 and its back working!

will go trough the changes after Christmas to adjust ... good for now!
[image]

TobiasHuber1980, 2023-12-23:
resetondisconnect auskommentieren bzw. komplett löschen!

parjanos, 2023-12-23:
> "resetondisconnect" in der yaml löschen, wird nicht mehr benötigt

Thank you, that works!

closed 2023-12-23 by andig

Ralf-Th, 2023-12-23:
thanks, works fine