evcc-io/evcc #11199: Battery Control for Solarmax

thebrainkafka, 2023-12-23:
For details see #10768.

Über Register 140 und 141 kann eine Vorgabe für die Lade-/Entladeleistung gegeben werden.

Die Werte sind „32 Bit Signed Integer“ Werte. Positive Werte stehen fürs Laden der Batterie und negative Werte fürs Entladen der Batterien.

Die Vorgabe wird jedoch nur umgesetzt, wenn im Watchdog Register ein Wert ungleich 0 steht.

Außerdem muss sich dieser Wert regelmäßig ändern.

Wenn der Wert im Watchdog Register länger als 10 Sekunden den gleichen Wert hat, wird es auf 0 zurückgesetzt, und die Vorgabe wird wieder ignoriert.

In der Praxis heißt das:

141 -> 0
142 -> 1

Unterbindet das Entladen der Hausbatterie.

Das ganze alle <59s wiederkehrend in 142 schreiben, solange wie die Sperre benötigt wird.
Sobald 142 nicht innerhalb 60s mit 1 beschrieben wurde oder aktiv 0 gesetzt wird geht die Batterie wieder in die Entladung.

Die 141 muss nicht erneut beschrieben werden.
Sobald der Watchdog 142 nicht mehr im Intervall gesetzt wird geht die 141 auf default zurück.

Also der 142 springt dann wieder auf 0 (wenn nicht innerhalb 60s die 1 geschrieben wird)
Zum halten des setting auf 141 (in unserem Fall die 0) muss dann immer nur der WD getriggert werden.

Zwangsladen geht nicht.

andig, 2023-12-23:
Das Thema hatten wir schon. Wir steuern aktuell keine Leistungen- daran hat sich erstmal nichts geändert!

manny4566, 2023-12-23:
Ich hänge mich mir mal mit rein. 
Schön, dass endlich darüber nachgedacht wird, Batterien zu "locken" wenn man das Auto lädt. Super, aber das habe mich schon vor Monaten hier angesprochen und es hat leider keinen interessiert -.- 

Gerne würde ich mir das auch für Sungrow Batterien wünschen. Die Register sind bekannt und können genutzt werden um die Batterie zu sperren, ähnlich wie bein Solarmax.
Gerne kann ich das bei mir testen, wenn es implementiert wird. 
Danke

andig, 2023-12-23:
@manny4566 bitte keine OT Postings. Sungrow ist im Release.

andig, 2023-12-25:
Das Problem bleibt hier das bereits mehrfach Besprochene. Leistung zu nullen ist kein Problem. Wir können den Wert allerdings nicht wieder zurücksetzen da wir den Default nicht kennen und nicht nicht kennen wollen. Bliebe also nur, des Watchdog abzuschalten womit die Null ungültig würde.

andig, 2023-12-25:
Depends on https://github.com/evcc-io/evcc/pull/11166 für Modbus LSW First.

thebrainkafka, 2023-12-25:
> Das Problem bleibt hier das bereits mehrfach Besprochene. Leistung zu nullen ist kein Problem. Wir können den Wert allerdings nicht wieder zurücksetzen da wir den Default nicht kennen und nicht nicht kennen wollen. Bliebe also nur, des Watchdog abzuschalten womit die Null ungültig würde.

Einfach den Watchdog nicht mehr im Intervall beschreiben oder die Null auf den Watchdog setzen. Genau.

thebrainkafka, 2024-01-11:
> Depends on #11166 für Modbus LSW First.

Gibt es hierzu ein update?

thebrainkafka, 2024-01-20:
> Depends on #11166 für Modbus LSW First.

Ist es nun von der Umsetzung #11663 abhängig oder?

thebrainkafka, 2024-01-28:
Hi @andig .
Nachdem 11663 nun merged ist. Kannst du mir sagen ob die Batteriesteueung des Solarmax nun implementiert werden kann?
Danke.

andig, 2024-02-03:
@thebrainkafka laut https://github.com/evcc-io/evcc/discussions/10768#discussioncomment-7645439 gibt es immer noch keine vollständige Doku. Die erste Voraussetzung wäre, dass Solarmax eine Doku zur Verfügung stellt.

Zweitens haben wir mehrfach gesagt, dass wir keine Leistungen steuern wollen. Bei Übergang nach normal müsste aber die "richtige" Maximalleistung geschrieben werden. Das wäre etwas anderes, wenn diese eine prozentuale Limitierung wäre- da könnte man wieder auf 100% gehen.

Long story short: aktuell wontfix, die kannst die Steuerung natürlich extern umsetzen.

closed 2024-02-03 by andig

thebrainkafka, 2024-09-11:
Hallo @andig .
Gibt es zum generellen Ansatz die Batteriesteuerung über die Vorgabe der Lade/Entladeleistung zu steuern ein Update?
Eine Dokumentation der Register liegt mir vor.

Der Übergang zu "normal" muss nicht geschrieben werden. Dieser stellt sich automatisch wieder ein wenn der Watchdog nicht mehr getriggert wird.

Wie bereits von mir getestet muss also "nur" die Entladeleistung auf 0 gesetzt werden und der Watchdog im Intervall getriggert (solange erforderlich)

Danke für deine Rückmeldung hierzu.

premultiply, 2024-09-11:
> Eine Dokumentation der Register liegt mir nun vor.

Dann bitte einfach hier anhängen.

> Der Übergang zu "normal" muss nicht geschrieben werden. Dieser stellt sich automatisch wieder ein wenn der Watchdog nicht mehr getriggert wird.

> Wie bereits von mir getestet muss also "nur" die Entladeleistung auf 0 gesetzt werden und der Watchdog im Intervall getriggert (solange erforderlich)

Kann man wie bei anderen WR auch so machen.

thebrainkafka, 2024-09-11:
Für mein Verständnis.
Heißt ihr könnt es implementieren und ich soll die Registerliste anfügen?
Danke

thebrainkafka, 2024-09-11:
@premultiply 

Hier nochmals wie ich es getestet habe.
Liste anbei.

Über Register 140 und 141 kann eine Vorgabe für die Lade-/Entladeleistung gegeben werden.

Die Werte sind „32 Bit Signed Integer“ Werte. Positive Werte stehen fürs Laden der Batterie und negative Werte fürs Entladen der Batterien.

Die Vorgabe wird jedoch nur umgesetzt, wenn im Watchdog Register (142) ein Wert ungleich 0 steht.

Außerdem muss sich dieser Wert regelmäßig ändern.

Wenn der Wert im Watchdog Register länger als 60 Sekunden den gleichen Wert hat, wird es auf 0 zurückgesetzt, und die Vorgabe wird wieder ignoriert.

In der Praxis heißt das:

141 -> 0
142 -> 1

Unterbindet das Entladen der Hausbatterie.

Das ganze alle <59s wiederkehrend in 142 (Watchdog) schreiben, solange wie die Sperre benötigt wird.
Sobald 142 nicht innerhalb 60s mit 1 beschrieben wurde oder aktiv 0 gesetzt wird geht die Batterie wieder in die Entladung.

Die 141 (Entladeleistung) muss nicht erneut beschrieben werden.
Sobald der Watchdog 142 nicht mehr im Intervall gesetzt wird geht die 141 auf default zurück.

Also der 142 springt dann wieder auf 0 (wenn nicht innerhalb 60s die 1 geschrieben wird)
Zum halten des setting auf 141 (in unserem Fall die 0) muss dann immer nur der WD getriggert werden.

Zwangsladen geht nicht.

[MAX.STORAGE ModbusTCP Datenpunktliste_extended 3.4.4.pdf](https://github.com/user-attachments/files/16966405/MAX.STORAGE.ModbusTCP.Datenpunktliste_extended.3.4.4.pdf)

thebrainkafka, 2024-09-12:
@andig 
@premultiply 

Hab es nochmals ausführlich getestet und die Anleitung im letzten post upgedated.
Passt das für euch?
Danke

thebrainkafka, 2024-09-13:
Benötigt ihr noch etwas (Infos, Tests) ? Muss ich etwas anstoßen?

andig, 2024-09-14:
Es braucht für alle 3 Modi Register + Wert

thebrainkafka, 2024-09-14:
Enable (Batteriesperre):
140 und 141 -> 0
142 -> 1 (Zufallszahl ungleich 0, welche alle 30s geschrieben wird, darf sich nicht wiederholen; z.b also 1 und dann 2 und dann wieder 1 usw).

Disable (Batterie wird wieder genutzt)
Keine Eingabe erforderlich, da nachdem der 142 nicht mehr mit ungleich 0 beschrieben wird alles auf default geht bzw. die defaultwerte sind:
140 und 141 -> 7000
142 0

Was ist der 3. Modi?
Laden aus dem Netz geht nicht.

andig, 2024-09-14:
Bitte mal testen

thebrainkafka, 2024-09-14:
Vielen Dank.
Muss ich dazu zum nightly wechseln?
Wenn ja vielleicht dumme Frage:
Wie wechsel ich die repo?

andig, 2024-09-14:
Bitte nicht bei jedem PR die gleiche Frage stellen :( Ein PR ist erst im Nightly wenn er gemerged ist. Also Branch auschecken und selber bauen.

thebrainkafka, 2024-09-14:
Sorry wenn ich dies schon einmal gefragt hatte.
Zum selber bauen fehlen mir leider die Kenntnisse.
Alternativen es zu testen?

andig, 2024-09-14:
Es gibt keine. Ich schiebs rein.

thebrainkafka, 2024-09-14:
D.h. ich kann es übers nightly morgen testen - richtig?

andig, 2024-09-14:
Auch die Frage gabs schon 1000 mal :(

thebrainkafka, 2024-09-14:
Sorry. Melde mich wenn ich es morgen aus dem Nightly getestet habe.

thebrainkafka, 2024-09-14:
Frage zum code 
Bedeutet type: writemultiple dass der Wert mehrmals oder wiederkehrend geschrieben wird.

Hold
140/141 müssen nur einmal beschrieben werden, danach nur wiederholend der 142 Watchdog (30s)

Zeile 118 heißt charge

thebrainkafka, 2024-09-15:
Log anbei.
Nach dem update auf das nightly startet evcc nicht mehr
[evcc-20240915-102114-debug.log.txt](https://github.com/user-attachments/files/17005067/evcc-20240915-102114-debug.log.txt)

andig, 2024-09-15:
Ups, da fehlte der Defaultwert. Neues Nightly in 30min.

thebrainkafka, 2024-09-15:
Top. Danke!

Sperrt, hält die sperre, entsperrt im schnell lademous.
Keine Fehler im debugblog.
Lass Nachmittags mal per schedule laden.

thebrainkafka, 2024-09-15:
Für die Doku - bitte ergänzen:
Zur Nutzung der Funktion Batterie Discharge Control muss über den Solarmax Support die Funktion "Connectivity+" freigeschaltet werden.

andig, 2024-09-15:
Was passiert wenn man das nicht hat? Müssen wir das im Template jetzt optional machen weils sonst Fehler gibt? Zurück rollen? Die Info kommt jetzt etwas überraschend…

thebrainkafka, 2024-09-15:
Dann kann man die Register nicht schreiben.
Passiert also nichts, funktioniert schlicht weg nicht. Fehler sollte es dadurch keine geben.

Man kann auch einfach den Schieber in der UI auf deaktiviert lassen.

Kurzer Anruf bei der Hotline und es wird freigeschaltet und geht dann.

Optional wäre ggf. eine gute Idee.
Du meinst ein template mit und eines ohne aktive Batteriesteuerung?