evcc-io/evcc #11202: Error at startup after update to 123.0

rwersuhn, 2023-12-23:
### Describe the bug

I have updated to 123.0 and have removed from yaml:

- mode
- Resetondisconnect
- Targetsoc

**FEHLER BEIM STARTEN**

cannot create charger 'wallbox3'
cannot create charger 'template'
cannot create charger 'eebus'
i/o timeout

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
#sponor evcc docu oder https://sponsor.evcc.io  
sponsortoken: *****  
  
# unique installation id  
plant: *****  
  
interval: 10s # control cycle interval  
  
meters:  
- type: template  
  template: tasmota   
  usage: grid    
  host: 192.168. …
  name: grid1  
  
chargers:  
- type: template  
  template: elliconnect   
  ski: AED …
  ip: 192.168. …  
  name: wallbox3  
  
vehicles:  
- type: template  
  template: id   
  title: eGolf    
  user: *****  
  password: *****  
  vin: *****  
  capacity: 35.8    
  name: ev2  
  phases: 2  
  #targetSoc: 75  
  maxcurrent: 13  
  
loadpoints:  
- title: Garage  
  charger: wallbox3  
  vehicle: ev2  
  #mode: pv  
  phases: 3  
  maxcurrent: 16  
  #resetOnDisconnect: true  
  
site:  
  title: Home 123.0  
  meters:  
    grid: grid1  
  
eebus:  
  certificate:  
      private: |  
          -----BEGIN EC PRIVATE KEY-----  
          MHc …  
          
          -----END EC PRIVATE KEY-----  
      public: |  
          -----BEGIN CERTIFICATE-----  
          MIIBv …
          -----END CERTIFICATE-----
```

### Log details

```text
Debian GNU/Linux
Last login: Sat Dec 23 11:22:26 2023
Userxy@debian:~$ evcc --log debug
[main  ] INFO 2023/12/23 11:54:54 evcc 0.123.0
[main  ] INFO 2023/12/23 11:54:54 using config file: /etc/evcc.yaml
[main  ] INFO 2023/12/23 11:54:54 starting ui and api at :7070
[db    ] INFO 2023/12/23 11:54:54 using sqlite database: /home/Userxy/.evcc/evcc.db
[eebus ] INFO 2023/12/23 11:54:54 Local SKI:  67b74 …
[eebus ] DEBUG 2023/12/23 11:54:54 mdns: announce
[eebus ] DEBUG 2023/12/23 11:54:54 mdns: using zeroconf
[eebus ] DEBUG 2023/12/23 11:54:54 starting websocket server on :4712
[eebus ] DEBUG 2023/12/23 11:54:54 mdns: start search
[main  ] FATAL 2023/12/23 11:56:24 cannot create charger 'wallbox3': cannot create charger 'template': cannot create charger 'eebus': i/o timeout
[main  ] FATAL 2023/12/23 11:56:24 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.123.0

StefanSchoof, 2023-12-23:
Hast du selbst gebaut? Wenn ja es gab ein Update der Go Crypto lib und du musst `make patch-asn1-sudo` wieder ausführen.

andig, 2023-12-23:
Falls das das offizielle Release ist: 

> cannot create charger 'wallbox3': cannot create charger 'template': cannot create charger 'eebus': i/o timeout

Die Box muss sich bei EEbus mit evcc verbinden. Macht sie anscheinend nicht...

andig, 2023-12-23:
PS.: Bitte sprechende Fehlerbeschreibungen

closed 2023-12-25 by andig