evcc-io/evcc #11208: 0.123.0: Request failed with status code 404: API request failed

exel90, 2023-12-23:
### Describe the bug

Seit dem Update auf 0.123.0 kann ich mein Fahrzeug nicht mehr auswählen.
Es wird mit roten Dreieck oben rechts folgender Fehler angezeigt: 
`Request failed with status code 404: API request failed http://192.168.69.8:7070/api/loadpoints/1/vehicle/Model 3
`

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2023/12/23 14:24:24 evcc 0.123.0
[main  ] INFO 2023/12/23 14:24:24 using config file: /etc/evcc.yaml
[db    ] INFO 2023/12/23 14:24:25 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] DEBUG 2023/12/23 14:24:26 charge total import: 1267.149kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- name: grid1
  type: custom
  power:
    source: http
    method: GET
    uri: http://192.168.69.8:8087/getPlainValue/javascript.0.Solarpower.Huawei.Meter.ActivePower
    scale: -1
  energy: # Energie (kWh)
    source: http
    method: GET
    uri: http://192.168.69.8:8087/getPlainValue/alias.0.Netzbezug-Gesamt.ReverseActiveEnergy
  currents: # optional currents (A)
    - source: http # L1 plugin type
      method: GET
      uri: http://192.168.69.8:8087/getPlainValue/javascript.0.Solarpower.Huawei.Meter.CurrentL1
    - source: http # L2 plugin type
      method: GET
      uri: http://192.168.69.8:8087/getPlainValue/javascript.0.Solarpower.Huawei.Meter.CurrentL2
    - source: http # L3 plugin type
      method: GET
      uri: http://192.168.69.8:8087/getPlainValue/javascript.0.Solarpower.Huawei.Meter.CurrentL3
- name: pvSum
  type: custom
  power:
    source: http
    method: GET
[... 173 lines cut ...]
[site  ] DEBUG 2023/12/23 14:21:38 pv power: 278W
[site  ] DEBUG 2023/12/23 14:21:38 grid power: 97W
[site  ] DEBUG 2023/12/23 14:21:38 grid currents: [0.46 -0.87 0.33]A
[site  ] DEBUG 2023/12/23 14:21:38 site power: 97W
[lp-1  ] DEBUG 2023/12/23 14:21:38 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/23 14:21:38 charge total import: 1267.149kWh
[lp-1  ] DEBUG 2023/12/23 14:21:38 charger status: A
[site  ] DEBUG 2023/12/23 14:22:08 ----
[lp-1  ] DEBUG 2023/12/23 14:22:08 charge power: 0W
[site  ] DEBUG 2023/12/23 14:22:08 pv power: 276W
[site  ] DEBUG 2023/12/23 14:22:08 grid power: 1554W
[site  ] DEBUG 2023/12/23 14:22:08 grid currents: [0.43 -0.86 -6.36]A
[site  ] DEBUG 2023/12/23 14:22:08 site power: 1554W
[lp-1  ] DEBUG 2023/12/23 14:22:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/23 14:22:08 charge total import: 1267.149kWh
[lp-1  ] DEBUG 2023/12/23 14:22:08 charger status: A
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.123.0

Sammy990, 2023-12-23:
+1, same here for a Model Y

andig, 2023-12-23:
@naltatis wir haben ` ` nicht für Namen zugelassen:

    "vehicle":          {[]string{"POST", "OPTIONS"}, "/vehicle/{name:[a-zA-Z0-9_.:-]+}", vehicleSelectHandler(site, lp)},

Sollen wir das ändern?

StefanSchoof, 2023-12-23:
Does it work if you use a name without space and a title the display name with the space?

CHB68, 2023-12-23:
Das liegt am Leerzeichen zwischen Model und 3
Ohne Leerzeichen funktioniert es

CHB68, 2023-12-23:
An dieser Stelle:
vehicles:
  - name: Model 3

andig, 2023-12-23:
Ich habs in die Releasenotes mit aufgenommen:

> Vehicles have been moved from `site` into top level `vehicles` object. Vehicles are no longer addressed by index, but using their config `name`. `name` MUST NOT contain spaces or special characters.

exel90, 2023-12-23:
Thank you very much.

closed 2023-12-23 by exel90