evcc-io/evcc #11214: Kia Niro EV (2023) Integration partially fails

Magnus8513, 2023-12-23:
### Describe the bug

I did not yet succeed to integrate my new 2023 Kia Niro EV. Here is what I configured:

`vehicles:
  - name: NiroEV
    type: template
    template: kia
    title: NiroEV # Will be displayed in the user interface (optional)
    user: mail@myprovider.de # User account (e.g. email address, user id, etc.)
    password: abcde123456 # Password of the user account (use single quotes in case of leading zeros)
    vin: W... # Required if you own multiple vehicles of the same brand (optional)
    capacity: 64 # Battery capacity in kWh (optional)
    language: de # 'en' for English and 'de' for German (optional) `

and this is what I get as a result:

`evcc vehicle NiroEV
[main  ] INFO 2023/12/23 16:32:09 evcc 0.123.0
[main  ] INFO 2023/12/23 16:32:09 using config file: /etc/evcc.yaml
[db    ] INFO 2023/12/23 16:32:10 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] ERROR 2023/12/23 16:32:20 creating vehicle NiroEV failed: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
Capacity: 64.0kWh
Features: [Offline]
`
In the UI I see an vehicle as "NiroEV (offline)" that I can use but without getting the SOC.

Any ideas of what is going wrong here?

Via the Kia App I can access the device and also the likes of Tronity or other API using counterparts I can access my data.

Regards
Magnus

### Steps to reproduce

1. configure additional devices as shown in config below
2. call: evcc vehicle NiroEV
3. get result as described above
...

### Configuration details

```yaml
vehicles:
  - name: NiroEV
    type: template
    template: kia
    title: NiroEV # Will be displayed in the user interface (optional)
    user: mail@myprovider.de # User account (e.g. email address, user id, etc.)
    password: abcde123456 # Password of the user account (use single quotes in case of leading zeros)
    vin: W... # Required if you own multiple vehicles of the same brand (optional)
    capacity: 64 # Battery capacity in kWh (optional)
    language: de # 'en' for English and 'de' for German (optional)
```

### Log details

```text
[main  ] INFO 2023/12/23 16:32:09 evcc 0.123.0
[main  ] INFO 2023/12/23 16:32:09 using config file: /etc/evcc.yaml
[db    ] INFO 2023/12/23 16:32:10 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] ERROR 2023/12/23 16:32:20 creating vehicle NiroEV failed: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
Capacity: 64.0kWh
Features: [Offline]
```

### What type of operating system are you running?

Linux

### Version

0.123.0

wolfthom, 2023-12-23:
I believe this could be related to a change in the KIA API or some downtime / issues. Getting my ev6 SoC stopped working a few hours ago with same error as mentioned above.

VolkerK62, 2023-12-23:
Same here with Hyundai Ioniq5.
NodeRed bluelinky is also on ERROR.

0x3d13f, 2023-12-23:
my EV6 also started being "offline" in the last hour.
Kia Connect App works, but seems a bit slow

andig, 2023-12-23:
We need a fix from one of the open source solutions to understand the changes.

MikeSchatz, 2023-12-23:
Same problem with Kia Niro EV

peregrines2, 2023-12-23:
Same problem here with Kia Niro EV MY19. 

"Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/12/23 21:53:01 evcc 0.123.1
[main  ] INFO 2023/12/23 21:53:01 using config file: /config/evcc.yaml
[main  ] INFO 2023/12/23 21:53:01 starting ui and api at :7070
[mqtt  ] INFO 2023/12/23 21:53:01 connecting evcc-1729022546 at tcp://192.168.178.129:1883
[main  ] ERROR 2023/12/23 21:53:02 creating vehicle niro failed: cannot create vehicle 'template': invalid key: targetsoc
[main  ] FATAL 2023/12/23 21:53:02 failed configuring loadpoints: failed configuring loadpoint: 1 error(s) decoding:

* '' has invalid keys: resetondisconnect
[main  ] FATAL 2023/12/23 21:53:02 will attempt restart in: 5m0s"

andig, 2023-12-23:
@peregrines2 lies bitte die 0.123 Releasenotes. Dein Problem hat hiermit nichts zu tun.

Stepsilein, 2023-12-23:
Sorry, aber Ioniq5 offline, auch  bei mir seit heute Nachmittag.
Gestriges Hyundai OTA Update hat API geändert oder was steckt dahinter???
Mit changelog Hinweis kann ich das nicht verifizieren... das hat doch nur mit dem Loadoint Dings Auswirkungen gehabt, oder?

emcepe, 2023-12-24:
Hyundai hatte etliche Stunden, bzw. eigentlich fast den ganzen 23.12. Probleme. Die sind nun wieder weg und API Anfragen funktionieren wieder.

wolfthom, 2023-12-24:
Die API-Probleme auf KIA/Hyundai-Seite sind gefixt. Bei mir bedurfte es eines Neustarts von evcc und nun ist der ev6 wieder online und die Daten werden von der API abgeholt und angezeigt. @Magnus8513 , @VolkerK62: passt es nun bei Euch auch wieder?

andig, 2023-12-24:
Doch keine Änderung? Super- danke für die Info 🎅🏻

closed 2023-12-24 by andig

VolkerK62, 2023-12-24:
jap ... geht wieder

0x3d13f, 2023-12-24:
Hier geht es nach evcc restart auch wieder. 
Auch wenn diesmal nichts an evcc anzupassen war,  an dieser Stelle ein großer Dank an das Entwickler Team. In Gegensatz zu manchem (siehe oben) sehe ich hier ein tolles Engagement ein gutes Produkt zu liefern, was angesichts der Komplexität (diverse Gerätschaften vom Wechselrichter über Batterien, Wallboxen und nicht zuletzt Autos müssen unterstützt werden, es ändert sich dauernd irgendwo etwas, die meisten Autos liefern nicht mal eine offizielle API etc.) aus meiner Sicht sehr gut gelingt. 
Und dann wird der Mut, bei der Weiterentwicklung des eigentlichen Produkts auch mal einen Schritt zu gehen, der ein paar kleine Anpassungen seitens der Anwender nötig macht und weit im Voraus angekündigt war mit der Eröffnung einer Vielzahl von Tickets belohnt, die anders als mit "bitte Release Notes lesen" einfach nicht zu beantworten sind und man bekommt auch noch Kommentare wie oben reingedrückt. Fair finde ich das nicht.

Magnus8513, 2023-12-24:
Es klappt auch bei mir nun. Danke für den konstruktiven Austausch hier!

Stepsilein, 2023-12-24:
Läuft auch bei mir wieder.
Ich bitte trotzdem meine konstruktive Kritik betr. besserer Info und Anleitungen zu beachten.
Dass ich Evcc und dessen Macher immer lobe, liest man ja auch.
Besser betr. Anleitungen und Info geht aber gewiss und ist teils nötig.

Nun Allen ein frohes Fest 🎄

andig, 2023-12-24:
Ach das war *konstruktive* Kritik? Las sich eher wie eine Forderungsliste bei ner Geiselentführung. Schöne Bescherung.  🎅🏻

Stepsilein, 2023-12-24:
> Ach das war _konstruktive_ Kritik? Las sich eher wie eine Forderungsliste bei ner Geiselentführung. Schöne Bescherung. 🎅🏻

Forderung hört sich bei mir anders an 😘.
Ich weiß, ist "Nerd Github"... trotzdem sollte besser erläutert werden für Nichtwissende.
Mir ist klar, dass die Macher wissen was sie meinen, aber auch auf Github darf man erklären, oder?
Weit lieber wäre mir da eine stets aktuelle Evcc Webseite, aber dort stehen teils veraltete Sachen die nicht mehr so funktionieren wie in deren Beispielen.
Mir ist auch klar, dass Alles quasi ehrenamtlich läuft und Lob hier nochmals für das tolle Evcc.
Aber nicht allergisch auf den/meinen Blick von außen reagieren.
Hat sonst leicht Elfenbeinturm Feeling.
Ich sage es so, wie ich es empfunde.
Habe oft über das Nichterklären und verkusterte hier auf Github geflucht...
Stört mich, da ich dazu nicht nerdig genug bin scheinbar 🤷🏼‍♂️.
Bitte trotzdem mal über meine angesprochenen Sachen nachdenken, danke.
Liebe Grüße 
Stephan 🙋🏼‍♂️

VolkerK62, 2023-12-24:
@Stepsilein 
>Weit lieber wäre mir da eine stets aktuelle Evcc Webseite, aber dort stehen teils veraltete Sachen die nicht mehr so funktionieren wie in deren Beispielen.

Doku kann jeder bearbeiten. Ist sogar besser, wenn das ein Nicht-Nerd mach.

Morpheus211178, 2023-12-31:
Also bei mir wird immer noch offline angezeigt , ich verstehe es nicht

0x3d13f, 2023-12-31:
Schau mal, ob Du bei vehicle noch veraltete Paramter drin hast. 
Siehe auch: https://github.com/evcc-io/evcc/issues/11338