evcc-io/evcc #11216: SoC via openWB: wieder DNS Probleme

martin-zh, 2023-12-23:
### Describe the bug

Das Problem hatten wir nun schon mehrfach:

2023-12-23 16:56:50: PID: 4850: Lp2: Error from EVCC: 2023/12/23 16:56:50 rpc error: code = Unknown desc = cannot create vehicle 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v1/users/xxx/garage/vehicles": dial tcp: lookup ola.prod.code.seat.cloud.vwgroup.com on [fdaa::3]:53: server misbehaving (LV0)
2023-12-23 16:56:44: PID: 4850: Lp2: Requesting SoC (LV0)

Es wäre toll, wenn die Ursache behoben werden könnte.

### Steps to reproduce

SOC via openWB abfragen

### Configuration details

```yaml
openWB
```

### Log details

```text
Siehe obem
```

### What type of operating system are you running?

Linux

martin-zh, 2023-12-23:
Sorry, wurde ja schon gemeldet und ist open.
Könnte man den Fehler anfangen und zb NULL zurückgeben anstatt des letzten SoCs? So würde das wenigstens auffallen anstatt mit zu wenig Batterie dazustehen

andig, 2023-12-23:
Duplikat: https://github.com/evcc-io/evcc/issues/11168

Die Fehlerbehebung muss in openWB erfolgen. Evcc nimmt konservativ soc 0 an wenn es keinen Wert bekommt. Bitte openWB Support kontaktieren.

closed 2023-12-23 by andig