evcc-io/evcc #11226: co2 tariff outdated, aber Daten vorhanden

VolkerK62, 2023-12-24:
### Describe the bug

DIe Ausgabe von `evcc tariff` liefert 
```
co2:
[main  ] FATAL 2023/12/24 08:47:15 outdated
```
Im UI sind die Daten vorhanden. Keine funktionalen Einschränkungen spürbar.

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
co2:
[gsi   ] TRACE 2023/12/24 08:51:28 GET https://api.corrently.io/v2.0/gsi/prediction?zip=***
[gsi   ] TRACE 2023/12/24 08:51:29 {"support":"If you have any question please contact dev@stromdao.com or visit https://premium.corrently.de/","license":"CC BY-NC-SA 4.0 ( https://creativecommons.org/licenses/by-nc-sa/4.0/legalcode.en )","info":"https://corrently.io/books/stromdao-gmbh/page/fair-use-policy","documentation":"https://corrently.io/books/grunstromindex","commercial":"https://corrently.energy/products/corrently-apis-grunstromindex-abrufe","signee":"0xE7Fe0626D7B8e3F2e5ECD146F9b11daac1DBE447","forecast":[{"epochtime":1703397600,"eevalue":9,"ewind":0,"esolar":0,"ensolar":0,"enwind":0,"sci":25,"gsi":8.549999999999999,"timeStamp":1703397600000,"energyprice":"-0.0045000","co2_g_standard":449,"co2_g_oekostrom":60,"timeframe":{"start":1703397600000,"end":1703401200000},"iat":1703402394761,"zip":"***","signature":"0xdeb35be3fb0cfa7ea6266b30eed22c06dd82dc9075c84848bd2fb0a36f873bc43183039cfb35cf3005714fd60cd305f55bd30d35dfff93ce3a46cba656c3c4dc1c"},{"epochtime":1703401200,"eevalue":100,"ewind":25,"esolar":67,"ensolar":67,"enwind":25,"sci":59,"gsi":95,"timeStamp":1703401200000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703401200000,"end":1703404800000},"iat":1703402394763,"zip":"***","signature":"0xbeea51c9bfd3de9eeb92e26b3af7c57a91dbf94f28cb4497c0b5087c944b0d1b0172924483fdb327ed45c6eb5bcc0b1a50d1b33ac1958400616f318f70f5a90e1b"},{"epochtime":1703404800,"eevalue":100,"ewind":33,"esolar":80,"ensolar":80,"enwind":33,"sci":65,"gsi":95,"timeStamp":1703404800000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703404800000,"end":1703408400000},"iat":1703402394775,"zip":"***","signature":"0xe38cd0045bca132f150fa5e9cb8d0759b8a8d1efc00169443fb141be7b61c1e9775deb2569a4a55922dc3866bcea5f626da4a11c658e3709b4466e23ca0bbd7c1b"},{"epochtime":1703408400,"eevalue":100,"ewind":34,"esolar":80,"ensolar":80,"enwind":34,"sci":65,"gsi":95,"timeStamp":1703408400000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703408400000,"end":1703412000000},"iat":1703402394776,"zip":"***","signature":"0x435bcd77e875b63b38359b89e94d95c6fec275783e393a691aa4ef9ec14807da35befaa7cc0af891fcf7a6800987f18b5ea69898880ac1b828cc366cc3ebe2d01c"},{"epochtime":1703412000,"eevalue":100,"ewind":37,"esolar":80,"ensolar":80,"enwind":37,"sci":65,"gsi":95,"timeStamp":1703412000000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703412000000,"end":1703415600000},"iat":1703402394778,"zip":"***","signature":"0xb468ad0861ea76c9061393d30deb6b8b5ce2726f914c80ab77c9ba293aaa6d01043823d45bc1c135de03bcae570e9c5f4c3374cd40a79afccc9b5f39e2c89ddd1c"},{"epochtime":1703415600,"eevalue":100,"ewind":41,"esolar":80,"ensolar":80,"enwind":41,"sci":65,"gsi":95,"timeStamp":1703415600000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703415600000,"end":1703419200000},"iat":1703402394785,"zip":"***","signature":"0x4f767945f0bb74dcd51ca611e9e989f0aa0afeab9064be6af4930360131cd0a36e63de71af0c325adca41617282b3961f6710ecb1abb4ea081198617cec9620f1c"},{"epochtime":1703419200,"eevalue":100,"ewind":40,"esolar":80,"ensolar":80,"enwind":40,"sci":65,"gsi":95,"timeStamp":1703419200000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703419200000,"end":1703422800000},"iat":1703402394788,"zip":"***","signature":"0xfe85f57ed7bb9497453fefe4665cf4b379a9f69c65206be1b7d7fa75cedf514e4a9366a4b8c5a6265ef8bb1e28a6ea12ff79884bf5c5f540df52e5b6e7bdc3ca1c"},{"epochtime":1703422800,"eevalue":100,"ewind":40,"esolar":80,"ensolar":80,"enwind":40,"sci":65,"gsi":95,"timeStamp":1703422800000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703422800000,"end":1703426400000},"iat":1703402394796,"zip":"***","signature":"0x47f5c022032c6b86891d447c9b7ccf91e9dd8d1a2786cfabed05371aa9e1c1b124d4a9f31993e688012c99b33ece7defb4d7ae46c3320d5281fe6f64678229bd1b"},{"epochtime":1703426400,"eevalue":100,"ewind":42,"esolar":80,"ensolar":80,"enwind":42,"sci":65,"gsi":95,"timeStamp":1703426400000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703426400000,"end":1703430000000},"iat":1703402394798,"zip":"***","signature":"0x3b5e2b09d61aa034e04408549f87ffe7222a97e48c1e43a9f7c3bb0a0444a5ca49eb03a03518dfcb0a24412aea036010a847af5dd40f1873cd74d0aef343b13d1b"},{"epochtime":1703430000,"eevalue":100,"ewind":39,"esolar":80,"ensolar":80,"enwind":39,"sci":65,"gsi":95,"timeStamp":1703430000000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703430000000,"end":1703433600000},"iat":1703402394799,"zip":"***","signature":"0x529dd9eb35f27cfdd303221eb55fdc7f541571f5b8079d5ccd120e35752125855a83ce0f7660eb6483aec2b620d1f31ec7fd2ff825c411ce28c4a0410eef0c4b1c"},{"epochtime":1703433600,"eevalue":100,"ewind":33,"esolar":80,"ensolar":80,"enwind":33,"sci":65,"gsi":95,"timeStamp":1703433600000,"energyprice":"-0.0500000","co2_g_standard":66,"co2_g_oekostrom":8,"timeframe":{"start":1703433600000,"end":1703437200000},"iat":1703402394801,"zip":"***","signature":"0x7fba4b965647bdfbc94c0b602e2e171eae5cc98005edae5012e98ac12e653b783792ee0b4c0b564601fba09833e3999fe4992e9c91aa51591ed68e61e4782f391b"},{"epochtime":1703437200,"eevalue":78,"ewind":23,"esolar":46,"ensolar":46,"enwind":23,"sci":48,"gsi":74.1,"timeStamp":1703437200000,"energyprice":"-0.0390000","co2_g_standard":159,"co2_g_oekostrom":21,"timeframe":{"start":1703437200000,"end":1703440800000},"iat":1703402394802,"zip":"***","signature":"0x467cc28c68634d0b1aa69c8217e96059ad4326cfcab14f9de47ad2d96110a719294b37ddfe2fba57a3c0e7c28cc4250b90d3b1e353f4254716c0b38408e572771c"},{"epochtime":1703440800,"eevalue":33,"ewind":23,"esolar":0,"ensolar":0,"enwind":23,"sci":25,"gsi":31.349999999999998,"timeStamp":1703440800000,"energyprice":"-0.0165000","co2_g_standard":348,"co2_g_oekostrom":46,"timeframe":{"start":1703440800000,"end":1703444400000},"iat":1703402394809,"zip":"***","signature":"0x2f7c5ef56da3d08116d2316fd31348442fb0b97dffa34fc971b3ed7c9da3abd97dc58d5d7fdd6dbca16c5f2d579a6c3663270c5a93c63b588332e17358e7d1b81b"},{"epochtime":1703444400,"eevalue":33,"ewind":25,"esolar":0,"ensolar":0,"enwind":25,"sci":25,"gsi":31.349999999999998,"timeStamp":1703444400000,"energyprice":"-0.0165000","co2_g_standard":348,"co2_g_oekostrom":46,"timeframe":{"start":1703444400000,"end":1703448000000},"iat":1703402394811,"zip":"***","signature":"0x87b5ad5ae4e64af341b2f1eedb879cb46a30013026c069ba768b20b0c15b36a506d1c2181dcad27bc56c50394e357e044e1480ff1525bbaee5a263ed03e6e3531b"},{"epochtime":1703448000,"eevalue":30,"ewind":21,"esolar":0,"ensolar":0,"enwind":21,"sci":25,"gsi":28.5,"timeStamp":1703448000000,"energyprice":"-0.0150000","co2_g_standard":361,"co2_g_oekostrom":48,"timeframe":{"start":1703448000000,"end":1703451600000},"iat":1703402394813,"zip":"***","signature":"0x9a6a64047152d3bf87b99e34df37b4da2edc758b1573aa1a1ab6c9b4271ee9797001bacd31bd8d43fdaa37fed01ab103734476e16c9c711aea7d0f4ffa997f681b"},{"epochtime":1703451600,"eevalue":29,"ewind":20,"esolar":0,"ensolar":0,"enwind":20,"sci":25,"gsi":27.549999999999997,"timeStamp":1703451600000,"energyprice":"-0.0145000","co2_g_standard":365,"co2_g_oekostrom":48,"timeframe":{"start":1703451600000,"end":1703455200000},"iat":1703402394818,"zip":"***","signature":"0x59e079a0aae7eec786a6508a82f4ace1d0328824562169e3b41a731d1e8f6c971cbbd49a87ebf2e7ab9d6a31e8521ce043144f1b95bfe8bb3badb62d949f00931b"},{"epochtime":1703455200,"eevalue":33,"ewind":25,"esolar":0,"ensolar":0,"enwind":25,"sci":25,"gsi":31.349999999999998,"timeStamp":1703455200000,"energyprice":"-0.0165000","co2_g_standard":348,"co2_g_oekostrom":46,"timeframe":{"start":1703455200000,"end":1703458800000},"iat":1703402394825,"zip":"***","signature":"0x78c23c26d55b7518e264205a88840caba2a568665c523129219e024ace9ebd022e3c45dffea0d76c69d57bce487a78c550309a2df2e6ce065529f3c9102be2c61c"},{"epochtime":1703458800,"eevalue":27,"ewind":18,"esolar":0,"ensolar":0,"enwind":18,"sci":25,"gsi":25.65,"timeStamp":1703458800000,"energyprice":"-0.0135000","co2_g_standard":373,"co2_g_oekostrom":50,"timeframe":{"start":1703458800000,"
[main  ] FATAL 2023/12/24 08:51:29 outdated
```

### What type of operating system are you running?

Linux

### Version

0.123.1

VolkerK62, 2023-12-24:
Hmm ... auf dem Laptop mit Debian WSL funktioniert es.

andig, 2023-12-24:
Es braucht ein trace log vom Fehler

VolkerK62, 2023-12-24:
>Es braucht ein trace log vom Fehler

trace log hab ich doch oben drin

andig, 2023-12-24:
Ah, böses scrolling

andig, 2023-12-24:
Geht bei mir. Du bist sicher, dass es nicht an einem anderen Deiner Tarife liegt?

GrimmiMeloni, 2023-12-24:
@VolkerK62 was ist mit Deiner Systemzeit vom evcc Host?

VolkerK62, 2023-12-24:
ich hab jetzt alle Tarife auskommentiert. Nur noch co2 übrig. Fehler bleibt.
Gleiche Version auf dem Laptop unter WSL gibt keinen Fehler.
Zeit müsste auch ok sein.

```
pi@raspberrypi:~ $ timedatectl
               Local time: So 2023-12-24 12:45:01 CET
           Universal time: So 2023-12-24 11:45:01 UTC
                 RTC time: n/a
                Time zone: Europe/Berlin (CET, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no

```

andig, 2023-12-24:
Ich denke das ist eine race condition. Die Initialisierung des Tarifs meldet schon "fertig" bevor der API Call die Daten abgeliefert und eingeloggt hat. Der Fehler ist in allen Tarifen drin und wird indirekt durch https://github.com/evcc-io/evcc/pull/11232 mit behoben da wir dort einmalig auf die Daten warten.

closed 2023-12-24 by andig

VolkerK62, 2023-12-25:
Super, geht wieder.