evcc-io/evcc #11227: Battery Control for SOLAX Hybrid G4

ioniks, 2023-12-24:
Hi,

The new feature Battery control is a very good feature.

For solax inverters, the only parameter that can be changed is the battery output A power that can be limited.

Is it possible during fast charging or smart grid to limit the output power of the battery and return it normally to the end of charging or back to a normal mode?

This parameter can be modified with a value defined in the configuration for example? (and not updated all the time, because according to the returns the eprom has a predetermined maximum write capacity)

Thanks

premultiply, 2023-12-24:
Yes please provide all technical details like Modbus registers, documentation, command flow examples, ...

ioniks, 2023-12-24:
Command flow is just when charging at full speed or smart grid (full speed too).

Change battery_discharge_max_current to 5 or 10A (it's better if i can change it in yaml for exemple)
Charge at fast mode
When stop charging (vehicle disconnected or fully charged) or mode change the battery_discharge_max_current is restored (can be a setting in yaml too)

For modbus register :
https://gbc-solino.cz/wp-content/uploads/2022/07/Hybrid-X1X3-G4-ModbusTCPRTU-V3.21-English_0622-public-version.pdf

It's 0x0091 "Battery_DischargeMaxCurrent", recommended 25 A and can be 0 to 30A

Say if you need more information.

premultiply, 2023-12-24:
Looks these two registers may fit the requirements:

[image]

ioniks, 2023-12-24:
First is the inverter mode for backup or other. (So no)
The second is for force charge the battery with the grid.

So I don't think this is useful for discharge control of the battery.

premultiply, 2023-12-24:
Please no guessing. Did you try it?
Switching to backup mode should stop discharging the battery.

ioniks, 2023-12-24:
I'm not guessing for backup and self use mode, I'm always on backup mode.

Backup mode just keeps a minimum soc more important (can be set) of the battery and when no solar power, inverter keep eps active, in self use mode if battery is at or less min soc (and no solar power ) the inverter is on standby and don't backup if grid lost.

So if you are in manual mode with no force charge and discharge, the inverter go to standby (I think because no backup mode) and don't go to eps mode when grid is lost.

premultiply, 2023-12-24:
[image]

ioniks, 2023-12-24:
Ok I have re read and i don't know why I think this. 

(Or it's another setting like hot standby setting)

andig, 2023-12-25:
Can someone confirm https://github.com/evcc-io/evcc/issues/11227#issuecomment-1868543284?

andig, 2023-12-27:
Bitte mal testen

closed 2023-12-28 by andig

TobiP3005, 2024-01-01:
@andig getting an decoding error on starting, please have a look.
I did not change anything in the config. It does not work since the last update:

FATAL 2024/01/01 13:59:27 cannot create meter 'battery': cannot create meter 'template': cannot create meter 'custom': battery mode: 1 error(s) decoding:
* error decoding 'Timeout': time: invalid duration "<no value>"

```
meters:
...
- type: template
  template: solax
  id: 1
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: 8N1
  usage: battery
  modbus: rs485serial
  name: battery
```

ioniks, 2024-01-04:
I have tested the battery control activation and i have a error  in EVCC :
`Request failed with status code 400: API request failed http://XXXXXXXXX:7070/api/batterydischargecontrol/true`

And in network tab, the response is : 
`{"error":"modbus: exception '4' (server device failure), function '6'"}`

andig, 2024-01-04:
@ioniks could you please tun
evcc meter --log error,modbus:trace --battery-mode hold --battery-mode-wait 30s
Thanks!

docsnyder7, 2024-01-04:
@andig  is there interessed for other users with Solax Hybrid WR´s?  could you need more informations?

ioniks, 2024-01-04:
@andig 

Can i set this mode in Yaml ? (i have docker version of evcc)

andig, 2024-01-04:
It's unclear to me if Solax works. If not we'll need an issue with relevant details.

ioniks, 2024-01-04:
@andig 

I have set in yaml 
```
log: debug
levels:
  modbus: trace
```

This is the log when i set the battery control to On, same API error and nothing in log for modbus

```
[lp-1  ] DEBUG 2024/01/04 12:28:34 charge power: 0W
[site  ] DEBUG 2024/01/04 12:28:34 pv power: 2108W
[site  ] DEBUG 2024/01/04 12:28:35 battery soc: 46%
[site  ] DEBUG 2024/01/04 12:28:35 battery power: -1560W
[site  ] DEBUG 2024/01/04 12:28:35 grid power: -0W
[site  ] DEBUG 2024/01/04 12:28:35 battery has priority at soc 46% (< 95%)
[site  ] DEBUG 2024/01/04 12:28:35 site power: 150W
[lp-1  ] DEBUG 2024/01/04 12:28:35 charge total import: 3428.009kWh
[lp-1  ] DEBUG 2024/01/04 12:28:35 charger status: A
[site  ] DEBUG 2024/01/04 12:28:44 set battery discharge control: true
[site  ] DEBUG 2024/01/04 12:28:49 ----
[lp-1  ] DEBUG 2024/01/04 12:28:49 charge power: 0W
[site  ] DEBUG 2024/01/04 12:28:50 pv power: 2008W
[site  ] DEBUG 2024/01/04 12:28:50 battery soc: 46%
[site  ] DEBUG 2024/01/04 12:28:50 battery power: -1438W
[site  ] DEBUG 2024/01/04 12:28:50 grid power: -0W
[site  ] DEBUG 2024/01/04 12:28:50 battery has priority at soc 46% (< 95%)
[site  ] DEBUG 2024/01/04 12:28:50 site power: 150W
[lp-1  ] DEBUG 2024/01/04 12:28:50 charge total import: 3428.009kWh
[lp-1  ] DEBUG 2024/01/04 12:28:50 charger status: A
[site  ] DEBUG 2024/01/04 12:29:04 ----
```

andig, 2024-01-04:
Run on command line:

    evcc meter --log error,modbus:trace --battery-mode hold --battery-mode-wait 30s

ioniks, 2024-01-04:
```
evcc meter --log error,modbus:trace --battery-mode hold --battery-mode-wait 30s
[main  ] INFO 2024/01/04 13:09:25 evcc 0.123.5
[main  ] INFO 2024/01/04 13:09:25 using config file: /etc/evcc.yaml
[modbus] TRACE 2024/01/04 13:10:55 modbus: send 00 01 00 00 00 06 01 06 00 1f 00 02
[modbus] TRACE 2024/01/04 13:10:57 modbus: recv 00 01 00 00 00 03 01 86 04
[main  ] FATAL 2024/01/04 13:10:57 set battery mode: modbus: exception '4' (server device failure), function '6'
```

In Home Assistant when i set Manual mode (with solax modbus plugin) it revert to Backup mode (my previous mode), and when i set it in Solax app, it works (but when it's set Stop charge and discharge, it continue charging ...)

andig, 2024-01-04:
> In Home Assistant when i set Manual mode (with solax modbus plugin) it revert to Backup mode (my previous mode), and when i set it in Solax app, it works (but when it's set Stop charge and discharge, it continue charging ...)

I have no idea what that means. Zero.

That said: if anyone has a problem with Solax please open new issue with required information.

andig, 2024-01-04:
> ```
> evcc meter --log error,modbus:trace --battery-mode hold --battery-mode-wait 30s
> [main  ] INFO 2024/01/04 12:14:22 evcc 0.123.5
> [main  ] INFO 2024/01/04 12:14:22 using config file: /etc/evcc.yaml
> [modbus] TRACE 2024/01/04 12:15:22 modbus: send 00 01 00 00 00 06 01 06 00 1f 00 02
> [main  ] FATAL 2024/01/04 12:15:22 set battery mode: read tcp 192.168.178.57:35484->192.168.178.88:502: read: connection reset by peer
> ```
> 
> ich setze mal die Ausgabe aus meinem system ein..

TCP Connection weg- ob/was das mit einem potentiellen Modbusproblem zu tun hat ist unklar.

andig, 2025-03-25:
> TCP Connection weg- ob/was das mit einem potentiellen Modbusproblem zu tun hat ist unklar.

Dafür gibt es im aktuellen Nightly eine Anpassung.