evcc-io/evcc #11228: Min- Maxeinstellungen funktionieren nicht mehr

AUSMAUS, 2023-12-24:
### Describe the bug

[image]

### Steps to reproduce

1.x
2.x
3.x
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: https
  host: https://192.168.178.45
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 
interval: 10s # control cycle interval

sponsortoken:
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
#- name: my_grid
#  type: template
#  template: tibber-pulse
#  usage: grid
#  token: 
#- name: my_grid
#  type: template
#  template: powerfox-poweropti
#  usage: grid
#  user: 
#  password: 
- name: mymeter
  type: custom
  power:
    source: mqtt
    topic: Strom/Aktuell
- name: pv2
  type: custom
  power:
    source: mqtt
    topic: PV/Aktuell
#- name: pv2
#  type: custom
#  power:
#    source: http
#    uri: http://192.168.178.34/cm?cmnd=Status%208
#    jq: .StatusSNS.SML.Power_Import_W

chargers:
- type: template
  template: go-e
  host: 192.168.178.22
  name: wallbox3

vehicles:
- type: template
  template: opel
  title: Mokka-e
  icon: car
  user: 
  password: 
  vin: 
  capacity: 50
  phases: 1
  cache: 15m
  mode: off
#  minSoc: 20
#  targetSoc: 80
  minCurrent: 6
  maxCurrent: 20
  name: ev2

loadpoints:
- title: Garage
  charger: wallbox3
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 20
 # resetOnDisconnect: true

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2864
  feedin:
    type: fixed
    price: 0.192
#  co2:
#    type: grünstromindex
#    zip: 

site:
  title: evcc64
  meters:
    grid: mymeter
    pv:
    - pv2
  residualpower: 10

mqtt:
  broker: 192.168.178.62:1883
  topic: evcc 
  clientid: evcc
  user: 
  password:
```

### Log details

```text
Keine
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

evcc version 0.123.1

andig, 2023-12-24:
Es bleibt leider unklar was da nicht funktioniert

closed 2023-12-24 by andig

AUSMAUS, 2023-12-24:
> Es bleibt leider unklar was da nicht funktioniert
Mit den obig angezeigten Einstellungen wird nicht geladen. Bei Schnell wird mit 20 A geladen.
Bei PV und Min + PV mit 6 A

andig, 2023-12-24:
Dann brauchts bitte ein Debug Log aus dem das kenntlich wird.

AUSMAUS, 2023-12-24:
Wie geht das bei HomeAssistant ?
Und wenn noch einmal, eine Bugmeldung sofort geschlossen wird. Werde ich sofort mein Sponsering kündigen. So geht man mit einem Unterstützer nicht um. Dies ist schon unverschämt.

andig, 2023-12-24:
Diese Diskussion, noch dazu mit Drohung, bin ich nicht bereit zu führen. Im Issuetemplate steht explizit drin dass Issues ohne Logfile geschlossen werden.

Frohe Weihnachten.