evcc-io/evcc #11233: Use semantic versioning

florianmutter, 2023-12-24:
**Is your feature request related to a problem? Please describe.**
From looking at the version number when upgrading (e.g. with apt-get) it is not obvious if an update contains breaking changes.

**Describe the solution you'd like**
Use [semantic versioning](https://semver.org). This is used widely in the software industry. It gives clear rules what to expect for major, minor or patch update.

**Describe alternatives you've considered**
No alternative considered

**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2023-12-24:
We already do. 0.x are pre-release versions and expected to be breaking. Happy to welcome something to the team as RelEng to improve capabilities.

closed 2023-12-24 by andig

florianmutter, 2023-12-28:
What are the criteria to move from pre-releases to actual releases?