evcc-io/evcc #11242: OpenWB Charger logic error and detecting 2p on a 1p car (corsa)

atze09, 2023-12-25:
### Describe the bug

Mir ist heute beim laden folgendes aufgefallen:
1. Beim laden einen charger logic error bekomme (siehe log)
2. Mein Auto (Corsa 1-phasiger Lader) plötzlich als 2 phasig erkannt 
3. Das Auto mit 32A lädt, wenn ich beim Auto 32A eingestellt habe, obwohl ich beim Ladepunkt als max.20A eingestellt habe (Ist in der unten aufgeführten config nicht mehr enthalten, hier habe ich das Auto manuell "gedrosselt")
4. Wenn ich in der Gui als Ladelimit 16A einstelle (Kommt im Log 29.1A = 20A + 9.1A (-2093W @ 1p), jedoch wird weiter mit 4,2kW geladen.

Hoffe ihr kommt mich hier etwas aufschlauen, bzw. evtl. Fehler in der Config ausmerzen. 
Vielen Dank und frohe Weihnachten.

### Steps to reproduce

1.Start EVCC
2.Connect car and start charging via PV
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: c35b3729cfe9020d0aaac81f785228a9ae1ad6170550d501403ac0e198a62728

interval: 30s # control cycle interval

sponsortoken: 
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.17.3  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.17.3  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.17.3  
  port: 502  
  name: battery3
  capacity: 29

chargers:
- type: template
  template: go-e 
  host: 192.168.17.83  
  name: wallbox5
- name: openwb
  type: template
  template: openwb-2.0
  # Modbus TCP
[... 165 lines cut ...]
[site  ] DEBUG 2023/12/25 13:37:49 battery power: -3352W
[site  ] DEBUG 2023/12/25 13:37:49 grid power: -35W
[site  ] DEBUG 2023/12/25 13:37:49 site power: -3187W
[lp-1  ] DEBUG 2023/12/25 13:37:49 vehicle odometer: 15336km
[lp-1  ] DEBUG 2023/12/25 13:37:49 charge currents: [6.23 655 0.03]A
[lp-1  ] DEBUG 2023/12/25 13:37:49 charge total import: 2272.710kWh
[lp-1  ] DEBUG 2023/12/25 13:37:49 charger status: C
[lp-1  ] DEBUG 2023/12/25 13:37:49 soc estimated: 43.03% (vehicle: 43.00%)
[lp-1  ] DEBUG 2023/12/25 13:37:49 vehicle soc: 43%
[lp-1  ] DEBUG 2023/12/25 13:37:49 vehicle range: 80km
[lp-1  ] WARN 2023/12/25 13:37:49 charger logic error: disabled but charging
[lp-1  ] DEBUG 2023/12/25 13:37:49 set ev4 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2023/12/25 13:37:49 pv charge current: 13.9A = 0A + 13.9A (-3187W @ 1p)
[lp-1  ] DEBUG 2023/12/25 13:37:49 pv timer reset
[lp-1  ] DEBUG 2023/12/25 13:37:49 pv timer inactive
[lp-1  ] DEBUG 2023/12/25 13:37:49 max charge current: 13.9A
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.1

andig, 2023-12-25:
Laut Log 655A auf L2.  Bitte mal mqtt: trace und neues Log damit wir sehen, woher das kommt.

atze09, 2023-12-26:
> Laut Log 655A auf L2. Wohl eher unwahrscheinlich. Bitte mal mqtt: trace und neues Log damit wir sehen, woher das kommt.

@andig könntest du mir kurz erläutern wie ich ein "mqtt: trace" erstelle?

VolkerK62, 2023-12-26:
die Konfig entsprechend erweitern
```
log: debug
levels:
  cache: error
  mqtt: trace
```

atze09, 2023-12-27:
> die Konfig entsprechend erweitern
> 
> ```
> log: debug
> levels:
>   cache: error
>   mqtt: trace
> ```

Super danke! Weißt du zufällig auch noch wie ich aus home assistant ein längeres Log bekomme? So wie ich es recherchiert habe, ist das so leider nicht möglich.

andig, 2023-12-27:
Nein, ist es nicht. Ich mache hier mal zu- 655A aus L2 sind jdfls. die Ursache. Wo die herkommen bleibt unklar.

closed 2023-12-27 by andig

atze09, 2023-12-27:
@andig:
Ich melde mich mit dem log sobald ich das Auto mal wieder per PV in lade. Danke

atze09, 2023-12-30:
Heute mal wieder Sonne zum laden da, anbei die log "Schnipsel" da ich aus HA bis dato kein komplettes log bekomme.
2 Phase steht laut log immer noch bei 655A.
Wie gesagt Corsa wird mit 2 Phasen angezeigt, ist aber definitiv nur ein 1P Lader verbaut!
Hoffe Ihr könnt mir helfen.

[lp-2  ] ERROR 2023/12/30 14:23:02 charge rater: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[lp-2  ] ERROR 2023/12/30 14:23:05 charge total import: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[lp-2  ] ERROR 2023/12/30 14:23:08 charger: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[site  ] DEBUG 2023/12/30 14:23:15 ----
[lp-1  ] DEBUG 2023/12/30 14:23:15 charge power: 4316W
[lp-2  ] ERROR 2023/12/30 14:23:24 charge meter: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[site  ] DEBUG 2023/12/30 14:23:24 pv power: 6073W
[site  ] DEBUG 2023/12/30 14:23:24 battery soc: 83%
[site  ] DEBUG 2023/12/30 14:23:24 battery power: -1018W
[site  ] DEBUG 2023/12/30 14:23:24 grid power: 308W
[site  ] DEBUG 2023/12/30 14:23:24 site power: -510W
[lp-1  ] DEBUG 2023/12/30 14:23:24 charge currents: [19 655 0.1]A
[lp-1  ] DEBUG 2023/12/30 14:23:24 detected active phases: 2p
[lp-1  ] DEBUG 2023/12/30 14:23:24 charge total import: 2281.496kWh
[lp-1  ] DEBUG 2023/12/30 14:23:24 charger status: C
[lp-1  ] DEBUG 2023/12/30 14:23:24 soc estimated: 55.42% (vehicle: 55.00%)
[lp-1  ] DEBUG 2023/12/30 14:23:24 vehicle soc: 55%
[lp-1  ] DEBUG 2023/12/30 14:23:24 vehicle range: 118km
[lp-1  ] WARN 2023/12/30 14:23:24 charger logic error: disabled but charging
[lp-1  ] DEBUG 2023/12/30 14:23:24 pv timer elapse
[lp-1  ] DEBUG 2023/12/30 14:23:24 pv timer inactive
[lp-1  ] DEBUG 2023/12/30 14:23:24 set ev4 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] WARN 2023/12/30 14:23:24 ignoring inconsistent phases: 1p < 2p observed active
[lp-1  ] DEBUG 2023/12/30 14:23:24 pv charge current: 22.2A = 20A + 2.22A (-510W @ 1p)
[lp-1  ] DEBUG 2023/12/30 14:23:24 pv timer reset
[lp-1  ] DEBUG 2023/12/30 14:23:24 pv timer inactive
[site  ] DEBUG 2023/12/30 14:23:45 ----
[lp-1  ] DEBUG 2023/12/30 14:23:45 charge power: 4133W
[lp-2  ] ERROR 2023/12/30 14:23:54 charge meter: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[site  ] DEBUG 2023/12/30 14:23:54 pv power: 6061W
[site  ] DEBUG 2023/12/30 14:23:54 battery soc: 83%
[site  ] DEBUG 2023/12/30 14:23:54 battery power: -888W
[site  ] DEBUG 2023/12/30 14:23:54 grid power: -237W
[site  ] DEBUG 2023/12/30 14:23:54 site power: -925W
[lp-2  ] ERROR 2023/12/30 14:23:57 charge meter: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[lp-2  ] ERROR 2023/12/30 14:24:00 charge meter: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[lp-2  ] ERROR 2023/12/30 14:24:03 charge rater: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[lp-2  ] ERROR 2023/12/30 14:24:06 charge total import: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[lp-2  ] ERROR 2023/12/30 14:24:10 charger: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[site  ] DEBUG 2023/12/30 14:24:15 ----
[lp-1  ] DEBUG 2023/12/30 14:24:15 charge power: 4140W
[lp-2  ] ERROR 2023/12/30 14:24:24 charge meter: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[site  ] DEBUG 2023/12/30 14:24:24 pv power: 5817W
[site  ] DEBUG 2023/12/30 14:24:24 battery soc: 83%
[site  ] DEBUG 2023/12/30 14:24:24 battery power: -888W
[site  ] DEBUG 2023/12/30 14:24:24 grid power: 142W
[site  ] DEBUG 2023/12/30 14:24:24 site power: -546W
[lp-1  ] DEBUG 2023/12/30 14:24:24 charge currents: [19.4 655 0.1]A
[lp-1  ] DEBUG 2023/12/30 14:24:24 detected active phases: 2p
[lp-1  ] DEBUG 2023/12/30 14:24:24 charge total import: 2281.567kWh
[lp-1  ] DEBUG 2023/12/30 14:24:24 charger status: C
[lp-1  ] DEBUG 2023/12/30 14:24:24 soc estimated: 55.56% (vehicle: 55.00%)
[lp-1  ] DEBUG 2023/12/30 14:24:24 vehicle soc: 56%
[lp-1  ] DEBUG 2023/12/30 14:24:24 vehicle range: 118km
[lp-1  ] WARN 2023/12/30 14:24:24 charger logic error: disabled but charging
[lp-1  ] DEBUG 2023/12/30 14:24:24 pv timer elapse
[lp-1  ] DEBUG 2023/12/30 14:24:24 pv timer inactive
[lp-1  ] DEBUG 2023/12/30 14:24:24 set ev4 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] WARN 2023/12/30 14:24:24 ignoring inconsistent phases: 1p < 2p observed active
[lp-1  ] DEBUG 2023/12/30 14:24:24 pv charge current: 22.4A = 20A + 2.37A (-546W @ 1p)
[lp-1  ] DEBUG 2023/12/30 14:24:24 pv timer reset
[lp-1  ] DEBUG 2023/12/30 14:24:24 pv timer inactive
[site  ] DEBUG 2023/12/30 14:24:45 ----
[lp-1  ] DEBUG 2023/12/30 14:24:45 charge power: 4434W
[lp-2  ] ERROR 2023/12/30 14:24:54 charge meter: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[... 313 lines cut ...]
[lp-2  ] ERROR 2023/12/30 14:27:05 charge total import: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[lp-2  ] ERROR 2023/12/30 14:27:08 charger: Get "http://192.168.17.83/status": dial tcp 192.168.17.83:80: connect: no route to host
[site  ] DEBUG 2023/12/30 14:27:15 ----
[lp-1  ] DEBUG 2023/12/30 14:27:15 charge power: 4331W
[... cut ...]
[lp-1  ] WARN 2023/12/30 14:27:24 charger logic error: disabled but charging
[lp-1  ] DEBUG 2023/12/30 14:27:24 pv timer elapse
[lp-1  ] DEBUG 2023/12/30 14:27:24 pv timer inactive
[lp-1  ] DEBUG 2023/12/30 14:27:24 set ev4 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] WARN 2023/12/30 14:27:24 ignoring inconsistent phases: 1p < 2p observed active
[lp-1  ] DEBUG 2023/12/30 14:27:24 pv charge current: 21.1A = 20A + 1.07A (-245W @ 1p)
[lp-1  ] DEBUG 2023/12/30 14:27:24 pv timer reset
[lp-1  ] DEBUG 2023/12/30 14:27:24 pv timer inactive

andig, 2023-12-30:
Am Problem hat sich nichts geändert. Deine OpenWB zeigt 655A auf L2. Lösch die doch bitte einfach mal raus, dann verschwindet auch das Problem. Und falls sie wieder auftauchen geht die Fragen vmtl. an OpenWB wo das her kommt...

atze09, 2023-12-30:
> Am Problem hat sich nichts geändert. Deine OpenWB zeigt 655A auf L2. Lösch die doch bitte einfach mal raus, dann verschwindet auch das Problem. Und falls sie wieder auftauchen geht die Fragen vmtl. an OpenWB wo das her kommt...
> 
> Ansonsten zeigt Dein Logfile diverse Fehler die einer problemlosen Funktion sicher auch nicht zuträglich sind.

Ok super danke, das versuche ich.
Die anderen Fehler sind entstanden, da mein anderer LP nicht vom WLAN erreichbar war, das sollte hier aber keine Rolle spielen, da das OpenWb Problem letztes mal auch da war als LP verfügbar war..

atze09, 2024-01-01:
> Am Problem hat sich nichts geändert. Deine OpenWB zeigt 655A auf L2. Lösch die doch bitte einfach mal raus, dann verschwindet auch das Problem. Und falls sie wieder auftauchen geht die Fragen vmtl. [...]

Erstmal frohes neues Jahr @andig 
Also ich habe mir das Ganze nochmal angesehen.
Die OpenWb meldet im Mqtt die richtigen Ladeströme an:
[image]

Jedoch in der EVCC MQTT steht hier auf der zweiten Phase die 655A. Also liegt der Fehler doch innerhalb EVCC oder?
[image]

Ich weiß auch nicht woher EVCC die Information über die zweite Phase nimmt, das Auto lädt ja nur mit einer Phase.

Welche detaillierteren Infos kann ich noch zum bugfix liefern?

andig, 2024-01-01:
> Jedoch in der EVCC MQTT steht hier auf der zweiten Phase die 655A.

Hast Du die denn aus der OpenWB rausgelöscht wie vorgeschlagen?

> Also liegt der Fehler doch innerhalb EVCC oder?

Nein. Mach doch bitte mal ein mqtt trace log damit wir sehen wo das her kommt. Aus evcc sicher nicht...