evcc-io/evcc #11243: API broken with christmas release - needs to be updated

creuter27, 2023-12-25:
I have some automations to automatically set the target time of the charge planner. They rely on the evcc api. Since the Christmas  update the automation isn't working anymore - most likely due to the api changes mentioned in the release notes. Unfortunately, the api documentation doesn't show any changes. when will it be updated? Any other way to figure out  how to set the target time now?

andig, 2023-12-25:
See release notes. We‘ll update the api asap.

Newan, 2023-12-29:
Ja die API wurde generell nun anders aufgebaut. Da brauch ich echt Zeit für. Den das bedeutet das ich den Adapter Grundlgend neu machen darf.
Aber werde ich machen!

closed 2024-02-10 by andig