evcc-io/evcc #11247: openWB series2: Wrong identifier encoding

premultiply, 2023-12-25:
### Discussed in https://github.com/evcc-io/evcc/discussions/11246

<div type='discussions-op-text'>

<sup>Originally posted by **Daikon2022** December 25, 2023</sup>
Hallo,

ich möchte gerne auf EVCC wechseln in Kombination mit einer openWB series2 mit aktueller 2.1.2 Software. Für mich ist es wichtig, dass ich per RFID Schlüsselanhänger unterschiedliche Autoprofile vorgeben kann. Leider habe ich das Problem, dass die erkannten Identifier extrem kryptisch sind bzw. nicht korrekt ausgelesen werden. Hier mal ein Beispiel:

Identifier:     〰〳㤲ㄱ㌳

Ich habe zum Auslesen der Identifier folgende Anleitung verwendet: https://docs.evcc.io/docs/guides/vehicles#erkennung-%C3%BCber-rfid-karten-und-andere-token

Gibt es mit der openWB und dem Softwarestand Probleme?</div>

Ist: Byte-Array als String,
Soll: Byte-Array als Hex-String

andig, 2023-12-25:
Gibts ein Log?

Daikon2022, 2023-12-25:
Ich weiß noch ob das hier weiterhilft:

> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 01 00 00 00 06 01 04 27 b0 00 0a
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 01 00 00 00 17 01 04 14 30 30 30 38 35 37 32 36 34 31 00 00 00 00 00 00 00 00 00 00
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 02 00 00 00 06 01 04 27 74 00 02
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 02 00 00 00 07 01 04 04 00 00 00 00
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 03 00 00 00 06 01 04 27 76 00 02
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 03 00 00 00 07 01 04 04 00 33 63 28
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 04 00 00 00 06 01 04 27 7b 00 03
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 04 00 00 00 09 01 04 06 00 00 00 00 00 00
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 05 00 00 00 06 01 04 27 78 00 03
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 05 00 00 00 09 01 04 06 5a 82 5a 63 59 6e
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 06 00 00 00 06 01 04 27 83 00 01
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 06 00 00 00 05 01 04 02 00 00
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 07 00 00 00 06 01 04 27 82 00 01
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 07 00 00 00 05 01 04 02 00 00
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 08 00 00 00 06 01 04 27 84 00 01
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 08 00 00 00 05 01 04 02 00 00
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: send 00 09 00 00 00 06 01 04 27 b0 00 0a
> [openwb-2.0] TRACE 2023/12/25 20:13:30 modbus: recv 00 09 00 00 00 17 01 04 14 30 30 30 38 35 37 32 36 34 31 00 00 00 00 00 00 00 00 00 00
> Power:          0W
> Energy:         3367.7kWh
> Current L1..L3: 0A 0A 0A
> Voltage L1..L3: 232V 231V 229V
> Charge status:  A
> Enabled:        false
> Identifier:     〰〸㔷㈶㐱

PR #11261 (2023-12-26): OpenWB2: fix identifier
Fixes #11247
closed 2023-12-26 by andig