evcc-io/evcc #11252: Victron Batteriesteuerung - does not retain

henne49, 2023-12-25:
### Describe the bug

Vielen Dank für das tolle Update. 

Leider wird bei meinem Victron ESS/Battery nicht Minimum SOC der Victron übernommen, sondern von evcc auf 20 also wohl das Default gesetzt.

So sieht das Victron SOC aus, im Winter bei 70%
[image]

### Steps to reproduce

1. Restart evcc
2. Batteriesteuerung ist aktiviert
3. Minimum SOC ist nicht 20 in Victron
...

### Configuration details

```yaml
meters:
  - name: Battery-Victron
    type: template
    template: victron-energy
    usage: battery
    host: X.X.X.X # IP-Adresse oder Hostname
    port: 502 # Port # Optional
    #minsoc: 70 # Min SOC ist nicht gesetzt
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/12/25 22:53:52 evcc 0.123.1
[main  ] INFO 2023/12/25 22:53:52 using config file: /config/evcc.yaml
[main  ] INFO 2023/12/25 22:53:52 starting ui and api at :7070
[db    ] INFO 2023/12/25 22:53:53 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2023/12/25 22:53:53 connecting evcc-503807776 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/12/25 22:53:53 tcp://localhost:1883 connected
[proxy-5200] DEBUG 2023/12/25 22:53:53 modbus proxy for 172.16.4.172:502 listening at :5200
[lp-1  ] DEBUG 2023/12/25 22:53:57 charge total import: 4494.460kWh
[site  ] INFO 2023/12/25 22:53:59 site config:
[site  ] INFO 2023/12/25 22:53:59   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/12/25 22:53:59     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/12/25 22:53:59     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/12/25 22:53:59     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/12/25 22:53:59   vehicles:
[site  ] INFO 2023/12/25 22:53:59     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2023/12/25 22:53:59     vehicle 2: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/12/25 22:53:59 loadpoint 1:
[lp-1  ] INFO 2023/12/25 22:53:59   mode:        pv
[lp-1  ] INFO 2023/12/25 22:53:59   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/12/25 22:53:59   meters:      charge ✓
[lp-1  ] INFO 2023/12/25 22:53:59     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2023/12/25 22:53:59 set buffer soc: 90
[site  ] DEBUG 2023/12/25 22:53:59 set smart cost limit: 70
[site  ] DEBUG 2023/12/25 22:53:59 set priority soc: 70
[lp-1  ] DEBUG 2023/12/25 22:53:59 phase timer inactive
[lp-1  ] DEBUG 2023/12/25 22:53:59 pv timer inactive
[lp-1  ] DEBUG 2023/12/25 22:53:59 guard timer inactive
[site  ] WARN 2023/12/25 22:53:59 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/12/25 22:53:59 ----
[lp-1  ] DEBUG 2023/12/25 22:53:59 charge power: 0W
[site  ] DEBUG 2023/12/25 22:53:59 pv power: 0W
[site  ] DEBUG 2023/12/25 22:53:59 battery soc: 69%
[site  ] DEBUG 2023/12/25 22:53:59 battery power: -14W
[site  ] DEBUG 2023/12/25 22:53:59 grid power: 484W
[site  ] DEBUG 2023/12/25 22:53:59 grid powers: [25 234 224]W
[site  ] DEBUG 2023/12/25 22:53:59 grid currents: [0.514 1.73 1.41]A
[site  ] DEBUG 2023/12/25 22:53:59 battery has priority at soc 69% (< 70%)
[site  ] DEBUG 2023/12/25 22:53:59 site power: 584W
[lp-1  ] DEBUG 2023/12/25 22:53:59 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/25 22:53:59 charge total import: 4494.460kWh
[lp-1  ] DEBUG 2023/12/25 22:53:59 charger status: B
[lp-1  ] DEBUG 2023/12/25 22:53:59 charge start energy: 4.49e+03kWh
[lp-1  ] INFO 2023/12/25 22:53:59 car connected
[lp-1  ] DEBUG 2023/12/25 22:53:59 vehicle api refresh
[lp-1  ] DEBUG 2023/12/25 22:53:59 pv timer elapse
[lp-1  ] DEBUG 2023/12/25 22:53:59 pv timer inactive
[lp-1  ] DEBUG 2023/12/25 22:53:59 charger: guard elapse
[lp-1  ] DEBUG 2023/12/25 22:53:59 guard timer inactive
[lp-1  ] DEBUG 2023/12/25 22:53:59 charge total import: 4494.460kWh
[lp-1  ] DEBUG 2023/12/25 22:53:59 charger vehicle id: 0462c662
[main  ] DEBUG 2023/12/25 22:54:00 vehicle status: A (Skoda Enyaq iV80)
[main  ] DEBUG 2023/12/25 22:54:00 vehicle status: B (Peugeot Rifter)
[lp-1  ] INFO 2023/12/25 22:54:00 vehicle updated: unknown -> Peugeot Rifter
[lp-1  ] DEBUG 2023/12/25 22:54:00 set charge mode: pv
[lp-1  ] DEBUG 2023/12/25 22:54:00 vehicle status: B
[lp-1  ] DEBUG 2023/12/25 22:54:00 vehicle soc: 80%
[lp-1  ] DEBUG 2023/12/25 22:54:00 vehicle range: 196km
[telegram] DEBUG 2023/12/25 22:54:00 sending to ***
[lp-1  ] DEBUG 2023/12/25 22:54:00 set rifter plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2023/12/25 22:54:00 limitSoc reached: 80.0% > 80%
[site  ] DEBUG 2023/12/25 22:54:00 set battery mode: normal
[site  ] DEBUG 2023/12/25 22:54:09 ----
[lp-1  ] DEBUG 2023/12/25 22:54:09 charge power: 0W
[site  ] DEBUG 2023/12/25 22:54:09 pv power: 0W
[site  ] DEBUG 2023/12/25 22:54:09 battery soc: 69%
[site  ] DEBUG 2023/12/25 22:54:09 battery power: 328W
[site  ] DEBUG 2023/12/25 22:54:09 grid power: -45W
[site  ] DEBUG 2023/12/25 22:54:09 grid powers: [25 -289 219]W
[site  ] DEBUG 2023/12/25 22:54:09 grid currents: [0.514 -1.7 1.37]A
[site  ] DEBUG 2023/12/25 22:54:09 site power: 383W
[lp-1  ] DEBUG 2023/12/25 22:54:09 vehicle odometer: 14295km
[lp-1  ] DEBUG 2023/12/25 22:54:09 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/25 22:54:09 charge total import: 4494.460kWh
[lp-1  ] DEBUG 2023/12/25 22:54:09 charger status: B
[lp-1  ] DEBUG 2023/12/25 22:54:09 set rifter plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2023/12/25 22:54:09 limitSoc reached: 80.0% > 80%
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

123.1

henne49, 2023-12-25:
minsoc: 70 unter Batterie behebt das Problem - allerdings keine Lösung für den Sommer. Hier würde ich sehen, dass Victron der Herrscher über den MinSOC ist, so dass ich den dort einstellen kann und Victron mit diesem SOC arbeitet.

andig, 2023-12-25:
Es braucht ein trace log für Victron um das zu beurteilen. Battery control MUSS aus sein.

closed 2023-12-25 by andig

henne49, 2023-12-25:
Es passiert eben auch mit Battery Control AUS. Trace Log aus evcc für die Victron Battery nehme ich? Besorge ich gern

andig, 2023-12-26:
Mir würde es schon reichen, diese Zeile hier:

> [site  ] DEBUG 2023/12/25 23:07:18 set battery mode: normal

Zu sehen wenn `batteryDischargeControl` deaktiviert ist. Im Code kann ich nix finden was das verursachen würde.

henne49, 2023-12-27:
In 0.123.3 habe ich auch mit deaktiviertem minsoc unter Battery nun keinen Setzen auf des SOC auf 20 mehr. Ich beobachte das weiter, habt ihr etwa geändert?

die gewünschte Zeile kann ich auch nicht mehr sehen. 

Hier ein Test vom Umschalten während evcc läuft

```
[site  ] DEBUG 2023/12/27 07:02:44 pv power: 0W
[site  ] DEBUG 2023/12/27 07:02:44 battery soc: 67%
[site  ] DEBUG 2023/12/27 07:02:44 battery power: 0W
[site  ] DEBUG 2023/12/27 07:02:44 grid power: 3599W
[site  ] DEBUG 2023/12/27 07:02:44 grid powers: [753 1889 956]W
[site  ] DEBUG 2023/12/27 07:02:44 grid currents: [4.14 8.45 4.88]A
[site  ] DEBUG 2023/12/27 07:02:44 site power: 3699W
[site  ] DEBUG 2023/12/27 07:03:13 set battery discharge control: true
[site  ] DEBUG 2023/12/27 07:03:13 set battery mode: normal
[site  ] DEBUG 2023/12/27 07:03:14 ----
[site  ] DEBUG 2023/12/27 07:03:14 pv power: 0W
[site  ] DEBUG 2023/12/27 07:03:14 battery soc: 67%
[site  ] DEBUG 2023/12/27 07:03:14 battery power: 0W
[site  ] DEBUG 2023/12/27 07:03:14 grid power: 2839W
[site  ] DEBUG 2023/12/27 07:03:14 grid powers: [791 1102 946]W
[site  ] DEBUG 2023/12/27 07:03:14 grid currents: [4.24 5.11 4.86]A
[site  ] DEBUG 2023/12/27 07:03:14 site power: 2939W
[site  ] DEBUG 2023/12/27 07:03:29 set battery discharge control: false
```

andig, 2023-12-27:
Nein…

Rodenberger, 2024-01-06:
Also es funktioniert immer noch nicht, oder habe ich was übersehen???

andig, 2024-01-06:
Ja.