evcc-io/evcc #11277: Oberflöche startet nicht mehr.

Juppme, 2023-12-27:
Hallo, seit zwei oder 3 Tagen geht evcc unter Home Assistent nicht mehr. 
Lief seit Mai fehlerfrei.
an der config wurde nicht verändert. 

habe jetzt mal den Token erneuert leider immer noch gleicher Fehler. 

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/12/27 11:21:47 evcc 0.123.2
[main  ] INFO 2023/12/27 11:21:47 using config file: /config/evcc.yaml
[main  ] INFO 2023/12/27 11:21:47 starting ui and api at :7070
[lp-1  ] WARN 2023/12/27 11:21:59 phases not configured, assuming 3p
[main  ] FATAL 2023/12/27 11:21:59 failed configuring site: 1 error(s) decoding:

* '' has invalid keys: buffersoc, bufferstartsoc, prioritysoc, smartcostlimit
[main  ] FATAL 2023/12/27 11:21:59 will attempt restart in: 5m0s

bitte um Hilfe

iz011235813, 2023-12-27:
Neuste Version?

Gab breaking changes, du musst die Werte auskommentieren, dann geht es (wahrscheinlich) wieder.

https://github.com/evcc-io/evcc/discussions/10978

closed 2023-12-27 by andig

Juppme, 2023-12-27:
Danke, das war es