evcc-io/evcc #11278: Fehler bei Plan ohne verbundenen Tesla

0163deralex, 2023-12-27:
### Describe the bug

Wenn der Wagen nicht angeschlossen ist und ich den Plan öffne, erscheint ein rotes ⚠ Warndreieck mit folgendem Inhalt;Request failed with status code 400: API request failed https://xxxxxxxxxxx.de/api/hassio_ingress/xxxxxxxxxxxxxxxxxxxxxx20Auhx7vnrK4Os/api/loadpoints/1/plan/preview/energy/10/2023-12-27T11:45:00.000Z

### Steps to reproduce

1. Evcc öffnen 
2. Loadpoint 1 auswählen und auf Plan klicken 
3. Ohne Auto oben genannte Fehlermeldung 
4. Mit Auto keine Fehlermeldung 
5. Nur beim Tesla Loadpoint 
[image]

### Configuration details

```yaml
*
```

### Log details

```text
[site  ] DEBUG 2023/12/27 11:31:59 ----
[lp-1  ] DEBUG 2023/12/27 11:31:59 charge power: 4W
[lp-2  ] DEBUG 2023/12/27 11:31:59 charge power: 0W
[site  ] DEBUG 2023/12/27 11:31:59 pv power: 783W
[site  ] DEBUG 2023/12/27 11:31:59 grid power: -610W
[site  ] DEBUG 2023/12/27 11:31:59 grid powers: [-203 -208 -199]W
[site  ] DEBUG 2023/12/27 11:31:59 grid currents: [-0.953 -1.01 -0.921]A
[site  ] DEBUG 2023/12/27 11:31:59 site power: -610W
[lp-1  ] DEBUG 2023/12/27 11:31:59 charge voltages: [232 232 233]V
[lp-1  ] DEBUG 2023/12/27 11:31:59 detected connected phases: 3p
[lp-1  ] DEBUG 2023/12/27 11:31:59 charge currents: [0.049 0.018 0.018]A
[lp-1  ] DEBUG 2023/12/27 11:31:59 charge total import: 2300.500kWh
[lp-1  ] DEBUG 2023/12/27 11:31:59 charger status: A
[lp-1  ] DEBUG 2023/12/27 11:31:59 set Valentino plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2023/12/27 11:32:09 ----
[lp-1  ] DEBUG 2023/12/27 11:32:09 charge power: 4W
[lp-2  ] DEBUG 2023/12/27 11:32:09 charge power: 0W
[site  ] DEBUG 2023/12/27 11:32:09 pv power: 775W
[site  ] DEBUG 2023/12/27 11:32:09 grid power: -617W
[site  ] DEBUG 2023/12/27 11:32:09 grid powers: [-203 -210 -203]W
[site  ] DEBUG 2023/12/27 11:32:09 grid currents: [-0.953 -1.02 -0.936]A
[site  ] DEBUG 2023/12/27 11:32:09 site power: -617W
[lp-2  ] DEBUG 2023/12/27 11:32:10 charger status: B
[lp-2  ] DEBUG 2023/12/27 11:32:10 pv charge current: 2.68A = 0A + 2.68A (-617W @ 1p)
[lp-2  ] DEBUG 2023/12/27 11:32:10 max charge current: 2.68A
[site  ] DEBUG 2023/12/27 11:32:19 ----
[lp-1  ] DEBUG 2023/12/27 11:32:19 charge power: 4W
[lp-2  ] DEBUG 2023/12/27 11:32:19 charge power: 0W
[site  ] DEBUG 2023/12/27 11:32:19 pv power: 787W
[site  ] DEBUG 2023/12/27 11:32:19 grid power: -620W
[site  ] DEBUG 2023/12/27 11:32:19 grid powers: [-206 -212 -202]W
[site  ] DEBUG 2023/12/27 11:32:19 grid currents: [-0.965 -1.02 -0.932]A
[site  ] DEBUG 2023/12/27 11:32:19 site power: -620W
[lp-1  ] DEBUG 2023/12/27 11:32:19 charge voltages: [232 231 233]V
[lp-1  ] DEBUG 2023/12/27 11:32:19 detected connected phases: 3p
[lp-1  ] DEBUG 2023/12/27 11:32:19 charge currents: [0.049 0.018 0.018]A
[lp-1  ] DEBUG 2023/12/27 11:32:19 charge total import: 2300.500kWh
[lp-1  ] DEBUG 2023/12/27 11:32:19 charger status: A
[lp-1  ] DEBUG 2023/12/27 11:32:19 set Valentino plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2023/12/27 11:32:29 ----
[lp-1  ] DEBUG 2023/12/27 11:32:29 charge power: 4W
[lp-2  ] DEBUG 2023/12/27 11:32:29 charge power: 0W
[site  ] DEBUG 2023/12/27 11:32:29 pv power: 791W
[site  ] DEBUG 2023/12/27 11:32:29 grid power: -625W
[site  ] DEBUG 2023/12/27 11:32:29 grid powers: [-205 -214 -206]W
[site  ] DEBUG 2023/12/27 11:32:29 grid currents: [-0.96 -1.02 -0.947]A
[site  ] DEBUG 2023/12/27 11:32:29 site power: -625W
[lp-2  ] DEBUG 2023/12/27 11:32:30 charger status: B
[lp-2  ] DEBUG 2023/12/27 11:32:30 pv charge current: 2.72A = 0A + 2.72A (-625W @ 1p)
[lp-2  ] DEBUG 2023/12/27 11:32:30 max charge current: 2.72A
[site  ] DEBUG 2023/12/27 11:32:39 ----
[lp-1  ] DEBUG 2023/12/27 11:32:39 charge power: 3W
[lp-2  ] DEBUG 2023/12/27 11:32:39 charge power: 0W
[site  ] DEBUG 2023/12/27 11:32:39 pv power: 800W
[site  ] DEBUG 2023/12/27 11:32:39 grid power: -635W
[site  ] DEBUG 2023/12/27 11:32:39 grid powers: [-210 -216 -209]W
[site  ] DEBUG 2023/12/27 11:32:39 grid currents: [-0.985 -1.04 -0.96]A
[site  ] DEBUG 2023/12/27 11:32:39 site power: -635W
[lp-1  ] DEBUG 2023/12/27 11:32:39 charge voltages: [233 232 233]V
[lp-1  ] DEBUG 2023/12/27 11:32:39 detected connected phases: 3p
[lp-1  ] DEBUG 2023/12/27 11:32:39 charge currents: [0.049 0.018 0.018]A
[lp-1  ] DEBUG 2023/12/27 11:32:39 charge total import: 2300.500kWh
[lp-1  ] DEBUG 2023/12/27 11:32:39 charger status: A
[lp-1  ] DEBUG 2023/12/27 11:32:39 set Valentino plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2023/12/27 11:32:49 ----
[lp-1  ] DEBUG 2023/12/27 11:32:49 charge power: 4W
[lp-2  ] DEBUG 2023/12/27 11:32:49 charge power: 0W
[site  ] DEBUG 2023/12/27 11:32:49 pv power: 806W
[site  ] DEBUG 2023/12/27 11:32:49 grid power: -643W
[site  ] DEBUG 2023/12/27 11:32:49 grid powers: [-213 -218 -211]W
[site  ] DEBUG 2023/12/27 11:32:49 grid currents: [-0.993 -1.05 -0.97]A
[site  ] DEBUG 2023/12/27 11:32:49 site power: -643W
[lp-2  ] DEBUG 2023/12/27 11:32:50 charger status: B
[lp-2  ] DEBUG 2023/12/27 11:32:50 pv charge current: 2.79A = 0A + 2.79A (-643W @ 1p)
[lp-2  ] DEBUG 2023/12/27 11:32:50 max charge current: 2.79A
[site  ] DEBUG 2023/12/27 11:32:59 ----
[lp-1  ] DEBUG 2023/12/27 11:32:59 charge power: 4W
[lp-2  ] DEBUG 2023/12/27 11:32:59 charge power: 0W
[site  ] DEBUG 2023/12/27 11:32:59 pv power: 814W
[site  ] DEBUG 2023/12/27 11:32:59 grid power: -643W
[site  ] DEBUG 2023/12/27 11:32:59 grid powers: [-214 -219 -210]W
[site  ] DEBUG 2023/12/27 11:32:59 grid currents: [-0.995 -1.05 -0.966]A
[site  ] DEBUG 2023/12/27 11:32:59 site power: -643W
[lp-1  ] DEBUG 2023/12/27 11:32:59 charge voltages: [233 232 233]V
[lp-1  ] DEBUG 2023/12/27 11:32:59 detected connected phases: 3p
[lp-1  ] DEBUG 2023/12/27 11:32:59 charge currents: [0.049 0.018 0.018]A
[lp-1  ] DEBUG 2023/12/27 11:32:59 charge total import: 2300.500kWh
[lp-1  ] DEBUG 2023/12/27 11:32:59 charger status: A
[lp-1  ] DEBUG 2023/12/27 11:32:59 set Valentino plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2023/12/27 11:33:09 ----
[lp-1  ] DEBUG 2023/12/27 11:33:09 charge power: 4W
[lp-2  ] DEBUG 2023/12/27 11:33:09 charge power: 0W
[site  ] DEBUG 2023/12/27 11:33:09 pv power: 803W
[site  ] DEBUG 2023/12/27 11:33:09 grid power: -636W
[site  ] DEBUG 2023/12/27 11:33:09 grid powers: [-211 -217 -209]W
[site  ] DEBUG 2023/12/27 11:33:09 grid currents: [-0.982 -1.04 -0.96]A
[site  ] DEBUG 2023/12/27 11:33:09 site power: -636W
[lp-2  ] DEBUG 2023/12/27 11:33:10 charger status: B
[lp-2  ] DEBUG 2023/12/27 11:33:10 pv charge current: 2.77A = 0A + 2.77A (-636W @ 1p)
[lp-2  ] DEBUG 2023/12/27 11:33:10 max charge current: 2.77A
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

*

andig, 2023-12-27:
@0163deralex kannst Du in der Browserkonsole mal in den Request rein schauen? Gibts da eine konkrete Fehlermeldung? Sowas wie:

    {"error":"energy planning not available for vehicles with known soc and capacity"}

Ich kann das provozieren wenn ein Fahrzeug mit Soc angeschlossen ist:

    curl http://localhost:7070/api/loadpoints/1/plan/preview/energy/80/2023-12-30T18:00:00.000Z

Gelingt mir aber nicht, wenn kein Auto angeschlossen oder Gastfahrzeug. Wir brauchen also konkrete Steps to Reproduce. in Deinem Logfile oben *ist* der Tesla übrigens angeschlossen:

> charger status: B

Passt also nicht zur Fehlerbeschreibung :(

0163deralex, 2023-12-27:
Wie finde ich die Browserkonsole?

Loadpoint 2 ist B, loadpoint 1 ist doch A? Der Tesla ist definitiv nicht angeschlossen an loadpoint 1.

StefanSchoof, 2023-12-27:
https://www.google.com/search?q=how+to+open+browser+console ?

0163deralex, 2023-12-27:
ok, die Konsole haben ich gefunden und folgende Errors entdeckt (ist aus 2 mal öffnen entstanden).
[image]
[image]

andig, 2023-12-27:
Ähh- schaut bitte ins Netzwerktab auf den Request ;)

0163deralex, 2023-12-27:
[image]

andig, 2023-12-27:
> schaut bitte ins Netzwerktab auf den Request ;)

andig, 2023-12-27:
@0163deralex bitte nicht irgendwas posten- das hilft nix.

0163deralex, 2023-12-27:
ich habe doch nur diese Error Meldung aus der Konsole gepostet :/

andig, 2023-12-27:
Die nutzt uns aber nichts! Es braucht den Netzwerkrequest aus dem Network Tab, wie schon 2x geschrieben:

> schaut bitte ins Netzwerktab auf den Request ;)

0163deralex, 2023-12-27:
Sorry, aber ich habe keine Ahnung was genau...Das vielleicht?
[image]

andig, 2023-12-27:
Bitte noch ein Blick auf Response und die Ausgabe von /api/state (kein Screenshot bitte).

0163deralex, 2023-12-27:
Hoffe, das sind die richtigen Zeilen:

Response:
`{
    "error": "energy planning not available for vehicles with known soc and capacity"
}`

api/state

`{
    "result": {
        "auth": {
            "vehicles": {}
        },
        "batteryConfigured": false,
        "batteryDischargeControl": false,
        "batteryMode": "unknown",
        "bufferSoc": 0,
        "bufferStartSoc": 0,
        "currency": "EUR",
        "greenShareHome": 0,
        "greenShareLoadpoints": 0,
        "gridConfigured": true,
        "gridCurrents": [
            0.601,
            1.141,
            0.966
        ],
        "gridEnergy": 1308.5554,
        "gridPower": 393.8,
        "gridPowers": [
            78.80000000000001,
            184.3,
            130.70000000000002
        ],
        "homePower": 389.8,
        "loadpoints": [
            {
                "chargeCurrent": 0,
                "chargeCurrents": [
                    0.049,
                    0.018,
                    0.018
                ],
                "chargeDuration": 159927000000000,
                "chargePower": 4,
                "chargeTotalImport": 2300.5,
                "chargeVoltages": [
                    234,
                    234,
                    235
                ],
                "chargedEnergy": 0,
                "chargerFeatureHeating": false,
                "chargerFeatureIntegratedDevice": false,
                "chargerIcon": null,
                "charging": false,
                "connected": false,
                "connectedDuration": 9223372036854775807,
                "disableThreshold": 0,
                "effectiveLimitSoc": 80,
                "effectiveMaxCurrent": 16,
                "effectiveMinCurrent": 6,
                "effectivePlanSoc": 0,
                "effectivePlanTime": "0001-01-01T00:00:00Z",
                "effectivePriority": 0,
                "enableThreshold": 0,
                "enabled": false,
                "guardAction": "inactive",
                "guardRemaining": 0,
[... 125 lines cut ...]
                "solarPercentage": 43.374919467044315
            },
            "total": {
                "avgCo2": 0,
                "avgPrice": 0.143617620940623,
                "chargedKWh": 1579.6670000000001,
                "solarPercentage": 43.374919467044315
            }
        },
        "tariffFeedIn": 0.2,
        "tariffGrid": 0.2746,
        "tariffPriceHome": 0.2746,
        "tariffPriceLoadpoints": 0.2746,
        "vehicles": {
            "Cowboy-Akku": {
                "title": "Cowboy-Akku (offline)"
            },
            "Valentino": {
                "title": "Valentino",
                "limitSoc": 80
            }
        },
        "version": "0.123.2"
    }
}`

andig, 2023-12-27:
@0163deralex Du sagtest der Fehler entsteht ohne Fahrzeug. Laut /api/state ist aber

      "vehiclePresent": true,

passen die Ausgaben wirklich zusammen?

0163deralex, 2023-12-27:
Der Screenshot ist von heute morgen und die Codes von heute Abend. Aber in beiden Fällen ist der Wagen nicht an der Box angeschlossen.
Da läuft ja noch ein zweiter Loadpoint, so ne tapo Steckdose. Da kommt auch in der ui der Hinweis, dass verbunden ist. Beim loadpoint 1 steht nicht, dass er verbunden ist.
[image]
[image]

andig, 2023-12-27:
Wir raten mangels Config leider schon wieder. Bitte im Issue ergänzen.

naltatis, 2023-12-30:
Jo, Problem ist hier, dass wir diesen Halbzustand am Loadpoint haben. Fahrzeug ist da (present, name, ...), aber es gibt keinen SoC (bzw. `soc: 0`). Daher wird Energy-based Charging verwendet, was (korrekterweise) failed.

Vorschlag: Wir publishen `socBasedPlanning` am Loadpoint (im Core ist das ja korrekt) und ersetzen die redundante Berechnung in der UI: https://github.com/evcc-io/evcc/blob/master/assets/js/components/Loadpoint.vue#L248

Robbi12, 2023-12-30:
Also ich habe das Problem auch. Aber bei einem Volvo. Auto ist definitiv nicht angeschlossen. Wenn ich auf Plan gehe kommt folgender Fehler: Request failed with status code 400: API request failed http://192.168.178.134:7070/api/loadpoints/1/plan/preview/energy/10/2023-12-31T06:00:00.000Z

PR #11603 (2024-01-10): Fix broken plan view for disconnected soc vehicles
fixes #11278 

Align `socBasedPlanning` Implementation in UI with the Go logic.

This is redundant logic. But currently this is the easiest solution without aggressively publishing plan mode every cycle.
closed 2024-01-10 by andig