evcc-io/evcc #11279: minSoc is no longer published for the loadpoint on MQTT

docolli, 2023-12-27:
### Describe the bug

After updating from previous version 0.122.1 to 0.123.0 I do no longer get the minSoc on the loadpoint MQTT topics. It is only published on the vehicle MQTT topics.

minSoc is defined to 65% in the new planer gui.

At least when the car is connected to the loadpoint and is loading to minSoc I would expect to get this value from the loadpoint topics. 

### Steps to reproduce

1. Use 0.122.1, set minSoc in planer and find this value as MQTT topic evcc/loadpoints/1/minSoc
2. Upgrade to 0.123.0
3. MQTT topic evcc/loadpoints/1/minSoc is no longer set
...

### Configuration details

```yaml
vehicles:
- type: template
  template: carwings 
  title: Nissan Leaf  
  user: *** 
  password: ***  
  capacity: 40  
  phases: 1  
  icon: car  
  cache: 15m  
  #mode: pv  
  #minSoc: 30  
  #targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 22  
  name: ev1

loadpoints:
- title: Garage Carmen
  charger: wallbox1
  vehicle: ev1
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 20
#  resetOnDisconnect: false
  soc:
    poll:
      mode: always
      interval: 720m
```

### Log details

```text
provided, if necessary
```

### What type of operating system are you running?

Linux

### Version

0.123.2

andig, 2023-12-27:
`minSoc` ist eine Fahrzeugeigenschaft und jetzt dort publiziert.

closed 2023-12-27 by andig

docolli, 2023-12-27:
Schon klar, aber sollte sich das nicht auf den loadpoint vererben, wenn das Auto angeschlossen ist? 

Ich habe nur 1 E-Auto, aber wenn es mehrere wären, könnte ich ja unterschiedliche minSoc definieren. Macht durchaus Sinn und ihr habt das jetzt korrekt in die Fahrzeugeigenschaften gepackt. Dann wäre es jedoch meiner Meinung nach sinnvoll, dass der loadpoint mir per MQTT zurückgibt, mit welchen soc-limits er gerade "arbeitet", je nach angeschlossenem Auto. Beim Laden bilden loadpoint und Fahrzeug ja eine "Einheit".

In meinem SmartHome Dashboard nutze ich die MQTT Infos zur dynamischen Anzeige der aktuelle Zustände. Hier gibt es für den loadpoint einen Bereich, in dem auch angezeigt wird auf welchen minSoc gerade geladen wird.

andig, 2023-12-27:
> Schon klar, aber sollte sich das nicht auf den loadpoint vererben, wenn das Auto angeschlossen ist?

Nein, wozu?

> Ich habe nur 1 E-Auto, aber wenn es mehrere wären, könnte ich ja unterschiedliche minSoc definieren. 

Ja, kannst du.

> Dann wäre es jedoch meiner Meinung nach sinnvoll, dass der loadpoint mir per MQTT zurückgibt

Er sagt dir ja welches Fahrzeug aktiv ist, darüber ist die Information verfügbar.

docolli, 2023-12-27:
> > Schon klar, aber sollte sich das nicht auf den loadpoint vererben, wenn das Auto angeschlossen ist?
> 
> Nein, wozu?

Weil ein Mapping eines Wertes von einem MQTT Topic für einen evcc User viel einfacher zu machen ist. 

> 
> > Dann wäre es jedoch meiner Meinung nach sinnvoll, dass der loadpoint mir per MQTT zurückgibt
> 
> Er sagt dir ja welches Fahrzeug aktiv ist, darüber ist die Information verfügbar.

So muss ich jetzt in der per MQTT an evcc angebundenen Anwendung erst Code schreiben (wenn das überhaupt so leicht geht) um herauszufinden, welches Auto am loadpoint angeschlossen ist und mir dann das passende MQTT Topic basteln, um den Wert zu bekommen. Der loadpoint hat diesen Wert doch nach Anschluss und Erkennen des Autos sowieso vom Auto übernommen und verwendet diesen, da kann er ihn auch publishen. So hat man auch ein Feedback, mit welchen Werten der loadpoint tatsächlich arbeitet.

andig, 2023-12-27:
Wenn ich meine Fahrzeuge loggen will ist es jetzt viel einfacher. So sieht halt ein normalisiertes Datenmodell aus. Irgendwas ist immer :/

docolli, 2023-12-28:
Ich unterstütze es auch, dass intern ein normalisiertes Datenmodell verwendet wird, da bin ich voll bei dir! Aus Anwendersicht ist es aber oftmals bequemer (und das erhöht die Zufriedenheit), wenn Daten anders sinnvoll aggregiert ausgegeben werden.

Beim Anpassen meines Dashboards und Sichtung der aktuellen MQTT Topics fällt mir dann aber zum Thema "normalisiertes Datenmodell" was auf:
Es gibt den Topic `evcc/loadpoints/1/vehicleCapacity`. Dieser müsste dann auch konsequenterweise nach `evcc/vehicles/ev1/capacity` wandern. Streng genommen ist dann auch `loadpoints_1_vehicleTitle` redundant... 😉

Wenn aber der loadpoint nur noch die Referenz zum vehicleName ausgibt und man die Infos nur noch dort findet, wird es aufwendiger die Infos zusammen zu bekommen, mit welchen Werten der loadpoint gerade arbeitet. Ja, irgendwas ist immer, aber ist es denn so viel extra Code, um die Werte (die der loadpoint vermutlich eh schon intern hat) per MQTT zu publishen?

andig, 2023-12-28:
Das ist dann https://github.com/evcc-io/evcc/pull/11181