evcc-io/evcc #11280: In mode "now" evcc uses only maxcurrent from vehicle, no longer from loadpoint

docolli, 2023-12-27:
### Describe the bug

I have defined maxcurrent for the vehicle to 22A, but for my loadpoint only to 20A, since the line to my loadpoint does not support more than 20A.

Until version 0.123.0 my loading current was always 20A, no matter what mode I was using. With the new christmas version now my car loads with 22A, when I use mode "now". I can confirm on my phoenix wallbox, that 22A is really set as limit.

I would expect, that the maxcurrent that the loadpoint is using, depends on the specified maxcurrent of the connected car AND the specified maxcurrent of the loadpoint, whatever is lower.

### Steps to reproduce

1. define maxcurrent of the car higher than maxcurrent of loadpoint
2. charge with mode pv, check max. power value
3. change to mode now (or charge to minSoc)
4. check max. power value

### Configuration details

```yaml
vehicles:
- type: template
  template: carwings 
  title: Nissan Leaf  
  user: *** 
  password: ***  
  capacity: 40  
  phases: 1  
  icon: car  
  cache: 15m  
  #mode: pv  
  #minSoc: 30  
  #targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 22  
  name: ev1

loadpoints:
- title: Garage Carmen
  charger: wallbox1
  vehicle: ev1
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 20
#  resetOnDisconnect: false
  soc:
    poll:
      mode: always
      interval: 720m
```

### Log details

```text
provided, if necessary.
```

### What type of operating system are you running?

Linux

### Version

0.123.2

andig, 2023-12-27:
This is by design according to https://github.com/evcc-io/evcc/issues/10330:

> vehicle has precendence over charger over loadpoint

/cc @premultiply

docolli, 2023-12-27:
> > I have defined maxcurrent for the vehicle to 22A, but for my loadpoint only to 20A, since the line to my loadpoint does not support more than 20A.
> 
> Why would you do that? Seems contradictory. This is by design according to #10330:

I have specified, what my objects "loadpoint" and "vehicle" are capable to do.
Let's assume I have two loadpoints, one (A) has a weak power supply line and is limited to 16A, the other one (B) however can support full 32A. Now I have a car, that can load with max. 22A.
If I connect my car to loadpoint A, I would expect that evcc uses max. 16A, however, when I connect the next day my car to loadpoint B I would expect to load it with 22A.

But I will read your reference to https://github.com/evcc-io/evcc/issues/10330

premultiply, 2023-12-27:
@andig Ich denke da müssen wir nochmal mit min() / max() bzw. funktionalem Equivalent ran. Da hab ich auch nicht dran gedacht.
Das schwächste Glied in der Kette sollte bzw. muss am Ende die Limits setzen.

docolli, 2023-12-27:
@premultiply : Ist ja schon hier angerissen worden: https://github.com/evcc-io/evcc/issues/10330#issuecomment-1849058398
👍

andig, 2023-12-27:
Solange es keinen Vorschlag gibt, der alle Anforderungen erfüllt bleibts ein Kompromiss. Die x-te Wiederholung von "das hatten wir schonmal" ändert daran nichts.

DerAndereAndi, 2023-12-27:
Ich hatte in dem erwähnten Thread schon mal einen Vorschlag ausführlich erklärt. Dieser ist auch durch @premultiply Umsetzungsidee beschrieben wie das bei euch technisch funktionieren kann. Und alle Anforderungen werden davon erfüllt.

Welche Anforderungen werden denn davon nicht erfüllt?

andig, 2023-12-27:
Siehe https://github.com/evcc-io/evcc/issues/10330#issuecomment-1785869768

DerAndereAndi, 2023-12-27:
It does. For defining the actually used minCurrent, use `max()` of the `minCurrent` of all involved (loadpoint, charger, vehicle).

andig, 2023-12-27:
Könntest Du erklären wir das funktioniert? Zoe ist der minA case, dazu gibts auch noch den maxA Case für 1p Ladung.

DerAndereAndi, 2023-12-27:
Du meinst den unterschiedlichen minCurrent für 1p und 3p? Das wird doch heute auch nicht unterstützt wenn ich mich nicht irre. Daher wäre das dafür keine Lösung aber für alle anderen und damit besser als der Status Quo. Irre ich mich hier?

Um den Fall zu unterstützen, müsstet ihr wohl unterschiedliche `minCurrent` Einstellungen für 1p und 3p auf vehicle Ebene ermöglichen. Und dann je nach Lademodus (1p oder 3p an der Wallbox) den entsprechenden nehmen oder entsprechende der Phases Einstellung der Wallbox. Einen anderen Weg um dieses Problem zu lösen sehe ich nicht. Spricht aber jetzt auch imho nicht gegen den gemachten Vorschlag.

docolli, 2023-12-28:
Die Lösung mit unterschiedlichen Werten für 1p und 3p ist auch schon mal als (vermutlich einzige) Lösung in https://github.com/evcc-io/evcc/issues/10330 diskutiert worden.
Da hängt intern aber dann wohl ein Haufen Änderungsnotwendigkeiten dran, ich denke darum wurde die Umsetzung nicht angegangen.

Wäre es in der config möglich die EInstellungen so getrennt für 1p und 3p zu definieren?
```
vehicles:
- type: template
  template: carwings 
  title: Nissan Leaf  
  user: *** 
  password: ***  
  capacity: 40  
  phases: 1
     minCurrent: 6  
     maxCurrent: 16  
  phases: 3
     minCurrent: 8  
     maxCurrent: 16  
  icon: car  
  cache: 15m 
  name: ev1
```

premultiply, 2023-12-28:
Das ist erstens ein ganz anderes Thema als hier (also OT) und macht zweitens die Komplexität der internen Steuerung / Fallunterscheidung nicht wirklich besser als andere Komfigurationsvarianten.

andig, 2023-12-28:
Unser Vorschlag (@premultiply) analog @DerAndereAndi:

- `minCurrent`: Use the max of loadpoint, charger and vehicle minCurrent
- `maxCurrent`: Use the min of loadpoint, charger and vehicle maxCurrent

Dabei:

- Das Loadpoint-Minimum kann in der Config entfallen bzw. muss entfallen falls `minCurrent` < Loadpoint-Minimum erwünscht ist. Dafür gibt es in der Implementierung eine Standard 6A Untergrenze falls sonst nichts definiert ist (Case: EEbus)
- Charger-Limits gibt es nur dynamisch, aber nicht aus der Konfiguration (Case: EEbus). Optional könnten diese Limits auch aus weiteren Chargern ausgelesen werden.

Offen:

- Abweichende Konfiguration für 1p (out of scope)

DerAndereAndi, 2023-12-28:
Als Ergänzung: Du erwähnst hier immer EEBUS, aber das wird bei jeder Wallbox mit ISO Unterstützung und passendem Fahrzeug auftreten. Auch über andere Protokolle ;)

VolkerK62, 2023-12-28:
Aktuell führt es zu Problemen, wenn am vehicle `maxcurrent` definiert ist. Dann kann man den Wert im UI ändern wie man will, es hat keine Auswirkungen.
Im MQTT Explorer kann man sehen, dass sich am Loadpoint zwar der Wert "maxcurrent" ändert, nicht aber der Wert "effective maxcurrent", der bleibt immer auf dem Fahrzeugwert.
Das sollte ja dann durch den PR ebenfalls gelöst werden?

andig, 2023-12-28:
Ja, genau 👍🏻

andig, 2023-12-30:
Nach Diskussion mit @naltatis würden wir das statische `minCurrent` Limit des Ladepunktes gerne behalten.

Um dem EEbus (ISO!) Usecase Rechnung zu tragen, würden wir https://github.com/evcc-io/evcc/issues/11280#issuecomment-1871230515 so anpassen wollen, dass der Charger die `minCurrent` übersteuern kann (Precedence), während die `maxCurrent` weiterhin das Mininum über alle Werte bleiben muss.

Gegenstimmen?

docolli, 2023-12-30:
Von mir aus könnt ihr dies so umsetzen. Kein Widerspruch, schaun wir mal. Irgendwas ist ja immer 😉
Vielen Dank für eure Mühe und Diskussion!

andig, 2023-12-30:
> Schaun wir mal.... Irgendwas ist ja immer 😉

Isso...

closed 2023-12-30 by andig

andig, 2024-01-12:
Scheint als hätten wir uns mit https://github.com/evcc-io/evcc/issues/11280#issuecomment-1872559005 ins Knie geschossen. Es ist jetzt nicht mehr möglich, bei EEbus einen Minimalstrom per Config festzulegen da er immer vom Charger überschrieben werden kann.