evcc-io/evcc #11298: vehicleSoc no longer written to InfluxDB since EVCC 0.123

cschlipf, 2023-12-28:
### Describe the bug

Even though `evcc vehicle` still shows the current SOC:

```
> cschlipf@home:~ $ evcc vehicle
[main  ] INFO 2023/12/28 00:59:24 evcc 0.123.2
[main  ] INFO 2023/12/28 00:59:24 using config file: /home/cschlipf/evcc.yaml
[db    ] INFO 2023/12/28 00:59:24 using sqlite database: /home/cschlipf/.evcc/evcc.db
[mqtt  ] INFO 2023/12/28 00:59:24 connecting evcc-1079426317 at tcp://localhost:1883
Soc:           85%
Capacity:      79.0kWh
Charge status: B
Charged:       12.7kWh
Range:         467km
Odometer:      25970km
Finish time:   2023-12-28 00:59:00 +0100 CET
Position:      0,0
Target Soc:    85%
OnIdentified:  Mode:pv
Features:      [CoarseCurrent]
```

It is no longer written to the connected InfluxDB:

```
> select value from vehicleSoc order by desc limit 10
name: vehicleSoc
time                 value
----                 -----
2023-12-27T18:31:42Z 0
2023-12-27T17:58:53Z 0
2023-12-27T17:55:41Z 0
2023-12-23T09:34:22Z 0
2023-12-22T19:33:52Z 0
2023-12-22T14:56:58Z 80.36742608284514
2023-12-22T14:55:58Z 79
2023-12-22T14:55:02Z 77.6010069030602
2023-12-22T14:51:25Z 77.60100572866092
2023-12-22T14:50:17Z 76.3003612438637
```

The 0's here are from recent restarts of EVCC. Last updated value is from an EVCC version prior to 0.123. Vehicle was charged after the last time stamp:
[image]

All data that is visualized here is from the EVCC connected InfluxDB.

At the same time the vehicleSoc however is still correctly sent via MQTT. Here logged via Home Assistant:
[image]

### Steps to reproduce

1. Charge vehicle
2. Using influx run these commands:
    `precision rfc3339`
    `select value from vehicleSoc order by desc limit 10`
4. See that no new values are written for vehicleSoc. All other values are written.
...

### Configuration details

```yaml
#General configuration
# open evcc at http://home:7070
network:
  schema: http
  host: home # .local suffix announces the hostname on MDNS
  port: 7070
log: info
levels:
  cache: error
# unique installation id
plant: xxxxxxxxx
# sponsortoken
sponsortoken: xxxxxxxxx
telemetry: true
interval: 10s # control cycle interval

site:
  title: Pirolweg
  meters:
    grid: sofar-grid
    pv: sofar-pv
    battery: 
      - sofar-battery-1

meters:
  - name: sofar-grid
    type: custom
    power:
      source: modbus
      id: 1
      uri: 192.168.158.13:8899
      register:
        address: 0x0488 # Total PCC active power. Selling electricity is positive, buying electricity is negative
        type: holding
        decode: int16
      scale: -10
    energy:
      source: modbus
      id: 1
      uri: 192.168.158.13:8899
      register:
        address: 0x068E # Energy_Purchase_Total
        type: holding
        decode: uint32
      scale: 0.1
    currents:
      - source: modbus
        id: 1
        uri: 192.168.158.13:8899
        register:
          address: 0x0492 # R phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
      - source: modbus
        id: 1
        uri: 192.168.158.13:8899
        register:
          address: 0x049D # S phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
      - source: modbus
        id: 1
        uri: 192.168.158.13:8899
        register:
          address: 0x04A8 # T phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
    powers:
      - source: modbus
        id: 1
        uri: 192.168.158.13:8899
        register:
          address: 0x0493 # ActivePower_PCC_R
          type: holding
          decode: int16
        scale: -10
      - source: modbus
        id: 1
        uri: 192.168.158.13:8899
        register:
          address: 0x049E # ActivePower_PCC_S
          type: holding
          decode: int16
        scale: -10
      - source: modbus
        id: 1
        uri: 192.168.158.13:8899
        register:
          address: 0x04A9 # ActivePower_PCC_T
          type: holding
          decode: int16
        scale: -10

  - name: sofar-pv
    type: custom
    power:
      source: modbus
      id: 1
      uri: 192.168.158.13:8899
      register:
        address: 0x05C4 # Total PV power
        type: holding
        decode: uint16
      scale: 100
    energy:
      source: modbus
      id: 1
      uri: 192.168.158.13:8899
      register:
        address: 0x0686 # PV_Generation_Total
        type: holding
        decode: uint32
      scale: 0.1
      
  - name: sofar-battery-1
    type: custom
    power:
      source: modbus
      id: 1
      uri: 192.168.158.13:8899
      register:
        address: 0x0606 # The charging and discharging power of the first battery pack. Charge is positive, discharge is negative
        type: holding
        decode: int16
      scale: -10
    energy:
      source: modbus
      id: 1
      uri: 192.168.158.13:8899
      register:
        address: 0x069A # Bat_Discharge_Total
        type: holding
        decode: uint32
      scale: 0.1
    soc:
      source: modbus
      id: 1
      uri: 192.168.158.13:8899
      register:
        address: 0x0608 # No. 1 battery pack SOC
        type: holding
        decode: uint16
    capacity: 10

loadpoints:
  - title: Garage
    charger: easee-garage
    # vehicle: tesla-m3-lr
    mode: pv
    phases: 0
    minCurrent: 6
    maxCurrent: 16
  - title: Stellplatz
    charger: easee-stellplatz
    # vehicle: tesla-m3-lr
    mode: pv
    phases: 0
    minCurrent: 6
    maxCurrent: 16

chargers:
  - name: easee-garage
    type: template
    template: easee
    user: xxxxxxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    charger: xxxxxxx
  - name: easee-stellplatz
    type: template
    template: easee
    user: xxxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    charger: xxxxxx

vehicles:
  - name: tesla-m3-lr
    type: template
    template: tesla
    title: Mr White # Wird in der Benutzeroberfläche angezeigt # Optional
    accessToken: xxxxxxx
    refreshToken: xxxxxxx
    mode: pv
    capacity: 79

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.227
  feedin:
    type: fixed
    price: 0.082
  co2:
    type: grünstromindex
    zip: 71126

messaging:
  events:
    # start:
    #   title: Ladestart - ${vehicleTitle}
    #   msg: |
    #     Modus:  ${mode} 
    #     PV Leistung: ${pvPower:%.1fk}kW
    #     Netz: ${gridPower:%.1fk}kW
    #     SOC (Haus): ${batterySoc}%
    # stop:
    #   title: Ladestop - ${vehicleTitle}
    #   msg: |
    #     SOC (${vehicleTitle}): ${vehicleSoc}%
    #     Geladen ${chargedEnergy:%.1fk}kWh in ${chargeDuration}
    #     PV Leistung: ${pvPower:%.1fk}kW
    #     SOC (Haus): ${batterySoc}%
    connect:
      title: Verbunden - ${vehicleTitle}
      msg: |
        Ladepunkt: ${title}
        Modus:  ${mode}
        PV Leistung: ${pvPower:%.1fk}kW
        SOC (Haus): ${batterySoc}%
    disconnect:
      title: Getrennt - ${vehicleTitle}
      msg: |
        SOC (${vehicleTitle}): ${vehicleSoc}%
        Geladen: ${chargedEnergy:%.1fk}kWh
        Verbindungsdauer: ${connectedDuration}
    guest:
      title: Unbekanntes Fahrzeug erkannt
      msg: Verbunden mit Ladepunkt ${title}
  services:
    - type: ntfy
      uri: https://ntfy.sh/schlipf-xxxxxxxxx
      priority: default
      tags: electric_plug,sunny

influx:
  url: http://localhost:8086
  database: evcc
  user: grafana
  password: 

mqtt:
  broker: localhost:1883
  topic: evcc
```

### Log details

```text
x
```

### What type of operating system are you running?

Linux

### Version

0.123.2

VolkerK62, 2023-12-28:
Also mit Version 0.123.1 (d3fe1fc0)  funktioniert es
 
[image]

cschlipf, 2023-12-28:
> Also mit Version 0.123.1 ([d3fe1fc](https://github.com/evcc-io/evcc/commit/d3fe1fc07dbab00558db91164f4765f8aeea4371)) funktioniert es
> 
> ![image](https://private-user-images.githubusercontent.com/99199865/293139950-a0f47764-43aa-4c35-857e-1914835548b8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM3NTk3NzAsIm5iZiI6MTcwMzc1OTQ3MCwicGF0aCI6Ii85OTE5OTg2NS8yOTMxMzk5NTAtYTBmNDc3NjQtNDNhYS00YzM1LTg1N2UtMTkxNDgzNTU0OGI4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI4VDEwMzExMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY5ZmM0NDZkMzhkMjc0NTdmNDJhYjc4ZTc5Yzc5NTIyYTJhYjllODRmOWQ3ZTE3NzgyOTA2YzNkMDk4MmE0MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ef0tvlY_Dvppkj7TmpHD32opnmpPd5i8iE4dWvPo6Bo)

0.123.1 habe ich übersprungen, aber wie Du sehen kannst, fehlen mir alle Daten seit dem 0.123 upgrade. Zufall? Oder habe ich in meiner Config seit dem Update noch einen Fehler drin?

VolkerK62, 2023-12-28:
ich hab jetzt das aktuelle nightly drauf. Bei mir kommen Daten an.
```
> select * from vehicleSoc order by time desc limit 10
name: vehicleSoc
time                loadpoint         value              vehicle
----                ---------         -----              -------
1703780996000000000 Carport           50                 (T)Raumschiff
1703780951000000000 Carport           49.495867768595154 (T)Raumschiff
1703780861000000000 Carport           49.37190082644622  (T)Raumschiff
1703780694000000000 Carport           49.247933884297574 (T)Raumschiff
1703780591000000000 Carport           49.12396694214893  (T)Raumschiff
1703780151000000000 WP SG+            50                 WasserSG
1703780096000000000 WP -Anzeige-      10                 WasserWP
1703780088000000000 Garage            95                 E-Bikes
1703780087000000000 Carport           49                 (T)Raumschiff
1703780070000000000 Waschen -Anzeige- 0                  Waschmaschine

> select value from vehicleSoc order by desc limit 10
name: vehicleSoc
time                value
----                -----
1703780996000000000 50
1703780951000000000 49.495867768595154
1703780861000000000 49.37190082644622
1703780694000000000 49.247933884297574
1703780591000000000 49.12396694214893
1703780151000000000 50
1703780096000000000 10
1703780088000000000 95
1703780087000000000 49
1703780070000000000 0
> 
```

cschlipf, 2023-12-28:
Ich glaube Dir ja, dass es bei Dir tut. Aber was tut bei mir nicht?

Aktueller Stand:
[image]

```
cschlipf@home:~ $ evcc vehicle
[main  ] INFO 2023/12/28 18:45:29 evcc 0.123.2
[main  ] INFO 2023/12/28 18:45:29 using config file: /home/cschlipf/evcc.yaml
[db    ] INFO 2023/12/28 18:45:30 using sqlite database: /home/cschlipf/.evcc/evcc.db
[mqtt  ] INFO 2023/12/28 18:45:30 connecting evcc-384595332 at tcp://localhost:1883
Soc:           55%
Capacity:      79.0kWh
Charge status: C
Charged:       0.4kWh
Range:         306km
Odometer:      26097km
Finish time:   2023-12-28 20:55:00 +0100 CET
Position:      0,0
Target Soc:    85%
OnIdentified:  Mode:pv
Features:      [CoarseCurrent]
```

EVCC erkennt das Fahrzeug, sieht den Soc. In die InfluxDB werden eifrige Daten geschrieben (die Charts aus dem Grafana Dashboard kommen alle aus EVCC Daten und chargePower wird parallel dazu, das vehicleSoc in der Influx absolut null neue Daten bekommt, ebenfalls eifrig aktualisiert (wie alle anderen InfluxDB Tabellen auch - eben nur nicht vehicleSoc).

Letzter Ladevorgang, bei dem noch vehicleSoc geschrieben wurde vor 0.123:
[image]

Heute während des aktuellen Ladevorganges:
[image]

Irgendeinen Tipp, was ich nun machen kann?

cschlipf, 2023-12-28:
Gerade nochmal EVCC neu gestartet während der Ladevorgang läuft und das Auto wach ist. Für den Loadpoint Stellplatz (an dem das Auto nicht dran hängt) wird ein Wert von 0 geschrieben beim Neustart. Für den Loadpoint Garage, wo gerade geladen wird, wird gar nichts geschrieben:

[image]

cschlipf, 2023-12-28:
Hm. Habe nun ein Downgrade auf 0.122.1 gemacht. Auch damit habe ich nun dasselbe Verhalten wie mit der 0.123.2.

VolkerK62, 2023-12-28:
Ich habe deine Konfig mit meiner verglichen und eigentlich nur einen Unterschied festgestellt.
Ich habe an meinen Loadpoints einen `soc` Eintrag.
Bei den Shellies nur mit `estimate: false` am Carport auch noch mit `poll`.

cschlipf, 2023-12-28:
Habe nun diese Loadpoint Config:

```yaml
loadpoints:
  - title: Garage
    charger: easee-garage
    # vehicle: tesla-m3-lr
    # mode: pv
    phases: 0
    minCurrent: 6
    maxCurrent: 16
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: true

  - title: Stellplatz
    charger: easee-stellplatz
    # vehicle: tesla-m3-lr
    # mode: pv
    phases: 0
    minCurrent: 6
    maxCurrent: 16
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: true
```

Keine Änderung :-(

cschlipf, 2023-12-28:
Habe nun mal die Config vom 17.12. aus dem Backup geholt mit 0.122.1 (beim Upgrade zu 0.123 mussten ja einige Parameter entfernt werden). Auch damit keine Änderung. Ich werde wahnsinnig. Alles wird geschrieben - außer dieser blöde vehicleSoc. 

Und wenn der Soc im EVCC UI angezeigt wird und `evcc vehicle` ihn auch korrekt erfasst, dann sollte EVCC den Wert doch auch wie alle anderen Werte in die Influx schreiben. Oder wo ist hier mein Gedankenfehler?

VolkerK62, 2023-12-28:
Was sagt denn das log mit `influx: trace`?

cschlipf, 2023-12-28:
Auch seltsam:
[image]

Beim EVCC restart wird für den Loadpoint 'Garage' nur ein Wert geschrieben, wenn kein Auto dran hängt oder wenn das Auto abgekoppelt wird.

VolkerK62, 2023-12-28:
Den Eintrag beim Log-Level ergänzen.
```
log: info
levels:
  cache: error
  influx: trace
```
evtl noch soc.poll auf "always" mit 1 Minute ändern (nur testweise)
Neustart, evtl Ladung starten und dann nach ein paar Minuten Log abrufen
https://docs.evcc.io/docs/guides/setup#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

cschlipf, 2023-12-28:
[evcc.influx.trace.log](https://github.com/evcc-io/evcc/files/13788837/evcc.influx.trace.log)

Ich habe mal einfach evcc -l trace ausgeführt und alle influx Einträge in obiges Files gegrepped. EVCC wurde gestartet. Auto war nicht verbunden. Dann Auto verbunden und einen Ladevorgang gestartet.

Bei der suche nach vehicleSoc bin hier auf diese Zeilen gestoßen:

```
cschlipf@home:~ $ grep vehicleSoc evcc.influx.trace.log 
[influx] TRACE 2023/12/28 19:54:20 write vehicleSoc=map[value:0] (map[loadpoint:Garage])
[influx] TRACE 2023/12/28 19:54:20 write vehicleSoc=map[value:0] (map[loadpoint:Garage])
[influx] TRACE 2023/12/28 19:54:29 write vehicleSoc=map[value:0] (map[loadpoint:Stellplatz])
[influx] TRACE 2023/12/28 19:54:29 write vehicleSoc=map[value:0] (map[loadpoint:Stellplatz])
[influx] TRACE 2023/12/28 19:55:35 write vehicleSoc=map[value:64] (map[loadpoint:Garage vehicle:Mr White])
```

Nach der letzten zu Urteilen hätte er also einen Wert für das Auto in der Garage von einem Soc von 64% schreiben müssen (welcher der aktuelle Wert ist).

Aber:
```
> select * from vehicleSoc order by desc limit 10
name: vehicleSoc
time                 loadpoint  value vehicle
----                 ---------  ----- -------
2023-12-28T19:05:25Z Stellplatz 0     
2023-12-28T18:54:29Z Stellplatz 0     
2023-12-28T18:54:20Z Garage     0     
2023-12-28T18:53:34Z Stellplatz 0     
2023-12-28T18:53:26Z Garage     0     
2023-12-28T18:44:42Z Stellplatz 0     
2023-12-28T18:44:23Z Garage     0     
2023-12-28T18:43:50Z Garage     0     
2023-12-28T18:42:40Z Stellplatz 0     
2023-12-28T18:26:48Z Stellplatz 0     
```

cschlipf, 2023-12-28:
Was mir auch noch auffällt:

Ich kann die Trace Einträge für den Soc von 0 auf die DB Mappen. Jedoch habe ich hier jedesmal zwei Trace Einträge für jeweils einen Influx Eintrag.

Beim Schreiben des Wertes von 64 fehlt mir der passende Eintrag in der Influx DB und im Trace Log wird die 64 nur einmal erwähnt. Nicht doppelt wie bei den 0er Werten, die wirklich geschrieben worden sind.

VolkerK62, 2023-12-29:
Ich habe bei mir mal geschaut. Wenn geladen wird, wird ca alle 2 Minuten einmal ein Wert geschickt. Vermutlich wegen einer Soc- Änderung durch den Estimator. Doppelwerte habe ich keine.
Sorry, dass ich dir nicht weiterhelfen kann.

VolkerK62, 2023-12-29:
Hab noch einen Strohhalm gefunden.
Letztes gab es irgendwie Probleme mit Leerzeichen im Namen (Details nicht bekannt).
Ändere doch mal deinen "Mr White" in "Mr-White".

cschlipf, 2023-12-29:
Sehr seltsam. Was hat sich geändert? Warum trifft es nur vehicleSoc und sonst keinen anderen Wert?

VolkerK62, 2023-12-29:
Bin kein Code-freak. mit einfachen Worten: Vehicle wurde vom Loadpoint entkoppelt.
UNd warum bei dir und bei mir nicht.

cschlipf, 2023-12-29:
Noch etwas hat sich in dem Zeitraum geändert: Das Tesla Weihnachtsupdate kam.

GrimmiMeloni, 2023-12-29:
@cschlipf ich habe gerade mal eine minimale Repro versucht.

Habe eine Influx erzeugt, und die demo.yaml um Deine Influx Konfiguration ergänzt.
Ich sehe hier Werte vom blauen e-Golf in Chronograph.

[image]

Query:  
`SELECT mean("value") AS "mean_value" FROM "evcc"."autogen"."vehicleSoc" WHERE time > :dashboardTime: AND time < :upperDashboardTime: GROUP BY time(:interval:) FILL(null)`

Für mich riecht das nach einem Problem mit Deiner Influx. Hast Du dort mal die Logs geprüft? Vielleicht hat die irgendwie Schluckauf.

cschlipf, 2023-12-29:
@GrimmiMeloni Ich habe mir jetzt mal das Write Logging der Influx angeschaltet. Unten sieht man was geschrieben wird während EVCC neu startet:

Dec 29 17:16:09 home influxd-systemd-start.sh[24096]: ts=2023-12-29T16:16:09.870494Z lvl=info msg="Write body received by handler" log_id=0mPaXlw0000 service=httpd body="gridCurrents l1=0,l2=0,l3=0 1703866569\ngridEnergy value=1566.8000000000002 1703866569\nhomePower value=590 1703866569\ngreenShareHome value=1 1703866569\ngreenShareLoadpoints value=0 1703866569\nchargeCurrents,loadpoint=Stellplatz l1=0,l2=0,l3=0 1703866569\n**vehicleSoc,loadpoint=Stellplatz value=0i 1703866569**\nchargeCurrent,loadpoint=Stellplatz value=0 1703866569\nsessionEnergy,loadpoint=Stellplatz value=0 1703866569\nsessionSolarPercentage,loadpoint=Stellplatz value=0 1703866569\neffectivePriority,loadpoint=Stellplatz value=0i 1703866569\nvehicleClimaterActive,loadpoint=Stellplatz value=\"<nil>\" 1703866569\n**vehicleSoc,loadpoint=Stellplatz value=0 1703866569**\nvehicleRange,loadpoint=Stellplatz value=0i 1703866569\nvehicleTargetSoc,loadpoint=Stellplatz value=0 1703866569\nphasesActive,loadpoint=Stellplatz value=3i 1703866569\neffectiveMinCurrent,loadpoint=Stellplatz value=6 1703866569\neffectivePlanSoc,loadpoint=Stellplatz value=0i 1703866569\neffectiveMaxCurrent,loadpoint=Stellplatz value=16 1703866569\neffectiveMaxCurrent,loadpoint=Stellplatz value=16 1703866569\neffectiveLimitSoc,loadpoint=Stellplatz value=100i 1703866569\nchargeCurrent,loadpoint=Stellplatz value=0 1703866569\nchargeTotalImport,loadpoint=Stellplatz value=36.9296586111111 1703866569\neffectivePriority,loadpoint=Stellplatz value=0i 1703866569\nchargedEnergy,loadpoint=Stellplatz value=0 1703866569\neffectivePlanSoc,loadpoint=Stellplatz value=0i 1703866569\neffectiveMinCurrent,loadpoint=Stellplatz value=6 1703866569\neffectiveLimitSoc,loadpoint=Stellplatz value=100i 1703866569\nphasesActive,loadpoint=Stellplatz value=3i 1703866569\nsessionEnergy,loadpoint=Stellplatz value=0 1703866569\ntariffPriceHome value=0.082 1703866569\ntariffCo2Home value=0 1703866569\nsessionSolarPercentage,loadpoint=Stellplatz value=0 1703866569\ntariffPriceLoadpoints value=0.227 1703866569\ntariffCo2Loadpoints value=86 1703866569\nchargePower,loadpoint=Garage,vehicle=Tesla value=0 1703866569\nplanEnergy,loadpoint=Stellplatz value=0 1703866569\nchargePower,loadpoint=Stellplatz value=0 1703866569\n"

In Fett habe ich die Daten markiert, die für vehicleSoc gesetzt werden: Auch hier fehlt die Garage, die gerade angeschlossen ist mit dem entsprechenden Fahrzeug. Nur der Stellplatz wird beim Neustart wieder auf 0 gesetzt. Sieht das so aus als ob EVCC die Daten für Garage und das angeschlossene Fahrzeug erst gar nicht übermittelt?

cschlipf, 2023-12-29:
Ich habe jetzt einen Eintrag gefunden wo für die Garage der vehicle Soc gesetzt wird. Allerdings in einem Request einmal auf 0 und einmal auf 80 (der aktuell, korrekte Wert). Warum werden hier innerhalb eines Requests die Werte zweimal mit geschickt und dann mit inkonsistenten Werten?:

Dec 29 17:18:41 home influxd-systemd-start.sh[24096]: ts=2023-12-29T16:18:41.549789Z lvl=info msg="Write body received by handler" log_id=0mPaXlw0000 service=httpd body="effectivePriority,loadpoint=Garage,vehicle=Tesla value=0i 1703866721\nvehicleRange,loadpoint=Garage,vehicle=Tesla value=0i 1703866721\nvehicleCapacity,loadpoint=Garage,vehicle=Tesla value=79 1703866721\nphasesActive,loadpoint=Garage,vehicle=Tesla value=1i 1703866721\nvehicleRange,loadpoint=Garage,vehicle=Tesla value=443i 1703866721\nvehicleClimaterActive,loadpoint=Garage,vehicle=Tesla value=\"<nil>\" 1703866721\nvehicleTargetSoc,loadpoint=Garage,vehicle=Tesla value=0 1703866721\n**vehicleSoc,loadpoint=Garage,vehicle=Tesla value=0 1703866721**\neffectiveMinCurrent,loadpoint=Garage,vehicle=Tesla value=6 1703866721\neffectiveMaxCurrent,loadpoint=Garage,vehicle=Tesla value=16 1703866721\neffectiveLimitSoc,loadpoint=Garage,vehicle=Tesla value=80i 1703866721\n**vehicleSoc,loadpoint=Garage,vehicle=Tesla value=80 1703866721**\nvehicleTargetSoc,loadpoint=Garage,vehicle=Tesla value=85 1703866721\neffectivePlanSoc,loadpoint=Garage,vehicle=Tesla value=0i 1703866721\neffectiveLimitSoc,loadpoint=Garage,vehicle=Tesla value=80i 1703866721\neffectivePlanSoc,loadpoint=Garage,vehicle=Tesla value=0i 1703866721\neffectiveMinCurrent,loadpoint=Garage,vehicle=Tesla value=6 1703866721\neffectiveMaxCurrent,loadpoint=Garage,vehicle=Tesla value=16 1703866721\neffectivePriority,loadpoint=Garage,vehicle=Tesla value=0i 1703866721\ngreenShareHome value=1 1703866721\ntariffPriceHome value=0.082 1703866721\ngreenShareLoadpoints value=0 1703866721\ntariffPriceLoadpoints value=0.227 1703866721\ntariffCo2Loadpoints value=90 1703866721\ntariffCo2Home value=0 1703866721\nchargePower,loadpoint=Garage,vehicle=Tesla value=0 1703866721\nchargePower,loadpoint=Stellplatz value=0 1703866721\n"

Allerdings fehlt für 17:18:41 (16:18:41 UTC) ein Eintrag in der DB:

```
> select * from vehicleSoc order by desc limit 10
name: vehicleSoc
time                 loadpoint  value
----                 ---------  -----
2023-12-29T16:17:48Z Stellplatz 0
2023-12-29T16:16:09Z Stellplatz 0
2023-12-29T16:13:33Z Stellplatz 0
2023-12-29T16:03:24Z Stellplatz 0
2023-12-29T16:01:31Z Stellplatz 0
2023-12-29T15:39:41Z Stellplatz 0
2023-12-29T15:39:14Z Stellplatz 0
```

Eine Fehlermeldung kann ich in der DB nicht finden.

GrimmiMeloni, 2023-12-30:
Soweit ich den Code richtig interpretiere, werden die Werte nur mit einem Vehicle Tag versehen, wenn auch ein Fahrzeug am Loadpoint erkannt wurde.

https://github.com/GrimmiMeloni/evcc/blob/fa0fb907ff588ad323bfa8a3142479b9b562ca37/server/influxdb.go#L157

Das würde zumindest beim Neustart ggf. erklären warum ohne Vehicle Tag geschrieben wird.
Dann ist auch 0 sicherlich der richtige Wert. Ob für "kein Auto" immer einen "Ladestand 0" zu loggen sinnvoll oder überflüssig ist könnte man nochmal hinterfragen. Ich sehe hier aber erstmal noch keinen direkten Zusammenhang zu _fehlenden_ Werten.

> Allerdings fehlt für 17:18:41 (16:18:41 UTC) ein Eintrag in der DB:

Ich kenne das Timing Deiner DB Abfragen nicht. Wurde die Query einfach zu früh abgesetzt? Oder bleibt `2023-12-29T16:17:48Z` jetzt für immer der aktuellste Timestamp in der DB? Wenn nein, ab wann siehst Du wieder Werte?

cschlipf, 2023-12-30:
Die Sache ist ja die: Solange das Auto an der Garage angehängt ist, wird hier gar nichts mehr geschrieben. Es werden nur die 0er für den Stellplatz geschrieben, wo kein Auto dran hängt.

Für alle anderen Measurements (also alles außer vehicleSoc) bekomme ich weiterhin aktuelle Timestamps. Für vehicleSoc derzeit nur, wenn ich EVCC neu starte, da die Werte für das verbundene Fahrzeug ja nicht mehr geschrieben werden. Auch nicht, wenn ich das Fahrzeug anstecke. Nur beim abstecken, erhalte ich dann wieder eine 0 für die Garage:
[image]
(Fahrzeug wurde gestern abend abgesteckt und heute morgen gerade eben zum testen).

Also wird der SoC nur aktualisiert, wenn kein Fahrzeug dran hängt.

andig, 2023-12-30:
Ich sehe hier kein Todo. Wir können gerne wieder auf machen, wenn es ein evcc Log gibt es dem sichtbar wird, dass ein Wert der eigentlich da wäre nicht geschrieben wird. Die unkorrelierten Influxlogs helfen leider nicht bei der Fehlersuche.

closed 2023-12-30 by andig

andig, 2023-12-30:
Ich habe auch überprüft, dass der soc geschrieben wird:

[image]

cschlipf, 2023-12-30:
Vielen Dank für all Eure Mühe.

Ich verstehe ja auch nicht was bei mir da schief läuft. An der EVCC Version scheint es nicht zu liegen (alte Version mit alter Config tut auch nicht mehr).  Im UI und über MQTT zeigt EVCC die korrekten Werte an. Laut Influx Log kommt ein Schreibzugriff.

Für mein Dashboard hole ich mir jetzt den SoC von TeslaMate als Workaround.

Ich melde mich, falls ich neue hilfreiche Erkenntnisse finde - aber aktuell kann ich nicht sagen, dass es an EVCC liegt. Sieht eher nicht so aus.

cschlipf, 2024-01-17:
Ich weiß jetzt nicht was genau ihr geändert habt, aber irgendwas in entweder 0.123.8 oder 0.123.9 hat das jetzt gefixt. Habe es schon aufgegeben gehabt und sonst nichts mehr geändert. Als ich aber heute von 0.123.7 aktualisiert habe wurden die Werte auf einmal wieder geschrieben:

```
> select * from vehicleSoc order by desc limit 10
name: vehicleSoc
time                 loadpoint  value              vehicle
----                 ---------  -----              -------
2024-01-17T12:54:49Z Garage     53.164412936077845 Tesla
2024-01-17T12:53:52Z Garage     52.001349683544284 Tesla
2024-01-17T12:47:52Z Garage     52                 Tesla
2024-01-17T12:46:23Z Stellplatz 0       
```

eilmi, 2024-02-17:
selbiges Problem in Version 0.124.2 heute beobachten müssen

Evcc erkennt Fahrzeug ohne Probleme (Tesla mit neuer Fleet-API) ins MQTT kommen sowohl Akkustand, wie auch vehicleRange und alles weitere, in die influxdb gelangt allerdings nur vehicleRange, vehicleOdometer,... aber vehicleSoC ist immer null :-/

andig, 2024-02-17:
Das wundert mich nicht, denn der Wert heisst `vehicleSoc`