evcc-io/evcc #11301: ID Template Fehler: wake-up vehicle: unexpected status: 502 (Bad Gateway)

allcoolusernamesaregone, 2023-12-28:
### Describe the bug

Ein E-Golf über das ID template konfiguriert, bringt seit der 0.123.1 beim Ladestart in der UI die Meldung
"Wallbox: wake-up vehicle: unexpected status: 502 (Bad Gateway)"

Sonst keine Auffälligkeiten, aber siehe Details im Log.

### Steps to reproduce

Nur das Auto anschließen zum Laden..

### Configuration details

```yaml
vehicles:
- type: template
  template: id 
  title: e-Golf  
  user: xxx
  password: xxx
  vin: xxx
  capacity: 32  
  phases: 2  
  icon: car  
  cache: 10m 
  mode: pv  
  name: egolf
```

### Log details

```text
Dez 28 10:29:45 evcc[2386]: [lp-1  ] DEBUG 2023/12/28 10:29:45 wake-up vehicle
Dez 28 10:29:45 evcc[2386]: [id    ] TRACE 2023/12/28 10:29:45 POST https://emea.bff.cariad.digital/vehicle/v1/vehicles/***/charging/start
Dez 28 10:29:46 evcc[2386]: [id    ] TRACE 2023/12/28 10:29:46 {"error":{"message":"Bad Gateway","info":"Upstream service responded with an unexpected status. If the problem persists, please contact our support.","code":4101,"group":2,"retry":true}}
Dez 28 10:29:46 evcc[2386]: [lp-1  ] ERROR 2023/12/28 10:29:46 wake-up vehicle: unexpected status: 502 (Bad Gateway)
```

### What type of operating system are you running?

Linux

### Version

0.123.1

andig, 2023-12-28:
Geht es wieder wenn du eine Version zurück gehst?

StefanSchoof, 2023-12-28:
Kannst du in deiner VW das Laden starten? Bei dem ID template ist das wake up ein start charge.

allcoolusernamesaregone, 2023-12-28:
[image]

ok, es geht nicht. Also Fehler VW, nicht hier vom Template.
Wie gut, dass VW da Wert auf die Schneeflocken legt, statt auf funktionsfähige Software... 
Danke für den Tipp. 

PS: Laden stoppen geht allerdings.. rock solid die Server bei VW.

closed 2023-12-28 by allcoolusernamesaregone