evcc-io/evcc #11306: Hinweis Batteriesteuerung überarbeiten

VolkerK62, 2023-12-28:
Danke euch. Da hat mich in der Tat der Hinweis ".... wird nicht unterstützt" verwirrt. Man kann ja in der Tat trotzdem noch die Werte für prioritySoC und bufferSoC in der GUI einstellen. Vielleicht sollte man den Hinweis nochmal sprachlich überarbeiten. Das verwirrt doch recht ordentlich. Aber Danke für die schnellen Hinweise!

_Originally posted by @ffm777 in https://github.com/evcc-io/evcc/discussions/11304#discussioncomment-7964127_

VolkerK62, 2023-12-28:
Da ist kein Einzelfall.
Evtl sollte man den Hinweis "Batteriesteuerung wird für diesen Speicher nicht unterstützt"  mit einem "darüberhinausgehende" ergänzen

andig, 2023-12-28:
> Edit: oder alternativ nur anzeigen, wenn ein ein-/ausschalten des feature möglich ist

Wäre auch mein Favorit

naltatis, 2023-12-30:
> > Edit: oder alternativ nur anzeigen, wenn ein ein-/ausschalten des feature möglich ist
> 
> Wäre auch mein Favorit

Effektiv heißt das, die Meldung einfach rauswerfen, richtig? Find ich ok. Gerade, wenn wir die Capability dann in der Doku (https://github.com/evcc-io/docs/issues/489]) und später auch in der Config-UI anzeigen.

closed 2024-01-01 by andig