evcc-io/evcc #11308: Problem with smartCostLimit "has invalid keys"

MirkoHuf, 2023-12-28:
### Describe the bug

Ich habe an der bestehenden evcc.yaml nichts geändert, ausser den paramater smartCostLimit hinzugefügt - hoffentlich wie in der Doku beschrieben. Jetzt startet aber evcc nicht mehr mit der Meldung * " has invalid keys: smartcostlimit
Die EVCC Version ist 0.123.2

Kann mir jemand helfen?

### Steps to reproduce

1. Konfiguration angepasst (siehe evcc.yaml)
2. evcc gestartet

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: synologynas # .local suffix announces the hostname on MDNS
  port: 7070

log: error
levels:
  cache: error
  ocpp: error

# unique installation id
plant: XXX

interval: 10s # control cycle interval

sponsortoken: XXX

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.XX  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.178.XX  
  password: XX  
  name: pv2
- type: template
  template: sma-inverter 
  usage: battery  
  host: 192.168.178.XX  
  password: XX  
  name: battery3

chargers:
- type: template
  template: ocpp 
  name: wallbox_garage
  stationId: XX
  connector: 1
  connecttimeout: 5m
  timeout: 2m
  getconfiguration: false
  bootnotification: false
- type: template
  template: ocpp 
  name: wallbox_carport
  stationId: XX
  connector: 1
  connecttimeout: 5m
  timeout: 2m
  getconfiguration: false
  bootnotification: false

loadpoints:
- title: Garage
  charger: wallbox_garage
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  vehicle: Polestar
- title: Carport
  charger: wallbox_carport
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  vehicle: Tesla

tariffs:
  grid:
    type: tibber
    token: XX

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  smartCostLimit: 0.20
  residualPower: 100
  
hems:
  type: sma
  AllowControl: false
  
vehicles:
  - name: Tesla
    type: template
    template: tesla
    title: Tesla Model Y
    accessToken: XX
    refreshToken: XX
    vin: XX
  - name: Polestar
    type: template
    template: polestar
    title: Polestar 2
    user: XX
    password: XX
```

### Log details

```text
Wie kann ich das denn ausführen, wenn evcc in einem Docker Container läuft auf einer synology nas?
```

### What type of operating system are you running?

Docker container

### Version

0.123.2

andig, 2023-12-28:
Bitte Release Notes lesen

closed 2023-12-28 by andig

MirkoHuf, 2023-12-28:
Hallo, sorry wenn ich das jetzt nochmal auf mache. Ich habe die Release Notes für die letzten 5 Versionen gelesen, aber keine Hilfe für mein Problem gefunden. In der Doku ist dieser Parameter auch nicht als deprecated markiert. Kannst du mir helfen? Welche Change bezieht sich den auf dieses Problem. Tür mit echt leid, aber ich finde es wirklich nicht.

andig, 2023-12-28:
https://github.com/evcc-io/evcc/releases/tag/0.123.0:

>  arious yaml settings like minSoc, targetSoc, resetOnDisconnect  **and others** have been removed. If you encounter 
> errors like: failed configuring loadpoint 1 error(s) decoding: * '' has invalid keys abc
> simply remove the offending abc entry.

MirkoHuf, 2023-12-28:
Danke @andig - ich hatte das zwar gelesen, aber meinen Parameter nicht gefunden. Jetzt habe ich den Wert in der UI gefunden.
Könnt Ihr die Doku noch anpassen (oder kann ich das machen?)
https://docs.evcc.io/docs/reference/configuration/site#smartcostlimit-ehemals-cheap