evcc-io/evcc #11312: Kein Zugriff mehr auf BMW i4 nach Update auf 0123.x

digitalonkel, 2023-12-28:
### Describe the bug

Der BMW ist mit der UI konfiguriert. Seit 0.123.x kein Zugriff mehr auf die Fahrzeugdaten.

### Steps to reproduce

1. BMW i4 mit UI konfigurieren 
2.
3.
...

### Configuration details

```yaml
Fahrzeug BMW i4 neu anlegen geht auch nicht mehr. Da aufgrund der Fehlermeldung nicht gespeichert werden kann.
```

### Log details

```text
cannot create vehicle 'template': cannot create vehicle 'bmw': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized)
```

### What type of operating system are you running?

Linux

### Version

0.123.2

andig, 2023-12-28:
…und alte Version geht dann wieder?

digitalonkel, 2023-12-28:
Zumindest können andere Dienste, wie z.B. EEVEE weiter mit den gleichen Daten auf das Fz. zugreifen.
Wie kann ich auf einem RasPi mit Linux einen Downgrade machen?

TweetsOfNiklas, 2023-12-28:
Habe dasselbe Problem mit dem i3. Ich werde morgen mal probieren mich bei der App aus- und einzuloggen. Das letzte Mal wollte er da dann ein neues Passwort haben.

andig, 2023-12-29:
> Zumindest können andere Dienste, wie z.B. EEVEE weiter mit den gleichen Daten auf das Fz. zugreifen. Gerade einen Downgrade auf 0.122.1 gemacht. Geht auch nicht.

Titel korrigiert, da gibts ja offensichtlich keinen Zusammenhang. Die Ursache dürfte damit bei BMW liegen.

fuxae, 2023-12-29:
> > Zumindest können andere Dienste, wie z.B. EEVEE weiter mit den gleichen Daten auf das Fz. zugreifen. Gerade einen Downgrade auf 0.122.1 gemacht. Geht auch nicht.
> 
> Titel korrigiert, da gibts ja offensichtlich keinen Zusammenhang. Die Ursache dürfte damit bei BMW liegen.

Also allgemein an BMW liegt es nicht. Bei meinem i4 ist alles in Ordnung und evcc zeigt alles korrekt an.
Allerdings habe ich meinen nicht im UI konfiguriert - wobei das eigentlich ja nichts ausmachen sollte?

TobiasHuber1980, 2023-12-29:
mein i3:

`Garage: vehicle soc: 403: Out of call volume quota. Quota will be replenished in 00:24:33. (unexpected status: 403 (Forbidden))`

TobiasHuber1980, 2023-12-29:
Ich hatte gerade beim i3
`Garage: vehicle soc: 403: Out of call volume quota. Quota will be replenished in 00:24:33. (unexpected status: 403 (Forbidden))`
hab mich testweise in der bmw app abgemeldet, neu angemeldet, geht jetzt wieder in evcc. weiss aber nicht ob das zusammenhängt.

digitalonkel, 2023-12-29:
> Ich hatte gerade beim i3 `Garage: vehicle soc: 403: Out of call volume quota. Quota will be replenished in 00:24:33. (unexpected status: 403 (Forbidden))` hab mich testweise in der bmw app abgemeldet, neu angemeldet, geht jetzt wieder in evcc. weiss aber nicht ob das zusammenhängt.

Das Verhalten kann ich bestätigen. Geht nach dem Ab- und Anmelden in der BMW-App wieder.

closed 2023-12-29 by digitalonkel