evcc-io/evcc #11316: Nach Update auf 0.123.2 kalb cvcc nicht mehr laden

Deanw1975, 2023-12-28:
### Describe the bug

Abend zusammen,

ich habe das update auf 0.123.2 und erhalte beim Aufruf von EVCC diese Fehlermeldung:

failed configuring loadpoints
failed configuring loadpoint

1 error(s) decoding: * '' has invalid keys: resetondisconnect, vehicles

Wenn ich den dorngrad auf 0.122.1 durchführe habe ich keine Probleme mehr.

Was manche ich falsch?

VG
Dean

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: xxxxxxxxxx

meters:
- type: template
  template: senec-home
  usage: grid
  host: 192.168.178.137
  name: grid1
- type: template
  template: senec-home
  usage: pv
  host: 192.168.178.137
  name: pv2
- type: template
  template: senec-home
  usage: battery
  host: 192.168.178.137
  name: battery3

chargers:
- type: template
  template: heidelberg
  id: 1
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: 8E1
  modbus: rs485serial
  name: wallbox5

vehicles:
- type: template
  template: enyaq
  title: black pearl
  user: xxxxxxx@xxxxxxxxx
  password: xxxxxxxx
  vin: xxxxxxxxxxxxxx
  capacity: 60
  name: ev4

loadpoints:
- title: carport
  charger: wallbox5
  vehicles:
  - ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Carport
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  prioritySoC: 30
  residualPower: 300

# mqtt message broker
mqtt:
  broker: 192.168.xxxx.xxxx:1884
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: xxxxxxx
  password: xxxxxxx
  clientid: 01

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.301 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh
```

### Log details

```text
0.122.1
```

### What type of operating system are you running?

Linux

### Version

 0.122.1

Balujuhu, 2023-12-28:
Resetondisconnect entfernen

andig, 2023-12-28:
Release notes lesen…

closed 2023-12-28 by andig

Deanw1975, 2023-12-29:
Resetondisconnect entfernt 
jetzt kommt noch dieser Fehler:

failed configuring loadpoints
failed configuring loadpoint
1 error(s) decoding: * '' has invalid keys: vehicles

Was habe ich hier noch übersehen?

andig, 2023-12-29:
Einfach löschen, genau wie bei den anderen identischen Fehlern

Deanw1975, 2023-12-29:
Habe jetzt jeden Fehler ein der Datei rausgenommen bis EVCC wieder gelaufen ist.
In der Oberfläche konnte ich den Wagen wieder anlegen.
Finde aber keine Konfiguration bezüglich der Senec Anlage.

Nun wird mir aber keine Daten mehr zum aktuellen Akkustand oder zum Hausverbrauch etc mehr angezeigt.
Somit werden auch keine ordentlichen Daten via MQTT versendet.
Wäre es ggf. sinnvoll die Konfiguration einmal komplett neu anzulegen.

VG