evcc-io/evcc #11318: Smartfox Template funktioniert nicht mit Smartfox Pro (aktuelle Firmware EM2 00.01.07.03)

rasenschachmatt, 2023-12-28:
### Describe the bug

Ich habe versucht, meinen Smartfox Pro (mit der [aktuellen Firmware Version](https://smartfox.at/downloads/software/) EM2 00.01.07.03) über das Template in evcc einzubinden. (Bei meinem Wechselrichter hat das wunderbar geklappt.)

Bei meinem Smartfox Pro funktioniert dies aber nicht, da die im Template verwendete URI http://{{ .host }}**/all** vom Gerät anscheinend nicht mehr unterstützt wird.

Das Web GUI von Smartfox selbst verwendet die URI http://{{ .host }}**/values.xml**, welches als Response nicht ein JSON, sondern ein XML Dokument zurückliefert (siehe Beispiel Response [values.xml](https://github.com/evcc-io/evcc/files/13789807/values.txt)).

Inzwischen habe ich es geschafft, die Leistung meines Smartfox Pro über ein Messgerät des Typs custom (statt template) auszulesen:
```
  name: SmartfoxCustom
  type: custom
  power: # power (W)
    source: http
    uri: http://192.168.178.20/values.xml
    method: GET # default HTTP method
    jq: .values.value[] | select(.attrid=="detailsPowerValue")."#content" | rtrimstr(" W") # parse response json (https://jqplay.org is your friend)
    scale: 1 # floating point factor applied to result, e.g. 0.001 for kW to W conversion
    timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration
```

Der Output von evcc meter sind nun folgendermassen aus:
```
C:\tools\evcc\smartfox>evcc meter
[main  ] INFO 2023/12/28 23:18:46 evcc 0.123.2
[main  ] INFO 2023/12/28 23:18:46 using config file: C:\tools\evcc\smartfox\evcc.yaml
[db    ] INFO 2023/12/28 23:18:46 using sqlite database: C:\Users\patri\.evcc\evcc.db
SE15K
-----
Power:  0W
Energy: 63456.4kWh

SmartfoxTemplate
----------------
Power:          Get "http://192.168.178.20/all": EOF
Energy:         Get "http://192.168.178.20/all": EOF
Current L1..L3: Get "http://192.168.178.20/all": EOF
Voltage L1..L3: Get "http://192.168.178.20/all": EOF
Power L1..L3:   Get "http://192.168.178.20/all": EOF

SmartfoxCustom
--------------
Power: 4500W
```

Schöner wäre es aber natürlich, den Smartfox Pro über ein angepasstes oder neues Template anzubinden. (Ich vermute, dass auch andere Benutzer des Smartfox auf dieses Problem stossen werden, wenn sie auf der aktuellen Firmware sind.)

Was haltet ihr davon?

### Steps to reproduce

1. Verwende einen Smartfox Pro mit der [aktuellen Firmware Version](https://smartfox.at/downloads/software/) EM2 00.01.07.03
2. Konfiguriere ihn über `evcc configure` und das Template [smartfox](https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/smartfox.yaml).
3. Versuche den Smartfox Pro über `evcc meter` auszulesen:
**Die HTTP GET Requests liefern EOF zurück, da die URI im Template nicht mehr unterstützt wird.**

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: c265a392155286ad587cd43ffc63798a17ca96ce5eac4fd58f4be01fec5cc3a8

interval: 30s # control cycle interval

meters:
- name: SE15K
  type: template
  template: solaredge-inverter
  id: 1
  host: 192.168.178.29
  port: 502
  usage: pv
  modbus: tcpip
  
- name: SmartfoxTemplate
  type: template
  template: smartfox
  host: 192.168.178.20
  usage: grid

- name: SmartfoxCustom
  type: custom
  power: # power (W)
    source: http
    uri: http://192.168.178.20/values.xml
    method: GET # default HTTP method
    jq: .values.value[] | select(.attrid=="detailsPowerValue")."#content" | rtrimstr(" W") # parse response json (https://jqplay.org is your friend)
    scale: 1 # floating point factor applied to result, e.g. 0.001 for kW to W conversion
    timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration

chargers:
- name: mystromGarage
  type: template
  template: mystrom
  host: 192.168.1.118
  standbypower: 15
  integrateddevice: true
  heating: false
  icon: bike

vehicles:
- name: eBike
  type: template
  template: offline
  title: e-Bike
  icon: bike
  capacity: 0.63
  phases: 1
  mode: pv
  mincurrent: 0.1
  maxcurrent: 1

loadpoints:
- title: Garage
  charger: mystromGarage
  vehicle: eBike
  # mode: off   # Charging is stopped, even if a vehicle is connected to the charger.
  # mode: now   # Start charging immediately at the maximum possible power.
  # mode: minpv # Start charging immediately at the minimum possible power. If sufficient PV surplus is available, charge faster.
  mode: pv    # Charge only using PV surplus and pause if there isn't enough power available.
  phases: 1
  mincurrent: 0.1
  maxcurrent: 10

site:
  title: Home
  meters: 
    pv: SE15K
    grid: SmartfoxTemplate
```

### Log details

```text
[main  ] INFO 2023/12/28 23:33:23 evcc 0.123.2
[main  ] INFO 2023/12/28 23:33:23 using config file: C:\tools\evcc\smartfox\evcc.yaml
[main  ] INFO 2023/12/28 23:33:23 starting ui and api at :7070
[db    ] INFO 2023/12/28 23:33:23 using sqlite database: C:\Users\patri\.evcc\evcc.db
[site  ] INFO 2023/12/28 23:33:24 site config:
[site  ] INFO 2023/12/28 23:33:24   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/12/28 23:33:24     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/12/28 23:33:24     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/12/28 23:33:24   vehicles:
[site  ] INFO 2023/12/28 23:33:24     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/12/28 23:33:24 loadpoint 1:
[lp-1  ] INFO 2023/12/28 23:33:24   mode:        pv
[lp-1  ] INFO 2023/12/28 23:33:24   charger:     power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/12/28 23:33:24   meters:      charge ✓
[lp-1  ] INFO 2023/12/28 23:33:24     charge:    power ✓ energy ✗ currents ✗
[lp-1  ] DEBUG 2023/12/28 23:33:24 phase timer inactive
[lp-1  ] DEBUG 2023/12/28 23:33:24 pv timer inactive
[lp-1  ] DEBUG 2023/12/28 23:33:24 guard timer inactive
[lp-1  ] INFO 2023/12/28 23:33:24 vehicle updated: unknown -> e-Bike
[lp-1  ] DEBUG 2023/12/28 23:33:24 set charge mode: pv
[site  ] DEBUG 2023/12/28 23:33:24 ----
[lp-1  ] DEBUG 2023/12/28 23:33:24 charge power: 0W
[site  ] DEBUG 2023/12/28 23:33:24 pv power: 0W
[site  ] ERROR 2023/12/28 23:33:25 grid meter: Get "http://192.168.178.20/all": EOF
[site  ] ERROR 2023/12/28 23:33:25 grid meter: Get "http://192.168.178.20/all": EOF
```

### What type of operating system are you running?

Windows

### Version

evcc version 0.123.2

premultiply, 2023-12-28:
Wie sieht denn http://192.168.178.20/values.xml aus?

rasenschachmatt, 2023-12-28:
In der Beschreibung oben ist ein Beispiel angehängt (das File heisst values.txt, da github keine xml Anhänge akzeptiert).

andig, 2023-12-30:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<values>
    <value id="hidWebinterfaceVersion">********************</value>
    <value id="macAddress">************</value>
    <value id="ipAddress">192.168.178.020</value>
    <value id="version">EM2  00.01.07.03</value>
    <value id="toGridValue">2.70 kW</value>
    <value id="lang">de</value>
    <value id="dateValue">2023-12-28</value>
    <value id="timeValue">22:52:12 Uhr</value>
    <value id="hidApiKey">********************************</value>
    <value id="socketIP">093.189.025.182</value>
    <value id="socketPort">5001</value>
    <value id="hidAoutMode">0</value>
    <value id="analogOutDescription">Boiler</value>
    <value id="analogOutPercent">0%</value>
    <value id="analogOutPower">0.00 &lt;span&gt;kW&lt;/span&gt;</value>
    <value id="analogOutTemp">0.0 &#176;C</value>
    <value id="hidAoutActivated">1</value>
    <value id="hidAoutStatus">9</value>
    <value id="hidAoutName">Boiler</value>
    <value id="hidCcType1">0</value>
    <value id="hidCcMode1">0</value>
    <value id="hidCcSt1">255</value>
    <value id="hidCcType2">0</value>
    <value id="hidCcMode2">0</value>
    <value id="hidCcSt2">255</value>
    <value id="hidCcType3">0</value>
    <value id="hidCcMode3">0</value>
    <value id="hidCcSt3">255</value>
    <value id="hidCcType4">0</value>
    <value id="hidCcMode4">0</value>
    <value id="hidCcSt4">255</value>
    <value id="hidCcType5">0</value>
    <value id="hidCcMode5">0</value>
    <value id="hidCcSt5">255</value>
    <value id="hidCcConnTimeout1">0</value>
    <value id="hidCcConnTimeout2">0</value>
    <value id="hidCcConnTimeout3">0</value>
    <value id="hidCcConnTimeout4">0</value>
    <value id="hidCcConnTimeout5">0</value>
    <value id="hidCc1Percent">0</value>
    <value id="hidCc2Percent">0</value>
    <value id="hidCc3Percent">0</value>
    <value id="hidCc4Percent">0</value>
    <value id="hidCc5Percent">0</value>
    <value id="hidCc1Rfid"></value>
    <value id="hidCc2Rfid"></value>
    <value id="hidCc3Rfid"></value>
    <value id="hidCc4Rfid"></value>
    <value id="hidCc5Rfid"></value>
    <value id="hidCc1Username"></value>
    <value id="hidCc2Username"></value>
    <value id="hidCc3Username"></value>
    <value id="hidCc4Username"></value>
    <value id="hidCc5Username"></value>
    <value id="hidCc1Info">0</value>
    <value id="hidCc2Info">0</value>
    <value id="hidCc3Info">0</value>
    <value id="hidCc4Info">0</value>
    <value id="hidCc5Info">0</value>
    <value id="hidCc1Name">CC</value>
    <value id="hidCc2Name">CC</value>
    <value id="hidCc3Name">CC</value>
    <value id="hidCc4Name">CC</value>
    <value id="hidCc5Name">CC</value>
    <value id="hidCc1Phase">1</value>
    <value id="hidCc2Phase">1</value>
    <value id="hidCc3Phase">1</value>
[... 364 lines cut ...]
    <value id="externalMeterConfigPane3Name">EM</value>
    <value id="externalMeterConfigPane3Ip">000.000.000.000</value>
    <value id="externalMeterConfigPane3Modbus">1</value>
    <value id="externalMeterConfigPane4Name">EM</value>
    <value id="externalMeterConfigPane4Ip">000.000.000.000</value>
    <value id="externalMeterConfigPane4Modbus">1</value>
    <value id="externalMeterConfigPane5Name">EM</value>
    <value id="externalMeterConfigPane5Ip">000.000.000.000</value>
    <value id="externalMeterConfigPane5Modbus">1</value>
    <value id="battery1ConfigPaneIp">000.000.000.000</value>
    <value id="carCharger1ConfigPaneIp">000.000.000.000</value>
    <value id="carCharger1ConfigPaneSerial">0</value>
    <value id="carCharger2ConfigPaneIp">000.000.000.000</value>
    <value id="carCharger2ConfigPaneSerial">0</value>
    <value id="carCharger3ConfigPaneIp">000.000.000.000</value>
    <value id="carCharger3ConfigPaneSerial">0</value>
    <value id="carCharger4ConfigPaneIp">000.000.000.000</value>
    <value id="carCharger4ConfigPaneSerial">0</value>
    <value id="carCharger5ConfigPaneIp">000.000.000.000</value>
    <value id="carCharger5ConfigPaneSerial">0</value>
    <value id="hidTcpConverterList">[]</value>
</values>
```

@premultiply hilft das?

andig, 2023-12-30:
@rasenschachmatt willst Du mal versuchen, ein neues Template zu erstellen? Für uns hat ads aktuell keine Priorität.

rasenschachmatt, 2023-12-30:
Hallo @andig
Ich schaue mir dies in den nächsten Tagen mal an. Ein YAML Template zu erstellen trau ich mir zu, mir ist nur noch nicht 100% klar, wie ich dies in evcc integrieren und testen kann, da die Software für mich noch neu ist und ich bisher nur eure Release Version verwendet habe. Würde mich dann wahrscheinlich wieder melden...

rasenschachmatt, 2023-12-30:
Das neue Template steht bereit. So weit wie möglich mit meinem SMARTFOX Pro getestet.

Siehe https://github.com/evcc-io/evcc/pull/11353

andig, 2024-01-02:
Mir ist nochmal aufgefallen, dass ja altes und neues Template auf den SF Pro abzielen. Damit bleibt für Naming und Doku die Frage, welches Template wann zu verwenden ist?

rasenschachmatt, 2024-01-02:
Das neue ist für die referenzierte Firmware Version.

Auf welcher Basis das alte entwickelt wurde, weiss ich leider nicht.

andig, 2024-01-02:
https://smartfox.at/softwareupdate/ - dann sollten wir das vllt. em2 nennen und in der Description mit der EM2 FW verknüpfen?

premultiply, 2024-01-02:
Oder die alte Version einfach ganz rauswerfen?

rasenschachmatt, 2024-01-02:
Das wäre aber ein Breaking Change für allfällige existierende Smartfox Benutzer, die noch auf der alten Firmware mit der "all" URI sind.
Würde ich persönlich nicht forcieren, aber ihr seid die Maintainers.

closed 2024-01-03 by andig

alaingiller, 2024-01-09:
Hallo @andig und @rasenschachmatt,

Ich habe den Smartfox template damals in evcc geschrieben. Ich habe selber ein SMARTFOX REG extended und bei mir funktionniert es nur mit /all, auch auf https://my.smartfox.at/liveview/D88039AC96E3 ist es mit /all gelöst. 
http://192.168.1.61/values.xml => gibt mir ein 404 zurück
http://192.168.1.61/all => kommt das json zurück, siehe https://github.com/evcc-io/evcc/pull/10253

Als ich den Template gemacht habe, bin ich davon gegangen dass alle Produkte gleich funktionieren sollen, aber sieht nicht so aus. => ich schlage vor in der smartfox template nur "Reg, Reg extended" in die description zu haben. Kann man auch die template umbennen aber es ist ein breaking change.

rasenschachmatt, 2024-01-09:
Hallo @alaingiller 
Danke für diese Information. Kannst du bitte einmal überprüfen, welche Firmware Version du auf deinem Smartfox hast?
Für den Smartfox Pro habe ich diese kürzlich auf die neuste Version aktualisiert (https://smartfox.de/downloads/software/), aber für den Smartfox Reg. Extended scheint es kein neues Update zu geben.

alaingiller, 2024-01-10:
Hallo, ja, ich habe schon der letzte FW