evcc-io/evcc #11324: Charging Plan: spread charging as much as possible to avoid peak loads

bolisaf, 2023-12-29:
**Is your feature request related to a problem? Please describe.**
charge plan charges the remaining charge after solar at peak power. I want to avoid peak charging as much as possible because of the 'capacity tariff' here in BE:
https://www.fluvius.be/nl/factuur-en-tarieven/capaciteitstarief/gezinnen-en-kleine-ondernemingen

**Describe the solution you'd like**
in addition to the current functionality for the charge plan, I would like to have an option to start charging right after sunset at the lowest power possible, based on the delta between current charge and target charge, and the time available (sunset to set departure time).

**Describe alternatives you've considered**
some automation in HA but then what's the use for EVCC...

**Additional context**
referring to a quote from @andig:
https://github.com/evcc-io/evcc/issues/9550#issuecomment-1722207893
> Plan charging tages as much PV as possible and covers the remainder at full speed.

StefanSchoof, 2023-12-29:
Sound similar to https://github.com/evcc-io/evcc/issues/9763

VolkerK62, 2023-12-29:
>in addition to the current functionality for the charge plan, I would like to have an option to start charging right after sunset at the lowest power possible, based on the delta between current charge and target charge, and the time available (sunset to set departure time).

with the latest Version of evcc this is possible. Not automatically, but just with a few clicks.
The Planner shows you a preview of the charging time and the chargePower. So you can adjust the maxcurrent within the ui to get the best solution.

bolisaf, 2023-12-29:
Yes, but would it retain that setting and would it adjust to changing sunset times throughout the year? I’d like to set and forget!

andig, 2023-12-29:
What‘s so particular over sunset that it matters?

bolisaf, 2023-12-29:
> What‘s so particular over sunset that it matters?

usually there's no more solar production after sunset so that can't influence the calculation for minimal charge power to reach the target charge in time :)

andig, 2024-01-02:
What specifically is a plan that spreads charging as much as possible? For which cost criteria would that optimise (that's what plans do). I'm not sure why this should become a plan. You can always set mode Min+Solar and adjust the session limit.

bolisaf, 2024-01-02:
Hi andig,

maybe if I provide some use-cases it's clearer:

- it's winter, so early sunset at 17h. car battery is at 40% and needs to be at 80% at 7AM. that's 14hours of charging, but the minimum power is 1400W so charging starts at say 21h to charge all night at minimal power
- same as before, only now it's summer and the sun sets at 22h. that means not enough time to charge at minimal power, so the power is increased a little bit to make sure the car is charget at the target SoC at 7AM.
- same scenario, but I arrive really late at 3AM. that means only 4 hours to get the car charged to Target SoC at 7AM, so the charging power is increased accordingly

does that make sense?

andig, 2024-01-09:
@naltatis @premultiply is "spread" charging something we want to consider?

premultiply, 2024-01-09:
Not really, as it would maximize losses and costs.

andig, 2024-01-10:
OPs point is that it minimises costs since he would pay more for power than for additional energy (Belgium amongst others).

naltatis, 2024-01-10:
One thing that's not yet clear for me: are we talking about evenly spreading through all remaining hours to keep max power as low as possible or must this now (or potentially later) also work with dynamic prices: "charge in the cheapest hours but don't exceed 5kW if possible". This would change how we'd have to think about the solution and calculate the cheapest option.

premultiply, 2024-01-10:
> OPs point is that it minimises costs since he would pay more for power than for additional energy (Belgium amongst others).

Ok, got it. But I'm still not sure whether it's really cheaper, for example, to buy twice the amount of energy than to pay for slightly higher power on the grid connection.

Oberreiter, 2024-01-12:
> @naltatis @premultiply is "spread" charging something we want to consider?

I would like this feature! It is god for the battery and for the grid.

StefanSchoof, 2024-01-12:
> It is god for the battery

Do you have any data that can support this? Does it really make a difference if you charging below 11 kW?

andig, 2024-01-12:
@StefanSchoof doesn't matter. We already have the case to reduce peak power.

andig, 2024-01-13:
Replaced by #11654 for getting broader view on requirements

closed 2024-01-13 by andig

Oberreiter, 2024-01-13:
> > It is god for the battery
> 
> Do you have any data that can support this? Does it really make a difference if you charging below 11 kW?

[https://ladehero.de/blog/so-behandeln-sie-ihren-e-auto-akku-richtig](https://ladehero.de/blog/so-behandeln-sie-ihren-e-auto-akku-richtig)
[https://www.e-mobileo.de/10-tipps-wie-der-akku-laenger-haelt/ ](https://www.e-mobileo.de/10-tipps-wie-der-akku-laenger-haelt/ )

hyperbart, 2024-02-02:
> One thing that's not yet clear for me: are we talking about evenly spreading through all remaining hours to keep max power as low as possible or must this now (or potentially later) also work with [...]

> > OPs point is that it minimises costs since he would pay more for power than for additional energy (Belgium amongst others).
> 
> Ok, got it. But I'm still not sure whether it's really cheaper, for example, to buy twice the amount of energy than to pay for slightly higher power on the grid connection.

@naltatis @premultiply 

Just as a reference/background info: Belgium system:

In every month they check per 15 min (h:00-h:15-h:30-h:45-h:00) what your average peak power consumption is.
For that month they write down "month peak power == 7kW"
They write this down for every month
and after a year they do an average of all your month peak power. You get a global single number, let' say 5kW
Every kW has a cost of around 40 EUR. So 200 EUR
If you would charge every month at 11kW that would be 440 EUR

What would be immensely cool (but you got my mind thinking) is if you would store the month peak for that month yourself (since you see that data if you have a meter) and adapt per month to allow EVCC to charge this fast since you are already penalized for that month because you did some other things...

premultiply, 2024-02-03:
Well setting `maxCurrent` by API from an external automation system should simply do the trick?

bolisaf, 2024-02-03:
I suppose so, but I don’t understand your logic here. I mean with the same reasoning I could just do everything in HA and not use EVCC at all.What we’re asking for here is a new feature to be added to this software. It doesn’t make sense if you then say ‘yes but you can do it using another software’, do you follow me? On the other hand I do understand you can’t add all the bells and whistles everyone is asking for, but here’s quite a few people who seem to be wishing for the same feature. And I’m sure that sooner or later this ‘capacity tariff’ will be introduced in some form in Germany as well ;)Thanks!L

bolisaf, 2024-02-03:
If this is correct then I’m looking at 440€ bill at the end of this year, and I the money I gained from solar charging doesn’t cover that.Thanks!L

hyperbart, 2024-02-03:
> Well setting `maxCurrent` by API from an external automation system should simply do the trick?

You are absolutely correct @premultiply and that's how I am doing it now but there is a certain opportunity for elegance to put this functionality at EVCC. In terms of energy usage and "flatten the curve". EVCC has evolved from an initially PV controlled charging solution to a cost efficient planner/charger and then my rationale is: "if you take into account cost efficiency we can't leave the peak power consumption/tariff behind in the requirements", but of course it is easy for me to say, I am the audience to which it concerns.

For me it should be a seperate mode, you don't have to bother everyone with it but a seperate mode or boolean to take this into account would be a good start for A LOT of people.

hyperbart, 2024-02-03:
> Hoeveel capaciteitstarief betaalt u voor een kilowatt?
>
> In 2024 betaalt u voor 1 kW op jaarbasis gemiddeld over Vlaanderen 39,93 euro (excl. btw); op maandbasis is dat 3,33 euro (excl. btw). Het bedrag verschilt van netbeheerder tot netbeheerder en hangt dus af van waar u woont.
>
> Iedereen betaalt een minimumbijdrage voor zijn gebruik van het net. Die komt overeen met 2,5 kW. Het gaat dus jaarlijks om gemiddeld 100 euro (excl. btw).
>
> Is uw maandpiek telkens 4 kW? Dan betaalt u op jaarbasis gemiddeld 160 euro (excl. btw) aan capaciteitstarief. Hebt u gedurende 1 van de 12 maanden uitzonderlijk een iets hogere piek van 6 kW? Dan kost u dat gemiddeld 6,65 euro (excl. btw) extra.
>
> De nettarieven voor 2024 per netbeheerder vindt u [hier](https://www.vreg.be/nl/hoeveel-bedragen-mijn-distributienettarieven).
>
> Alle info over het capaciteitstarief vindt u [hier](https://www.vreg.be/nl/nieuwe-nettarieven).

You doubt me? 😏 Never doubt me 😆 .
Just joking!

https://www.vreg.be/nl/veelgestelde-vragen/hoeveel-capaciteitstarief-betaalt-u-voor-een-kilowatt

naltatis, 2024-02-05:
@hyperbart thanks for the detailed input. This is quite a lot of change, you're proposing here :)
I don't see, that we are adding another mode to evcc in this form. But we've definitely also outgrown the initial meaning of our PV and Min+PV modes (planner, dynamic pricing, ...). Replacing those with one or maybe two "smart"-modes that come with configurability is something I think is likely to happen.

We will do a larger mode-redesign once the configuration UI is out the door. If you haven't already seen it, I'd recommend looking in https://github.com/evcc-io/evcc/discussions/3530. Even though it has been open for quite some time, we will definitely go through all the feedback and ideas in there before we'll make a proposal for a new mode-concept. I'd recommend you put a reference to this, a small summary of the core changes, in there so that we will find it later.